zephyrproject-rtos/zephyr #62831: Elf loader

mkschreder, 2023-09-19:
**Is your feature request related to a problem? Please describe.**

What's wrong with adding an elf loader to Zephyr? 

**Describe the solution you'd like**

I would like to be able to execute position independent elf binaries from flash formatted with ext2 that is already supported. This would pave the way towards porting busybox to zephyr and being able to run programs from flash. Of course the biggest issue is that this will have to be position independent executables that can be loaded into any portion of memory and execute in place from there. 

I think that if full posix compliance is implemented in zephyr then porting many existing tools will be very easy. It shouldn't really matter that zephyr is an rtos and not a desktop os. It should still be possible to build and run a wide variety of portable software on zephyr. 

**Describe alternatives you've considered**

Even minix has a collection of standalone utilities and is able to run x11. Zephyr today is bigger in size than minix yet it doesn't have the tools to port x11 to it. 

**Additional context**

https://github.com/Stichting-MINIX-Research-Foundation/minix/blob/master/minix/lib/libexec/exec_elf.c

http://edeproject.org/wiki/installingonminix

henrikbrixandersen, 2023-09-19:
Duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/2746

closed 2023-09-19 by henrikbrixandersen

nordicjm, 2023-09-20:
Loading code is one thing, what you ask for is something that simply is not going to happen, instead of a static application, you want a huge kernel that has every single feature enabled (which would be required for external applications to run). Instead of trying to make zephyr into something it is not i.e. linux, just go and use linux.

mkschreder, 2023-09-20:
> Loading code is one thing, what you ask for is something that simply is not going to happen, instead of a static application, you want a huge kernel that has every single feature enabled (which [...]

Zephyr is already reimplementing many of the basic utilities anyway (filesystem, network etc). This will get worse with time when custom utilities are continuously added for things like network config, file manipulations etc. I believe it would be better to make it an option to include many of these small utilities as they already exist. Busybox utilities are tiny. But they do require more complete posix support to work. From there, the new functionality can be added to zephyr in ways that are compatible with already existing unix tools. I do not see this as a drawback. Zephyr is small, modular and clean kernel that can be used as basis for building more complex systems that do not require an mmu.