zephyrproject-rtos/zephyr #62838: Twister fails at  raise Exception(f"Duplicate platform identifier {platform.name} found")

carriegong, 2023-09-20:
raise Exception(f"Duplicate platform identifier {platform.name} found")

I am running a custom board. or device tree. I am trying to run device testing on it.
I use the -A for the Board_root of the actual project. 
what causes this?.. 
i have added the device tree to the zephyr boards folder also.. but if i delete it i get unrecognized board. 

whats the correct way to run twister with a custom device tree?

PerMac, 2023-09-20:
Since recently boards from other projects are added by default.  FYI this is the PR https://github.com/zephyrproject-rtos/zephyr/pull/57484 Most likely this is your case, hence you shouldn't add additional -A in your call

carriegong, 2023-09-20:
> Since recently boards from other projects are added by default. FYI this is the PR #57484 Most likely this is your case, hence you shouldn't add additional -A in your call

ok but if i don't i get ERROR   - platform_filter - unrecognized platform XXXX
how do i fix that..

carriegong, 2023-09-20:
got it.nvm.

PerMac, 2023-09-21:
What was the issue then? What did help?

carriegong, 2023-09-28:
running into a different issue now, but i am not sure if i am adding in custom device tree in correctly. Currently i have it sitting in zephyr/boards/arm/ ...where does the overlay file go?

PerMac, 2023-10-03:
The github issues in zephyr don't work as "help desk" but are meant to point to issues to be resolved by other contributors. Also, one "real issue" : one github issue. You can try github discussions or discord channels (there are dozens of them, also device tree-specific) Please, don't reopen a closed issues unless the same problem is back or was not resolved in the first place. And the best is to first check the docs, since a lot of things are already explained there. Devicetree overlays are already explained there https://docs.zephyrproject.org/latest/build/dts/index.html#devicetree

closed 2023-10-03 by PerMac