zephyrproject-rtos/zephyr #62845: Hard fault when reading sensors using shell on nRF51

sense-Jo, 2023-09-20:
**Describe the bug**
When reading sensors using the `sensor get` command on the Zephyr shell on a nRF51 device, a hard fault occurs.
Reading sensors in the application seems to work though (I tried the `samples/sensor/die_temp_polling` sample).
I observed this on `nrf51dk_nrf51422` and a nRF51822-based board. The type of sensor does not seem to make a difference (I first noticed this with an external SHT30 connected via I²C).
Other platforms don't seem to be affected (I tried nRF52840 and STM32WB55RG).

This is a regression; using `git bisect` I found that commit ca5bf10143 (by @yperess) is the first with this bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a `nrf51dk_nrf51422` board to your PC
2. `cd zephyr/samples/sensor/sensor_shell`
3. `west build -p -b nrf51dk_nrf51422`
4. `west flash`
5. connect to the nRF51DK serial using a serial monitor, e.g. `picocom /dev/ttyACM0 -b 115200`
6. `sensor info` (optional)
7. `sensor get temp@4000c000`

**Expected behavior**
I expect to get sensor value readings as follows:
```
uart:~$ sensor info
device name: temp@4000c000, vendor: Nordic Semiconductor, model: nrf-temp, friendly name: (null)
uart:~$ sensor get temp@4000c000
channel idx=12 die_temp =  23.500000
uart:~$
```

**Impact**
No relevant impact on me.

**Logs and console output**
The hard fault (on commit ed380de152):
```
uart:~$ sensor info
device name: temp@4000c000, vendor: Nordic Semiconductor, model: nrf-temp, friendly name: (null)
uart:~$ sensor get temp@4000c000
[00:00:15.710,723] <err> os: ***** HARD FAULT *****
[00:00:15.710,876] <err> os: r0/a1:  0x00000005  r1/a2:  0x00000005  r2/a3:  0x00000005
[00:00:15.710,968] <err> os: r3/a4:  0x20001331 r12/ip:  0x00000005 r14/lr:  0x00007a09
[00:00:15.711,029] <err> os:  xpsr:  0x21000000
[00:00:15.711,090] <err> os: Faulting instruction address (r15/pc): 0x00007ada
[00:00:15.711,273] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:15.711,456] <err> os: Current thread: 0x200008a0 (shell_uart)
[00:00:15.784,423] <err> os: Halting system
```

When attaching GDB, I can get the following backtrace (image and text are identical):
[image]
```
(gdb) bt
#0  arch_system_halt (reason=reason@entry=0) at /home/jo/zephyrproject/zephyr/kernel/fatal.c:32
#1  0x00008f8c in k_sys_fatal_error_handler (reason=reason@entry=0, esf=esf@entry=0x20003038 <z_interrupt_stacks+1984>) at /home/jo/zephyrproject/zephyr/kernel/fatal.c:46
#2  0x00009058 in z_fatal_error (reason=0, reason@entry=536883240, esf=esf@entry=0x20003038 <z_interrupt_stacks+1984>) at /home/jo/zephyrproject/zephyr/kernel/fatal.c:131
#3  0x00005d7c in z_arm_fatal_error (reason=536883240, reason@entry=0, esf=0x20003038 <z_interrupt_stacks+1984>, esf@entry=0x20003058 <z_interrupt_stacks+2016>) at /home/jo/zephyrproject/zephyr/arch/arm/core/aarch32/fatal.c:63
#4  0x00006078 in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>) at /home/jo/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:1132
#5  0x0000612c in z_arm_hard_fault () at /home/jo/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:102
#6  <signal handler called>
#7  0x00007ada in sensor_submit_fallback (iodev_sqe=0x0 <_vector_table>, dev=0x20001b38 <_sqe_pool_sensor_read_rtio_sqe_pool>) at /home/jo/zephyrproject/zephyr/drivers/sensor/default_rtio_sensor.c:181
#8  sensor_iodev_submit (iodev_sqe=0x0 <_vector_table>) at /home/jo/zephyrproject/zephyr/drivers/sensor/default_rtio_sensor.c:26
#9  0x0000ccc4 in rtio_iodev_submit (iodev_sqe=<optimized out>) at /home/jo/zephyrproject/zephyr/subsys/rtio/rtio_executor.c:28
#10 0x0000cd08 in rtio_executor_submit (r=r@entry=0x20000340 <sensor_read_rtio>) at /home/jo/zephyrproject/zephyr/subsys/rtio/rtio_executor.c:94
#11 0x000074c2 in z_impl_rtio_submit (r=0x20000340 <sensor_read_rtio>, wait_count=0) at /home/jo/zephyrproject/zephyr/include/zephyr/rtio/rtio.h:1424
#12 rtio_submit (r=0x20000340 <sensor_read_rtio>, wait_count=0) at /home/jo/zephyrproject/zephyr/samples/sensor/sensor_shell/build/zephyr/include/generated/syscalls/rtio.h:161
#13 sensor_read (iodev=0x2000036c <iodev_sensor_shell_read>, ctx=0x20000340 <sensor_read_rtio>, userdata=0x20002668 <shell_uart_stack+1520>) at /home/jo/zephyrproject/zephyr/include/zephyr/drivers/sensor.h:893
#14 cmd_get_sensor (sh=<optimized out>, argc=2, argv=<optimized out>) at /home/jo/zephyrproject/zephyr/drivers/sensor/sensor_shell.c:309
#15 0x000047e4 in exec_cmd (help_entry=0x200026d0 <shell_uart_stack+1624>, argv=0x200026e8 <shell_uart_stack+1648>, argc=<optimized out>, sh=0xd980 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:558
#16 execute (sh=sh@entry=0xd980 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:800
#17 0x00004a32 in state_collect (sh=0xd980 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1004
#18 shell_process (sh=0xd980 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1475
#19 0x0000b692 in shell_signal_handle (sh=sh@entry=0xd980 <shell_uart>, sig_idx=sig_idx@entry=SHELL_SIGNAL_RXRDY, handler=handler@entry=0x4875 <shell_process>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1292
#20 0x000050f4 in shell_thread (arg_log_backend=<optimized out>, arg_log_level=<optimized out>, shell_handle=0xd980 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1351
#21 shell_thread (shell_handle=0xd980 <shell_uart>, arg_log_backend=<optimized out>, arg_log_level=<optimized out>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1308
#22 0x0000a840 in z_thread_entry (entry=0x5061 <shell_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /home/jo/zephyrproject/zephyr/lib/os/thread_entry.c:36
#23 0xaaaaaaaa in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.1

yperess, 2023-09-20:
@sense-Jo while I'm working on getting a board to test this on, could you try one quick thing before we go too far. The decoding logic is about to change anyway so I want to see if that fixes the issue. Could you checkout https://github.com/zephyrproject-rtos/zephyr/pull/61022 and try the same thing?

Thanks.

sense-Jo, 2023-09-21:
> @sense-Jo while I'm working on getting a board to test this on, could you try one quick thing before we go too far. The decoding logic is about to change anyway so I want to see if that fixes the issue. Could you checkout #61022 and try the same thing?

Sure, I just tried the same with the PR checked out. Unfortunately, with pretty much the same result:

```
uart:~$ sensor info
device name: temp@4000c000, vendor: Nordic Semiconductor, model: nrf-temp, friendly name: (null)
uart:~$ sensor get temp@4000c000
[00:00:25.567,749] <err> os: ***** HARD FAULT *****
[00:00:25.567,932] <err> os: r0/a1:  0x00000005  r1/a2:  0x00000005  r2/a3:  0x00000005
[00:00:25.567,993] <err> os: r3/a4:  0x20001a35 r12/ip:  0x00000005 r14/lr:  0x000089f1
[00:00:25.568,054] <err> os:  xpsr:  0x21000000
[00:00:25.568,145] <err> os: Faulting instruction address (r15/pc): 0x00008ac2
[00:00:25.568,328] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:25.568,511] <err> os: Current thread: 0x200008a8 (shell_uart)
[00:00:25.641,540] <err> os: Halting system
```

```
(gdb) bt
#0  arch_system_halt (reason=reason@entry=0) at /home/jo/zephyrproject/zephyr/kernel/fatal.c:32
#1  0x00009f80 in k_sys_fatal_error_handler (reason=reason@entry=0, esf=esf@entry=0x20003738 <z_interrupt_stacks+1984>) at /home/jo/zephyrproject/zephyr/kernel/fatal.c:46
#2  0x0000a04c in z_fatal_error (reason=0, reason@entry=536885032, esf=esf@entry=0x20003738 <z_interrupt_stacks+1984>) at /home/jo/zephyrproject/zephyr/kernel/fatal.c:131
#3  0x000068ec in z_arm_fatal_error (reason=536885032, reason@entry=0, esf=0x20003738 <z_interrupt_stacks+1984>, esf@entry=0x20003758 <z_interrupt_stacks+2016>) at /home/jo/zephyrproject/zephyr/arch/arm/core/aarch32/fatal.c:73
#4  0x00006bec in z_arm_fault (msp=<optimized out>, psp=<optimized out>, exc_return=<optimized out>, callee_regs=<optimized out>) at /home/jo/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:1157
#5  0x00006cb0 in z_arm_hard_fault () at /home/jo/zephyrproject/zephyr/arch/arm/core/aarch32/cortex_m/fault_s.S:102
#6  <signal handler called>
#7  0x00008ac2 in sensor_submit_fallback (iodev_sqe=0x0 <_vector_table>, dev=0x2000223c <_sqe_pool_sensor_read_rtio_sqe_pool>) at /home/jo/zephyrproject/zephyr/drivers/sensor/default_rtio_sensor.c:181
#8  sensor_iodev_submit (iodev_sqe=0x0 <_vector_table>) at /home/jo/zephyrproject/zephyr/drivers/sensor/default_rtio_sensor.c:26
#9  0x0000dc7a in rtio_iodev_submit (iodev_sqe=<optimized out>) at /home/jo/zephyrproject/zephyr/subsys/rtio/rtio_executor.c:28
#10 0x0000dcbe in rtio_executor_submit (r=r@entry=0x20000344 <sensor_read_rtio>) at /home/jo/zephyrproject/zephyr/subsys/rtio/rtio_executor.c:94
#11 0x000084fe in z_impl_rtio_submit (r=0x20000344 <sensor_read_rtio>, wait_count=0) at /home/jo/zephyrproject/zephyr/include/zephyr/rtio/rtio.h:1418
#12 rtio_submit (r=0x20000344 <sensor_read_rtio>, wait_count=0) at /home/jo/zephyrproject/zephyr/samples/sensor/sensor_shell/build/zephyr/include/generated/syscalls/rtio.h:163
#13 sensor_read (iodev=0x20000374 <iodev_sensor_shell_read>, ctx=0x20000344 <sensor_read_rtio>, userdata=0x20002d68 <shell_uart_stack+1520>) at /home/jo/zephyrproject/zephyr/include/zephyr/drivers/sensor.h:915
#14 cmd_get_sensor (sh=<optimized out>, argc=2, argv=<optimized out>) at /home/jo/zephyrproject/zephyr/drivers/sensor/sensor_shell.c:399
#15 0x00005354 in exec_cmd (help_entry=0x20002dd0 <shell_uart_stack+1624>, argv=0x20002de8 <shell_uart_stack+1648>, argc=<optimized out>, sh=0xeb38 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:558
#16 execute (sh=sh@entry=0xeb38 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:800
#17 0x000055a2 in state_collect (sh=0xeb38 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1004
#18 shell_process (sh=0xeb38 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1475
#19 0x0000c662 in shell_signal_handle (sh=sh@entry=0xeb38 <shell_uart>, sig_idx=sig_idx@entry=SHELL_SIGNAL_RXRDY, handler=handler@entry=0x53e5 <shell_process>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1292
#20 0x00005c64 in shell_thread (arg_log_backend=<optimized out>, arg_log_level=<optimized out>, shell_handle=0xeb38 <shell_uart>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1351
#21 shell_thread (shell_handle=0xeb38 <shell_uart>, arg_log_backend=<optimized out>, arg_log_level=<optimized out>) at /home/jo/zephyrproject/zephyr/subsys/shell/shell.c:1308
#22 0x0000b810 in z_thread_entry (entry=0x5bd1 <shell_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /home/jo/zephyrproject/zephyr/lib/os/thread_entry.c:48
#23 0xaaaaaaaa in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

k0d, 2023-09-28:
I'm getting exactly the same using a nucleo-l053r8 and a rp2040, with various sensors.

I can confirm that the committ https://github.com/zephyrproject-rtos/zephyr/commit/ca5bf1014390008d2e5cc04caca4f9c24aa60057 is what breaks it for me also.

fabiobaltieri, 2023-10-03:
cc @MaureenHelm @avisconti @teburd @tristan-google

fabiobaltieri, 2023-10-03:
This is probably due to https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/default_rtio_sensor.c#L124, it's picking a potentially unaligned point in the buffer and using it as a 32 bit value. That function needs one or more copies of that and memcpy the data into and out of it.

k0d, 2023-10-03:
@fabiobaltieri Do you have a suggestion of some code I can test to fix it?

fabiobaltieri, 2023-10-03:
> @fabiobaltieri Do you have a suggestion of some code I can test to fix it?

hey I think @yperess is looking into it already

yperess, 2023-10-03:
@sense-Jo I have a PR that's already in flight to tweak some sensor code paths, I've added a fix for this issue into it and was able to verify on an M7 core with `CONFIG_TRAP_UNALIGNED_ACCESS` enabled. Can you verify https://github.com/zephyrproject-rtos/zephyr/pull/60063 works for you?

k0d, 2023-10-03:
@yperess I've tested this on a Nucleo-G071RB and it's still getting the same error. I don't see anything in 'default_rtio_sensor.c' that seems to fix it anyway. Did you forget to commit something?

yperess, 2023-10-03:
@k0d I'm sorry, it was close, the fix was adding the padding to the `struct sensor_data_generic_header` but I used the CONFIG__TRAP_UNALIGNED_ACCESS to guard that  padding (I didn't realize that the M0 cores didn't `select TRAP_UNALIGNED_ACCESS` #63481 should fix things

k0d, 2023-10-03:
@yperess That works...thanks 🥇

sense-Jo, 2023-10-06:
I can confirm that with https://github.com/zephyrproject-rtos/zephyr/pull/63481, the sensor shell works on nRF51 as well :+1:

fabiobaltieri, 2023-10-06:
Thanks for reporting and for to help testing folks.

closed 2023-10-06 by fabiobaltieri