zephyrproject-rtos/zephyr #62872: Define Requirements Management process

gregshue, 2023-09-20:
**Is your feature request related to a problem? Please describe.**
The Security Committee and Safety Committee are each responsible for identifying, capturing and approving requirements in their respective areas - but there is no publicly defined process for content outside of those areas or for establishing consistency across those areas. Specifically, the following are missing from public view:
1. How any conflicts between Security Requirements and Safety Requirements are resolved.
2. How constraints on the Zephyr Project are submitted, reviewed, and approved. (e.g., Originated code shall be released under Apache 2.0 license).
3. How user needs are submitted, reviewed, and approved.
4. How the set of system requirements (those to be verified for a particular release) are chosen and the set verified for completeness and consistency.

**Describe the solution you would like to see**
We need to have a single public process defined for everything related to managing requirements. This public process will need to satisfy **all** certifications sought by the Zephyr Project. If that content cannot be made public, it should be described using terms from http://swebokwiki.org/Chapter_1:_Software_Requirements, which is one of the sources that is the basis of the **free** self-paced course [Linux Foundation: Software Engineering Basics for Embedded Systems](https://training.linuxfoundation.org/training/software-engineering-basics-for-embedded-systems-lfd116/).
 
**Describe alternatives you've considered**
ISO/IEC/IEEE standards for requirements management processes are behind a paywall, which we want to avoid.

**Additional context**

gregshue, 2023-09-20:
A solution is needed to address https://github.com/zephyrproject-rtos/reqmgmt/pull/5.

gregshue, 2023-10-06:
Requirements Management for software engineering (SwE) has parallel issues as Requirements Management for Systems Engineering (SE). Here are some guides on requirements management from the [International Council on Systems Engineering (INCOSE)](incose.org):

- [INCOSE Managing Requirements for Design - pamphlet](https://www.incose.org/docs/default-source/TWG-Documents/003-requirements-management-and-se-pamphlet.pdf?sfvrsn=24c882c6_2). More details available in:
  - [INCOSE Requirements Management Tutorial (2019) - 30 pgs](https://sdincose.org/wp-content/uploads/2019/11/RickHefner-Oct2019-INCOSE-SD-Requirements-Tutorial.pdf)
  - [INCOSE Guide to Writing Requirements (2023) - 140 pgs](https://www.incose.org/docs/default-source/working-groups/requirements-wg/gtwr/[incose_rwg_gtwr_v4_040423_final_drafts.pdf](https://www.incose.org/docs/default-source/working-groups/requirements-wg/gtwr/incose_rwg_gtwr_v4_040423_final_drafts.pdf?sfvrsn=5c877fc7_2)?sfvrsn=5c877fc7_2)

There is some conflict between INCOSE terminology and that used in ISO/IEC/IEEE 15288 _Systems and software engineering - Systems life cycle processes_ and ISO/IEC/IEEE 29148 _Systems and software engineering - life cycle processes - Requirements engineering_.  I found a discussion of this [here](https://www.linkedin.com/pulse/what-stakeholder-requirements-keith-collyer-2e).

> The authors of the manual are using the term “stakeholder needs” to mean what has traditionally been called “stakeholder requirements” and using “stakeholder requirements” to define a specific subset of system requirements, again going against decades of usage.

One of the early decisions we will need to resolve is about terminology. The Security Architect @ceolin and the Safety Architect @simhein need to be in this discussion. BTW, who is overseeing the requirements that are not related to security or safety?

ceolin, 2023-10-06:
> Requirements Management for software engineering (SwE) has parallel issues as Requirements Management for Systems Engineering (SE). Here are some guides on requirements management from the [International Council on Systems Engineering (INCOSE)](incose.org):
> 
> * [INCOSE Managing Requirements for Design - pamphlet](https://www.incose.org/docs/default-source/TWG-Documents/003-requirements-management-and-se-pamphlet.pdf?sfvrsn=24c882c6_2). More details available in:
>   
>   * [INCOSE Requirements Management Tutorial (2019) - 30 pgs](https://sdincose.org/wp-content/uploads/2019/11/RickHefner-Oct2019-INCOSE-SD-Requirements-Tutorial.pdf)
>   * [INCOSE Guide to Writing Requirements (2023) - 140 pgs](https://www.incose.org/docs/default-source/working-groups/requirements-wg/gtwr/%5Bincose_rwg_gtwr_v4_040423_final_drafts.pdf%5D(https://www.incose.org/docs/default-source/working-groups/requirements-wg/gtwr/incose_rwg_gtwr_v4_040423_final_drafts.pdf?sfvrsn=5c877fc7_2)?sfvrsn=5c877fc7_2)
> 
> There is some conflict between INCOSE terminology and that used in ISO/IEC/IEEE 15288 _Systems and software engineering - Systems life cycle processes_ and ISO/IEC/IEEE 29148 _Systems and software engineering - life cycle processes - Requirements engineering_. I found a discussion of this [here](https://www.linkedin.com/pulse/what-stakeholder-requirements-keith-collyer-2e).
> 
> > The authors of the manual are using the term “stakeholder needs” to mean what has traditionally been called “stakeholder requirements” and using “stakeholder requirements” to define a specific subset of system requirements, again going against decades of usage.
> 
> One of the early decisions we will need to resolve is about terminology. The Security Architect @ceolin and the Safety Architect @simhein need to be in this discussion. BTW, who is overseeing the requirements that are not related to security or safety?

Functional requirement ? 

Do we really need special ways to handle safety / security / functional requirements ?

gregshue, 2023-10-07:
Functional and non-functional requirements. We need one process that covers all requirement management whether or not they are for safety or security.

simhein, 2023-10-11:
> Functional requirement ?
>
> Do we really need special ways to handle safety / security / functional requirements ?

I think we do not need special ways to handle the requirements different, but what we need is a way to label the different requirements correctly and a way to filter them as needed. And the safety WG/Committee took that into account while choosing a tool for requirements management tool.

About the topic of having a process right from the start I'm a little bit unsure right now because the safety WG is at the start of working with requirements and in the process of exploring such a process which maybe could work for the project.
So my proposal right now would be that the safety WG( and everyone who is interested ) work further on that topic and if we think we have a good idea of a working process for the process, we can present it to the process working group to get more feedback and discuss the topic further.

nashif, 2023-10-11:
as already discussed, this is already being dealt with in the safety wg. No need to have a parallel discussion in the process WG right now especially if all those involved are already participating in the discussions.

gregshue, 2023-10-12:
For the record, this issue was created after multiple statements in the Safety WG that there were no Security concerns because there were not any Security requirements yet, and that the functional and non-functional requirements stated in the Zephyr Project Charter were out of scope. 

Now that the Safety WG has explicitly agreed it will be looking at Requirements Management for the whole project (including Security concerns) it is appropriate to close this issue.

closed 2023-10-12 by gregshue