zephyrproject-rtos/zephyr #62883: [Interrupt][Multi-level] Configurable level bits

ycsin, 2023-09-21:
**Is your enhancement proposal related to a problem? Please describe.**
#61050 introduced the support of configurable bitmasks for multi-level interrupt, however, there might be parts of Zephyr still hardcoding the number of bits for the levels, i.e.:
- https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/dts/gen_defines.py#L463
- https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/gpio/gpio_sifive.c#L107

**Describe the solution you'd like**
Identify the files that hardcode the values, update them to use Kconfig

**Describe alternatives you've considered**
Make the bits used by first level to be constantly 8

cfriedt, 2023-09-21:
This technically would fall into the category of a bug. Can you grab the template for a bug report and update the description?

cfriedt, 2023-10-05:
@fabiobaltieri @jhedberg - this will be pretty critical for functional risc-v plic.

PR #63289 (2023-10-30): arch: build: interrupt: encode multi level IRQ with C macro magic
This PR encodes the Zephyr multi-level interrupt using C macro magic in the devicetree, rather than depending on the [`gen_defines.py`](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/dts/gen_defines.py) script during build:

https://github.com/zephyrproject-rtos/zephyr/blob/1fc16e65656e742fbe0adc89e0fdf685c51525a5/scripts/dts/gen_defines.py#L435-L441

https://github.com/zephyrproject-rtos/zephyr/blob/1fc16e65656e742fbe0adc89e0fdf685c51525a5/scripts/dts/gen_defines.py#L456-L469

https://github.com/zephyrproject-rt [... cut]
closed 2023-10-30 by cfriedt