zephyrproject-rtos/zephyr #62886: tests-ci : libraries: libc: newlib.thread_safety.userspace.stress test Timeout

hakehuang, 2023-09-21:
**Describe the bug**
newlib.thread_safety.userspace.stress test is Timeout on zephyr-v3.4.0-3926-g4ba97c22555e on mimxrt595_evk_cm33
testcase path is libraries/libc/newlib/thread_safety/userspace/stress/malloc_thread_safety

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt595_evk_cm33  --sub-test libraries.libc
```
or
```
# cd tests/libraries/libc/newlib/thread_safety/userspace/stress/malloc_thread_safety
# west build -b mimxrt595_evk_cm33
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.4.0-3926-g4ba97c22555e ***
E: no free partition slots available
Running TESTSUITE newlib_thread_safety_locks
===================================================================
START - test_env_lock
PASS - test_env_lock in 0.001 seconds
===================================================================
START - test_malloc_lock
PASS - test_malloc_lock in 0.001 seconds
===================================================================
START - test_retargetable_lock_mutex
PASS - test_retargetable_lock_mutex in 0.001 seconds
===================================================================
START - test_retargetable_lock_sem
PASS - test_retargetable_lock_sem in 0.001 seconds
===================================================================
START - test_sfp_lock
PASS - test_sfp_lock in 0.001 seconds
===================================================================
START - test_sinit_lock
PASS - test_sinit_lock in 0.001 seconds
===================================================================
START - test_tz_lock
PASS - test_tz_lock in 0.001 seconds
===================================================================
TESTSUITE newlib_thread_safety_locks succeeded
Running TESTSUITE newlib_thread_safety_stress
===================================================================
START - test_malloc_thread_safety
Created 64 worker threads.
Waiting 30 seconds to see if any failures occur ...
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x301a0ee4
E: r0/a1:  0x00000020  r1/a2:  0x00000120  r2/a3:  0x00000020
E: r3/a4:  0x00000121 r12/ip:  0x00000002 r14/lr:  0x1800ece1
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x1800eaea
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x301817b8 (unknown)
E: Halting system

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.4.0-3926-g4ba97c22555e

hakehuang, 2023-09-21:
related to #62885

decsny, 2024-02-14:
does not reproduce on latest main 61f065230ad9bca91633cd57b9dcd5e90c5d8bc7