zephyrproject-rtos/zephyr #62887: tests-ci : kernel: threads: tls.userspace test Timeout

hakehuang, 2023-09-21:
**Describe the bug**
tls.userspace test is Timeout on zephyr-v3.4.0-3926-g4ba97c22555e on mimxrt595_evk_cm33
testcase path is kernel/threads/tls/userspace/tls_userspace

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt595_evk_cm33  --sub-test kernel.threads
```
or
```
# cd tests/kernel/threads/tls/userspace/tls_userspace
# west build -b mimxrt595_evk_cm33
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.4.0-3926-g4ba97c22555e ***
E: no free partition slots available
Running TESTSUITE thread_tls
===================================================================
E: num_parts of 3 exceeds maximum allowable partitions (2)
ASSERTION FAIL [ret == 0] @ WEST_TOPDIR/zephyr/tests/kernel/threads/tls/src/main.c:219
k_mem_domain_init() failed -22
E: r0/a1:  0x00000004  r1/a2:  0x000000db  r2/a3:  0x00000016
E: r3/a4:  0x00000000 r12/ip:  0x0000000c r14/lr:  0x1800d831
E:  xpsr:  0x69000000
E: Faulting instruction address (r15/pc): 0x1800d84a
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x301812e8 (main)
E: Halting system

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.4.0-3926-g4ba97c22555e

hakehuang, 2023-09-21:
related to #62885

decsny, 2023-10-18:
I think this is another issue caused because of the number of static MPU regions being used up on these platforms

closed 2024-01-01 by github-actions[bot]