zephyrproject-rtos/zephyr #62889: tests-ci : portability: posix: common.dynamic_stack test Failed

hakehuang, 2023-09-21:
**Describe the bug**
common.dynamic_stack test is Failed on zephyr-v3.4.0-3926-g4ba97c22555e on mimxrt1060_evk
testcase path is portability/posix/common/dynamic_stack/nanosleep_execution

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1060_evk  --sub-test portability.posix
```
or
```
# cd tests/portability/posix/common/dynamic_stack/nanosleep_execution
# west build -b mimxrt1060_evk
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
START - test_nanosleep_execution

Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/nanosleep.c:153: common_lower_bound_check: (actual_ns >= exp_ns is false)
actual: 96774 expected: 500000000
FAIL - test_nanosleep_execution in 0.016 seconds

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.4.0-3926-g4ba97c22555e

mmahadevan108, 2023-10-04:
@hakehuang , can you please try on the latest master. The test passed for me on RT685 EVK board.

mmahadevan108, 2023-10-04:
The behavior is strange, it hangs at test
`START - test_nanosleep_errors_errno` when I run it from a reset, however the test passes on board powerup.
It hangs at a call to `k_sleep` because the value passed in to sleep is very high. Not sure what is causing this.

hakehuang, 2023-10-05:
> hake, do we have a bisect on when this started failing?

per my analysis, the posix test cases is newly updated and cause this problem. some of them may be resolved by align the timer.

mmahadevan108, 2023-10-06:
@hakehuang , can you provide more details on what additions are needed to the timer?

hakehuang, 2023-10-07:
> @hakehuang , can you provide more details on what additions are needed to the timer?
ewfwe to this, basically, we need add a sleep before testing.
https://github.com/zephyrproject-rtos/zephyr/pull/63210

hakehuang, 2023-10-07:
> @hakehuang , can you provide more details on what additions are needed to the timer?
please refer to below pr.  Basically, we need add a sleep before testing.
https://github.com/zephyrproject-rtos/zephyr/pull/63210

PR #63667 (2023-10-13): test: posix: common: nanosleep test fix
in test_nanosleep_execution, the

actual_ns = k_cyc_to_ns_ceil64((now - then * selection));

is wrong, according to posix driver code

lib/posix/clock.c

line: 156
	if ((flags & TIMER_ABSTIME) == 0) {
		ns += uptime_ns;
	}

so we need change to

actual_ns = k_cyc_to_ns_ceil64((now + then * selection));

also to increase accuracy in comparison as the sleep is using k_sleep, change to k_cycle_get_64

Fixes: #62889
closed 2023-10-13 by jhedberg