zephyrproject-rtos/zephyr #62892: tests-ci : portability: posix: common.no_spin_validate test Failed

hakehuang, 2023-09-21:
**Describe the bug**
common.no_spin_validate test is Failed on zephyr-v3.4.0-3926-g4ba97c22555e on mimxrt1060_evk
testcase path is portability/posix/common/no_spin_validate/nanosleep_execution

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1060_evk  --sub-test portability.posix
```
or
```
# cd tests/portability/posix/common/no_spin_validate/nanosleep_execution
# west build -b mimxrt1060_evk
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
START - test_nanosleep_execution

Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/nanosleep.c:153: common_lower_bound_check: (actual_ns >= exp_ns is false)
actual: 96629 expected: 500000000
FAIL - test_nanosleep_execution in 0.016 seconds

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.4.0-3926-g4ba97c22555e

hakehuang, 2023-09-21:
related: #62889

decsny, 2024-02-12:
does not reproduce on latest main

closed 2024-02-12 by decsny