zephyrproject-rtos/zephyr #62897: tests-ci : portability: posix: common test Failed

hakehuang, 2023-09-21:
**Describe the bug**
common test is Failed on zephyr-v3.4.0-3926-g4ba97c22555e on mimxrt1060_evk
testcase path is portability/posix/common/nanosleep_execution

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1060_evk  --sub-test portability.posix
```
or
```
# cd tests/portability/posix/common/nanosleep_execution
# west build -b mimxrt1060_evk
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
START - test_nanosleep_execution

Assertion failed at WEST_TOPDIR/zephyr/tests/posix/common/src/nanosleep.c:153: common_lower_bound_check: (actual_ns >= exp_ns is false)
actual: 96449 expected: 500000000
FAIL - test_nanosleep_execution in 0.016 seconds

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.4.0-3926-g4ba97c22555e

hakehuang, 2023-09-21:
related: #62889

hakehuang, 2023-09-27:
maybe we need simliar solution like https://github.com/zephyrproject-rtos/zephyr/pull/63086

decsny, 2024-02-12:
Hi @hakehuang , I was not able to reproduce this on the latest 08f163cdf6c23e25a1b6a76d1ed13c792d536542

decsny, 2024-02-12:
does not reproduce on latest main

closed 2024-02-12 by decsny