zephyrproject-rtos/zephyr #62898: tests-ci : kernel: timer: timer test Failed

hakehuang, 2023-09-21:
**Describe the bug**
timer test is Failed on zephyr-v3.4.0-3926-g4ba97c22555e on lpcxpresso54114_m4
testcase path is kernel/timer/timer/jitter_drift_timer_period

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso54114_m4  --sub-test kernel.timer
```
or
```
# cd tests/kernel/timer/timer/jitter_drift_timer_period
# west build -b lpcxpresso54114_m4
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
START - test_jitter_drift_timer_period
periodic timer behavior test using built-in restart mechanism
collecting time samples for approx 10 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 48000000, kernel tick rate 10000
period duration statistics for 10000 samples (0 rollovers):
expected: 1000 us,       	48000.000000 cycles
min:      899.916667 us, 	43196 cycles
max:      1000.000000 us, 	48000 cycles
mean:     999.989992 us, 	47999.519600 cycles
variance: 1.001567 us, 	2307.610816 cycles
stddev:   1.000783 us, 	48.037598 cycles
timer start cycle 1345313, end cycle 481340509,
total time 9999899.916667 us, expected time 10000000.000000 us,
expected time drift 0.000000 us, difference -100.083333 us

Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:248: do_test_using: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than expected 10%)
FAIL - test_jitter_drift_timer_period in 10.328 seconds

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v3.4.0-3926-g4ba97c22555e

hakehuang, 2023-09-21:
related: #58729

teburd, 2023-09-26:
See #63086 as a possible fix

mmahadevan108, 2023-10-04:
@hakehuang , can this be closed?

closed 2023-10-05 by hakehuang