zephyrproject-rtos/zephyr #62913: Add hostapd OSS project to Zephyr as a Wi-Fi NM

krish2718, 2023-09-21:
## Origin

hostap: https://w1.fi/cgit/hostap/

## Purpose

WPA supplicant (which is a subset of hostap) is a user-space daemon for IEEE 802.1x supplicant that handles IEEE 802.11
authentication and association of the WLAN driver. It implements key negotiation with a WPA Authenticator and it controls
the roaming and IEEE 802.11 authentication/association of the wlan driver.

## Mode of integration

This would be a Zephyr module with `hostap` as the name.

## Maintainership

Maintainers - @krish2718 @jukkar

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/62784

## Description

In Zephyr most existing Wi-Fi chipsets offload the entire Wi-Fi functionality to the chip. They only rely
on Zephyr's Wi-Fi management APIs to configure the chip. But for some chipsets, the Wi-Fi functionality is
not offloaded to the chip. In such cases, there is a need for a module that can handle the Wi-Fi functionality
esp. the IEEE 802.11 authentication and association. And WPA supplicant the de-facto standard for IEEE 802.11
authentication and association. It supports all the IEEE 802.11 authentication methods and it is widely used
in Linux and other OSes. So it makes sense to use WPA supplicant as the Wi-Fi management module for such chipsets.

Apart from authentication and association, WPA supplicant also handles a few other Wi-Fi features such as
roaming, WPS, FILS, DPP etc. But these features are not mandatory for a Wi-Fi management module. So we can choose to
not use these features and still use WPA supplicant as the Wi-Fi management module.

## Dependencies

Zephyr is integrated in to this component via NM feature (Network Manager).

## Revision

e792f38db8471f35ee94f109a4093dece8f78c6a is used as a starting point for the integration.

## License

The OSS project is under `BSD`. See https://w1.fi/cgit/hostap/tree/COPYING for details (it migrated from BSD/GPL dual license to BSD only in 2012).
All customizations for Zephyr are under Apache-2.0.

Note: MbedTLS port is taken from ESP-IDF and is also under Apache-2.0, copyright is updated.

@jukkar  @sachinthegreen @carlescufi

rftafas, 2023-09-21:
FWIW: https://github.com/zephyrproject-rtos/zephyr/issues/34507

krish2718, 2023-09-21:
@rftafas thanks, it was an old proposal from the same team :), so, this proposal supersedes that.

krish2718, 2023-09-22:
> BTW, will there be any specification on how to write drivers compatible with this? I'm thinking something like 'mac80211'.

We have plans to documents the wpa_supplicant to driver interface, so, that should help develop drivers that use wpa_supplicant. It would be analogous to `cfg80211` in the Linux world.

dleach02, 2024-01-25:
this was approved and the module now exists

closed 2024-01-25 by dleach02