zephyrproject-rtos/zephyr #62917: GENMASK and FIELD_PREP macro can't get correct bits value of register

snowuyl, 2023-09-21:
**Describe the bug**
I use GENMASK and FIELD_PREP macro to get bits value of register. But it failed to get correct bits value.
#define GENMASK(h, l) (((~0UL) - (1UL << (l)) + 1) & (~0UL >> (BITS_PER_LONG - 1 - (h))))
#define LSB_GET(value) ((value) & -(value))
#define FIELD_PREP(mask, value) (((value) * LSB_GET(mask)) & (mask))
#define LP5861_DEV_INITIAL_RESERVED_MSK      GENMASK(7, 3)
#define LP5861_DEV_INITIAL_RESERVED(x)       FIELD_PREP(LP5861_DEV_INITIAL_RESERVED_MSK, x)
#define LP5861_DEV_INITIAL_DATA_REF_MODE_MSK GENMASK(2, 1)
#define LP5861_DEV_INITIAL_DATA_REF_MODE(x)  FIELD_PREP(LP5861_DEV_INITIAL_DATA_REF_MODE_MSK, x)
#define LP5861_DEV_INITIAL_PWM_FRE_MSK       BIT(0)
#define LP5861_DEV_INITIAL_PWM_FRE(x)        FIELD_PREP(LP5861_DEV_INITIAL_PWM_FRE_MSK, x)
const uint8_t u8=0x5e;
devInit.reserved    = LP5861_DEV_INITIAL_RESERVED(u8);
devInit.dataRefMode = LP5861_DEV_INITIAL_DATA_REF_MODE(u8);
devInit.pwmFre      = LP5861_DEV_INITIAL_PWM_FRE(u8);

**Logs and console output**
u8=0x5e
devInit.reserved=0xF0, devInit.dataRefMode=0x4, devInit.pwmFre=0x0

**Environment (please complete the following information):**
 - OS: Windows 11
 - Toolchain Zephyr SDK v3.3.99

jhedberg, 2023-09-26:
These macros have been in Zephyr for about five years and there are 756 users of `GENMASK()` alone, so it's quite surprising that they would be broken.

snowuyl, 2023-09-26:
Thanks for your reply! Does Zephyr provide sample code for me to test?

gramsay0, 2023-09-28:
@snowuyl I think you want `FIELD_GET()` to **get** the bits rather than `FIELD_PREP()` in this case

snowuyl, 2023-09-29:
Thanks for your reply! I will try it.

nashif, 2023-10-19:
no response after last comment, assuming the suggestion works and this is not a bug.