zephyrproject-rtos/zephyr #62918: net: l2: ieee802154: (slightly) simplify CSL API and generalize "CSL RX time" to "expected RX time"

fgrandel, 2023-09-21:
**Problem Description**

As has been pointed out in #61227, the current CSL configuration approach could be better aligned with standard primitives. This enhancement proposal outlines a way how this could be done. This is a prerequisite for HW offloading support for TSCH time synchronization (as well as offloading support for other standard timing sensitive protocols, notably beacon enabled PANs/DSME).

**Solution**

The following source code snippet documents the proposed API change to `IEEE802154_CONFIG_CSL_PERIOD` and `IEEE802154_CONFIG_CSL_RX_TIME`:

```
	/** Enables or disables a device as a CSL receiver and configures its
	 *  CSL period.
	 *
	 *  @details Configures the CSL period in units of 10 symbol periods.
	 *  Values greater than zero enable CSL if the driver supports it and
	 *  the device starts to operate as a CSL receiver. Setting this to
	 *  zero disables CSL on the device. If the drive does not support CSL,
	 *  the configuration call SHALL return -ENOTSUP.
	 *
	 *  See section 7.4.2.3 and section 8.4.3.6, table 8-104, macCslPeriod.
	 *
	 *  @note Confusingly the standard calls the CSL receiver "CSL
	 *  coordinator" (i.e. "coordinating the CSL protocol timing", see
	 *  section 6.12.2.2), although, typically, a CSL coordinator is NOT
	 *  also an IEEE 802.15.4 FFD coordinator or PAN coordintor but a simple
	 *  RFD end device (compare the device types outlined in sections 5.1,
	 *  5.3, 5.5 and 6.1). To avoid this kind of confusion we prefer to call
	 *  CSL coordinators (often an RFD end device) "CSL receivers" and the
	 *  CSL peer device (often a FFD coordinator or PAN coordinator) the
	 *  "CSL transmitter". Also note that at this time, we do NOT support
	 *  unsynchronized transmission with CSL wake up frames as specified in
	 *  section 6.12.2.4.4.
	 *
	 *  In order to offload CSL receiver timing to the driver the upper
	 *  layer SHALL combine several configuration options in the following
	 *  way:
	 *
	 *    1. Use @ref IEEE802154_CONFIG_ENH_ACK_HEADER_IE once with an
	 *       appropriate pre-filled CSL IE that leaves the CSL phase zero.
	 *       The phase SHALL be injected on-the-fly by the driver at runtime
	 *       as outlined in 2. below. Adding a short and extended address
	 *       will inform the driver of the short and extended addresses of
	 *       the CSL receiver to which it should inject CSL IEs. This SHALL
	 *       be done before enabling CSL by setting a CSL period greater
	 *       than zero.
	 *
	 *    2. Configure @ref IEEE802154_CONFIG_EXPECTED_RX_TIME immediately
	 *       followed by @ref IEEE802154_CONFIG_CSL_PERIOD. Upper layers
	 *       SHALL ensure that the receiver is not enabled between the two
	 *       calls (e.g. by a previously configured RX slot) nor shall it
	 *       transmit a frame in between to prevent race conditions.
	 *
	 *       The expected RX time SHALL point to the end of SFD of an
	 *       ideally timed RX frame in the nearest CSL channel sample (i.e.
	 *       whose "start of SHR" lies exactly at the center of the CSL
	 *       channel sample).
	 *
	 *       Drivers SHALL derive CSL phase of any frame containing a CSL IE
	 *       as follows:
	 *
	 *          cslAnchorPointNs = last expected RX time
	 *                             + PHY-specific PHR duration in ns
	 *
	 *          startOfMhrNs = start of MHR of the frame containing the
	 *                         CSL IE relative to the local network clock
	 *
	 *          cslPhase = (startOfMhrNs - cslAnchorPointNs)
	 *                     / (10 * PHY specific symbol period in ns)
	 *                     % cslPeriod
	 *
	 *       The driver SHALL set the CSL Phase in the IE configured in 1.
[... 88 lines cut ...]
	 *    be calculated inside the driver, see IEEE 802.15.4z.
	 *
	 *  Additionally this parameter MAY be used by drivers to discipline
	 *  their local representation of a distributed network clock by
	 *  deriving synchronization instants related to a remote representation
	 *  of the same clock.
	 *
	 *  @note Available in any interface operational state.
	 */
	IEEE802154_CONFIG_EXPECTED_RX_TIME,
```

**Alternatives Considered**

I first considered replacing CSL RX time by an additional "expected RX time" parameter in the RX slot configuration:
 * This would have required us to re-calculate the next CSL channel sample center for each RX slot (which may be considered a feature as it moves that calculation from the driver to L2 - but as the CSL phase calculation would still be the driver's responsibility, such an advantage is minor IMO).
 * More importantly: This would have reduced the generality of such a parameter to timed RX while it may also be useful in case the receiver is permanently on.

I also considered keeping CSL RX time and adding expected RX time independently:
 * That turned out to generate a lot of ambivalence and overlap in the API definition.
 * The existing CSL RX time parameter is not a standard parameter therefore replacing it with something more generally useful and tied to the existing RMARKER/end of SFD definition used on the TX side seemed more intuitive

**Additional context**

Setting an expected RX time is required to support offloading of time synchronization to the driver in the context of TSCH protocol (see #50336) or beacon enabled PANs - maybe even future ranging use cases. Therefore some standard based common ground between CSL and TSCH should be found to keep the driver implementation as simple as possible in both cases.

fgrandel, 2023-09-21:
@edmont @jciupis @rlubos That's the solution proposed - related to the Discord group I opened to discuss the preliminaries with you. Feel free to comment/discuss here or on Discord as appropriate.

hubertmis, 2023-09-22:
I like generic `IEEE802154_CONFIG_EXPECTED_RX_TIME` replacing the CSL-specific one a lot!

I'm thinking if we could make the CSL interface more scalable. The proposed one seems to perfectly fit devices running a single CSL connection (exactli like the one being replaced), but maybe we could somehow slithtly change it to support multiple CSL connections in parallel.

I didn't think this through, so if that would complicate API or implementations a lot, we can consider an API change needed for this later, when we have actual use-case.

fgrandel, 2023-09-22:
> I'm thinking if we could make the CSL interface more scalable.

@hubertmis Thanks for your review! :-) I'd love to understand that comment better. Will write to you over Discord and then document the results here.

fgrandel, 2023-09-25:
> > I'm thinking if we could make the CSL interface more scalable.
> 
> @hubertmis Thanks for your review! :-) I'd love to understand that comment better. Will write to you over Discord and then document the results here.

We quickly discussed this over Discord and the result is:
* @hubertmis request was to consider having several cslPhase/Period pairs per receiver, such that different transmitters could be served with different periods (i.e. some basic "TDMA" emulation very similar to what TSCH actually does natively through slotframes and links).
* We agreed that this is probably not in line with IEEE 802.15.4's CSL protocol description, as they don't define dedicated pairs of cslPeriod/cslPhase per transmitter counterpart (some kind of a "neighbor table" as in TSCH). Although this would have to be confirmed in detail first.
* This, however, doesn't mean that such a requirement could not be implemented if it should at any time be specified by the Thread standard. That's what we have our Thread-specific architecture block and namespace for. Anything can be done in there if it turns out to be required for (Open)Thread.

PR #63252 (2023-10-20): drivers: ieee802154: improve CSL standard conformance
*NOT meant for Zephyr v3.5. A section will be added to the release notes for 3.6 once those are out.*

~~DNM: Requires https://github.com/zephyrproject-rtos/hal_nordic/pull/142 to be merged first.~~

UPDATE: Required conversion function has been merged and this PR was rebased on top of it. See https://github.com/zephyrproject-rtos/hal_nordic/pull/145.

This change slightly simplifies the configuration of a CSL receiver and generalized CSL_RX_TIME to EXPECTED_RX_TIME as a re-usable primitive across several timing-sensitive IEEE 802.15.4 stan [... cut]
commit e9d5a98e (2023-10-20): drivers: ieee802154: improve CSL standard conformance

This change slightly simplifies the configuration of a CSL receiver and
generalized CSL_RX_TIME to EXPECTED_RX_TIME as a re-usable primitive
across several timing-sensitive IEEE 802.15.4 standard sub-protocols
(namely BE-PANs/DSME/CSL/RIT/TSCH).

This API change is based on the rules outlined in RFC #61227.

Fixes: #62918
closed 2023-10-20 by carlescufi