zephyrproject-rtos/zephyr #62926: Twister: no paths in tests' names

PerMac, 2023-09-21:
## Introduction

Make twister to not use paths in tests' names.

### Problem description

Twister creates names of tests as <RELATIVE_PATH_TO_ZEPHYR>/<TEST_NAME_FROM_YAML>. E.g. `samples/hello_world/sample.basic.helloworld`. These names are treated as identifiers by twister. It means options such as `-s` which allow running a single test also have to match such names. However, when a test is located outside of the zephyr tree the path is not resolved uniformly, i.e. the name (hence identifier) will vary, depending on the place where twister will start scanning for tests. This effect is especially severe when one tries to use a test_plan.py script: test names in the script's output won't match the names from a twister call, and twister will fail due to not finding matching tests.

Moreover, if a test moves to another location, it is still the same test. With the path in the name, it will be treated as a different test.

### Proposed change

Twister will no longer add paths to test names. Make the transition smooth giving time for the community to adapt.

## Detailed RFC

I am adding new flags: `--no-path-name` and `path-name` in #62332. They both set the same value, to either False or True.  When False, twister skips the part where a path is added to the name. 
At first, the value is set to True by default. I.e. when nothing is added to a twister command, everything works as before (names are with paths). The community will be advised to try using "--no-path-name". This should prevent a backlash with users complaining about their ways of calling twister/analyzing reports suddenly not working. The community will have some time to try this option out and align their CIs accordingly. 

After a "deprecation" phase, the variable will be set to True by default, however, one will be able to use "--path-name" to maintain the original behavior. 

### Proposed change (Detailed)

- Inform the community (during testing wg meetings, on discord's #twister, #testing and testing wg mailing list) about the idea of not including path by default in tests' names. Inform that an incoming behavior can be tested by adding `--no-path-name`. Ask to verify their workflows/CIs with this flag. 
- Add this flag to twister calls in upstream's CIs. 
- Inform about the new flags and the upcoming change in changelog for Zephyr 3.5
- At the beginning of Zephyr 3.5 add a warning printed at the start of twister when the flag is not used: "Please consider adding `--no-path-name` to your twister call. This option will be the default one from Zephyr 3.6" 
- In Zephyr 3.6 make the option default. Remove (now redundant) `--no-path-name` from CLI calls.

### Dependencies

This change can affect workflows (CIs) related to executing specific tests and tracking their results (if paths were not stripped already). With the change in names, test plans will also change and the corresponding modifications have to be included in such scrips (e.g. test_plan.py)

### Concerns and Unresolved Questions

We will have to define how `-s` (choose a specific test) should work to allow more flexibility and seamless experience. However, it should be quite simple 

## Alternatives

Leave things as they are and try patching/working around the problems described above.

PerMac, 2023-09-21:
cc @hakehuang @golowanow @SzymonRichert @gopiotr @gchwier @yperess @erwango @evgeniy-paltsev

nashif, 2024-01-26:
This has been addressed AFAIK

closed 2024-01-26 by nashif

PerMac, 2024-01-29:
The RFC contain a proposal for deprecation of paths in names and making non-path the default. But I got no reply in this RFC. I don't mind getting this closed, since I added --no-detailed-test-id as a flag in whichever CI I needed it, I though @nashif you were for making the no-path the default.