zephyrproject-rtos/zephyr #62940: drivers: ieee802154: improved driver support for enhanced ACK IE offloading

fgrandel, 2023-09-21:
**Problem Description**

As has been pointed out in #61227, configuring IE offloading for enhanced ACK IEs could be improved to make it easier for drivers that do not want to implement their own IE parser/writer internally by introducing improved types and appropriate helpers to be re-used across all drivers (optionally including nRF5).

In #61227 this was referred to as "moving soft MAC parts out of proprietary drivers into cross-vendor L2 callbacks".

**Proposed Solution**

In the context of #50336 (TSCH roadmap) many types and IE parser/writer helpers need to be introduced into L2 (soft MAC). Some of these are immediately relevant to #61227 and make it easier for other drivers to support CSL IE offloading. More specifically the existing implementation could be moved forward immediately by providing more structured IE types that help driver maintainers to construct CSL IEs with little effort. Later on in the TSCH roadmap, additional writers and parsers will be introduced that will make it even easier to support IE offloading in any driver.

A PR with the proposed immediate changes will follow.

**Possible Alternatives**

nRF5's IE parser/writer could be ported out of the driver internals and placed into a cross-vendor layer shared by OT and native L2. The nRF5 IE parser/write is however less generic and complete than the one required for TSCH so it probably makes more sense to either keep it in parallel or successively replace it as additional re-usable helpers are introduced into Zephyr. These helpers already exist in the [TSCH development branch](https://github.com/fgrandel/zephyr/tree/feat/ieee802154-tsch-ie-support) and can be previewed there, see ieee802154_frame_ie.* there.

**Additional Context**

The changes proposed in this enhancement are a precondition to support hardware offloading for ACK IE injection in TSCH. Additionally, the TSCH roadmap not only introduces IEs specific to TSCH but will immediately provide types also that help to offload other IEEE 802.15.4 timing sensitive protocols that involve IE signaling in enhanced ACKs (one of which is CSL).

fgrandel, 2023-09-21:
@edmont @jciupis @rlubos Another enhancement proposal, this time for improved (more flexible) enhanced ACK IE offloading. This is a precondition for TSCH support, too, but is more generally useful than that. PR follows as soon as the two blocking PRs #62413 and #62503. You can preview the proposed change here: https://github.com/fgrandel/zephyr/commit/e70933b9914687a437dcf90f88b90f885cb6f5c7

Prasanna1717, 2023-09-21:
Hello sir,
From my understanding, the problem is about the `CONFIG_GPIO_SCH311X` configuration not being enabled for the `nrf_sch311x` board. To resolve this issue, the GPIO driver for the SCH311X chip needs to be enabled for the `nRF Connect SCH311x` board.

To fix this issue, we need to enable the CONFIG_GPIO_SCH311X configuration option for the nrf_sch311x board. This can be done by editing the board's **Kconfig. board** file. I guess this code snippet which help you enable it.

     config BOARD_NRF_SCH311X
     bool "nRF Connect SCH311x"
     source "drivers/gpio/Kconfig.gpio"
     config GPIO_SCH311X
     bool "GPIO Driver for SCH311X"
     select GPIO
     help
     Enable the GPIO driver for SCH311X

If I'm correct I can make a PR regarding this after some testing is done Locally. If I misunderstood the problem, You can let me know, I try my best to work on it.
Thank You

Regards 
Prasanna

PR #63251 (2023-10-20): drivers: ieee802154: improve ACK header IE config
*NOT meant for Zephyr v3.5. A section will be added to the release notes for 3.6 once those are out.*

Improves standard conformance of the IEEE802154_CONFIG_ENH_ACK_HEADER_IE option and introduces certain "soft MAC" capabilities around header IEs:
 * Introduces types and helpers that allow driver maintainers to represent, parse, write and validate header IEs.
 * Introduces helper functions to access non-aligned fields in header IEs, namely element IDs.

Updates the only existing L2 and driver pair that uses IEEE802154_CONFIG_ENH_ACK_HEADER_IE: [... cut]
commit 80da9ddf (2023-10-20): drivers: ieee802154: improve ACK header IE config

Improves standard conformance of the IEEE802154_CONFIG_ENH_ACK_HEADER_IE
option and introduces certain "soft MAC" capabilities around header IEs:
 * Introduces types and helpers that allow driver maintainers to
   represent, parse, write and validate header IEs.
 * Introduces helper functions to access non-aligned fields in header
   IEs, namely element IDs.

Updates the only existing L2 and driver pair that uses
IEEE802154_CONFIG_ENH_ACK_HEADER_IE: OpenThread platform radio and nRF5
and improves header IE validation in the nRF5 driver.

This  [... cut]
closed 2023-10-20 by carlescufi