zephyrproject-rtos/zephyr #62953: Treat the safely temporary node as the prev node and call remove

LuoZhongYao, 2023-09-22:
https://github.com/zephyrproject-rtos/zephyr/blob/8951bab7e741f3750532d4af68497b9af06a453c/subsys/bluetooth/host/gatt.c#L5157

henrikbrixandersen, 2023-09-22:
Please elaborate.

LuoZhongYao, 2023-09-23:
The third parameter of `SYS_SLIST_FOR_EACH_NODE_SAFE` is `next`, and `sys_slist_remove` requires `prev`, so `list` is broken here

```c
		SYS_SLIST_FOR_EACH_NODE_SAFE(&sub->list, node, tmp) {
			if (node == &params->node) {
				gatt_sub_remove(conn, sub, tmp, params);
				break;
			}
		}
```
```c
static void gatt_sub_remove(struct bt_conn *conn, struct gatt_sub *sub,
			    sys_snode_t *prev,
			    struct bt_gatt_subscribe_params *params)
{
	if (params) {
		/* Remove subscription from the list*/
		sys_slist_remove(&sub->list, prev, &params->node);
		/* Notify removal */
		params->notify(conn, params, NULL, 0);
	}

	if (gatt_sub_is_empty(sub)) {
		gatt_sub_free(sub);
	}
}
```

henrikbrixandersen, 2023-09-23:
I see you are reporting a bug? Please use our bug template when doing so. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

closed 2023-09-27 by carlescufi