zephyrproject-rtos/zephyr #62963: lpcxpresso55s69: Support for the flash hardware is broken

butok, 2023-09-22:
**Describe the bug**
If to enable the FLASH support for lpcxpresso55s69, an application is stuck and never reach its main() function.

**To Reproduce**
Steps to reproduce the behavior:
1. Use zephyr\samples\basic\blinky sample example.
2. Enable the FLASH hardware support in the application, by adding CONFIG_FLASH=y to the project configuration. 
5. Build and run it for lpcxpresso55s69_cpu0.
6. The application is stuck and never reach the main() function.

If to delete CONFIG_FLASH=y, the application is functional as expected.

**Expected behavior**
Enabling of FLASH should not break the application functionality.

**Impact**
Any application, that requires the FLASH support, is not functional.

**Logs and console output**
N/A

**Environment**

OS: Windows
Toolchain: Zephyr SDK 0.16.1
Commit: https://github.com/zephyrproject-rtos/zephyr/commit/4129b10b6900133e41829893c36a01527f293548 Date: 9/12/2023

**Additional context**
N/A

danieldegrasse, 2023-10-03:
I've bisected this issue down to c6e3bac4f20a659. It appears the LPC55S69 requires the CPU core clock be less than 100MHz when flash operations are being performed:
[image]
(section 5.3 of the LPC55S69 reference manual)

butok, 2023-10-04:
The #62963 is flash initialization issue. No write/erase is done yet.

danieldegrasse, 2023-10-04:
@butok This is a good point- however, I see flash API tests able to pass prior to c6e3bac. I wonder if the issue is that we call the `flash_init` ROM API during flash init? Perhaps the ROM code there runs into issues if the clock is over 100MHz. The SDK `FLASH_Init` function seems to suggest 96MHz is expected: https://github.com/zephyrproject-rtos/hal_nxp/blob/master/mcux/mcux-sdk/drivers/iap1/fsl_iap.c#L237

butok, 2023-10-05:
Hi Daniel (@danieldegrasse )

>  I see flash API tests able to pass
What test application are you using?

>The SDK FLASH_Init function seems to suggest 96MHz
It is true. Checked on practice for LPC55S69, it should be <100 for a flash write.
But for reading, it had no issue.

Now, I am working on LPC55S36 & FLASH & MCUboot. I have  a working prototype, I need to prepare/clean code for upstreaming and to add final improvements.
When it is finished (during next week), I will switch to this LPC55S69/16/06 issue.

BTW: 55S69/16/06 and 55S36 are too different. 
Do we want to split them to separate source files to have cleaner code and to avoid repetitive #ifdef ?

danieldegrasse, 2023-10-05:
> What test application are you using?

`tests/drivers/flash/common`

> BTW: 55S69/16/06 and 55S36 are too different.
Do we want to split them to separate source files to have cleaner code and to avoid repetitive #ifdef ?

Yes, we should probably split the SOC initialization for the LPC5536 into another file, if that is what you mean.

closed 2023-10-08 by fabiobaltieri

butok, 2023-10-10:
Tested on my side, on LPC55S16/69. 
Confirm, it is fixed.

PR #63486 (2023-10-08): soc: arm: nxp_lpc: Only clock core from PLL1 when CONFIG_FLASH=n
Do not clock the LPC55xxx cores from PLL1 when CONFIG_FLASH is set. This is required due to the following limitation of the flash controller (documented in the reference manual):

Flash operations (erase, blank check, program) and reading a single word can only be performed for CPU frequencies of up to 100 MHz. These operations cannot be performed for frequencies above 100 MHz.

The PLL1 clock source will result in a core clock of 150MHz, which violates this requirement.

Fixes #62963
commit faf55932 (2023-10-08): soc: arm: nxp_lpc: Only clock core from PLL1 when CONFIG_FLASH=n

Do not clock the LPC55xxx cores from PLL1 when CONFIG_FLASH is set. This
is required due to the following limitation of the flash controller
(documented in the reference manual):

Flash operations (erase, blank check, program) and reading a single word
can only be performed for CPU frequencies of up to 100 MHz. These
operations cannot be performed for frequencies above 100 MHz.

The PLL1 clock source will result in a core clock of 150MHz, which
violates this requirement.

Fixes #62963