zephyrproject-rtos/zephyr #63011: Snippets as extra arguments on Twister CLI leads to early CMake error

rexut, 2023-09-23:
**Describe the bug**

Whenever you call Twister with the command line argument `--extra-args SNIPPET="<valid-snippet-name>"`, CMake aborts with the error message `zephyr_set(... SCOPE snippets) doesn't exist.`. This happens very early on in Zephyr bootstrapping, more specifically when including `cmake/modules/snippets.cmake`.

The analogous usage for a simple build, on the other hand, does not result in any error message, so both `west build -S <valid-snippet-name>` and `west build -- -DSNIPPET="<valid-snippet-name>"` work as described and documented. So far, I myself was not able to find the cause of the non-existent scope for snippets in the case of Twister and need help from experts.

**To Reproduce**

Here is a more fancy console log:

```
west twister --verbose --inline-logs --cmake-only \
     --extra-args SNIPPET="cdc-acm-console" \
     --platform nucleo_f767zi \
     --test tests/drivers/uart/uart_async_api/drivers.uart.async_api
```

…breaks with (CMake output in `build.log`):

```
-- Application: /.../workspace-zephyr/zephyr/tests/drivers/uart/uart_async_api
-- CMake version: 3.27.4
-- Found Python3: /.../workspace-zephyr/.env/bin/python (found suitable version "3.8.10", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.4.99 (/.../workspace-zephyr/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: nucleo_f767zi
CMake Error at zephyr/cmake/modules/extensions.cmake:2764 (message):
  zephyr_set(...  SCOPE snippets) doesn't exists.
Call Stack (most recent call first):
  twister-out/nucleo_f767zi/tests/drivers/uart/uart_async_api/drivers.uart.async_api/zephyr/snippets_generated.cmake:25 (zephyr_set)
  zephyr/cmake/modules/snippets.cmake:94 (include)
  zephyr/cmake/modules/snippets.cmake:121 (zephyr_process_snippets)
  zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  zephyr/cmake/package_helper.cmake:115 (find_package)
```

**Expected behavior**

The expectation here would be that Twister behaves the same way with snippets as it does when passing a shield stack via extra arguments. Both `--extra-args SNIPPET="<valid-snippet-name>"` and `--extra-args SHIELD="<valid-shield-name>"` should not lead to any CMake errors, alone or together.

**Impact**

Announcement for the community, but showstopper for [Bridle](https://github.com/tiacsys/bridle). Currently I'm trying to convert a [board-specific shield with CPU clock tweaks](https://github.com/tiacsys/bridle/tree/main/boards/shields/can_timing_adj) to a snippet so that the default [Zephyr CAN timing test](https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/drivers/can/timing) passes.

**Logs and console output**
See above.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.1
 - main branch, commit SHA 341590545ac80b7a2abd9902b7f10d577323ccde

**Additional context**
Non.

rexut, 2023-09-28:
Last few days I was playing with the new test case specification `required_snippets` in our downstream project with similar effect. On downstream we have the `bridle/tests/drivers/can/timing/testcase.yaml` looks like:
```
tests:
  drivers.can.timing:
    tags:
      - drivers
      - can
    depends_on: can
    filter: dt_chosen_enabled("zephyr,canbus")
    required_snippets:
      - can-timing-adj
```
Even the snippet `can-timing-adj` provides a downstream board specific overlay, details in: https://github.com/tiacsys/bridle/tree/devel/snippets/can-timing-adj/snippets/can-timing-adj

With `west twister --cmake-only --alt-config-root bridle/tests/drivers/can/timing --testsuite-root zephyr/tests/drivers/can/timing` Twister exits early with following error message:
```
ERROR   - Can't find snippet 'can-timing-adj' for test 'tests/drivers/can/timing/drivers.can.timing'
```

That's a result of the Twister test plan logic in [apply_filters() about line 839](https://github.com/zephyrproject-rtos/zephyr/blob/main/scripts/pylib/twister/twisterlib/testplan.py#L839):
```
if ts.required_snippets:
    missing_snippet = False
    snippet_args = {"snippets": ts.required_snippets}
    found_snippets = snippets.find_snippets_in_roots(snippet_args, [Path(ZEPHYR_BASE), Path(ts.source_dir)])
```

`found_snippets` expands here to nothing, because it looks only into the snippets and test spec folder on Zephyr upstream, never on any (well known) downstream repositories, although these are declared in `bridle/zephyr/module.yml`.

nordicjm, 2023-09-28:
Give https://github.com/zephyrproject-rtos/zephyr/pull/63232 a try

rexut, 2023-09-28:
> Give #63232 a try

Wow, quick response. Many thanks. Looks like right direction. Twister now finds downstream snippets in modules, too. BUT: now we run into exactly the same error as descript on top of this bug report. Let me give you the sample session.

Twister called by west:
```
west twister --cmake-only \
             --alt-config-root bridle/tests/drivers/can/timing \
             --testsuite-root zephyr/tests/drivers/can/timing
```

Exit now with:
```
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-4509-g5f7fee5b9139
INFO    - Using 'zephyr' toolchain.
INFO    - Using alternative configuration from /.../workspace/bridle/tests/drivers/can/timing/testcase.yaml
INFO    - Selecting default platforms per test case
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /.../workspace/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - Cmake build failure: /.../workspace/zephyr/tests/drivers/can/timing for tiac_magpie
ERROR   - tiac_magpie               tests/drivers/can/timing/drivers.can.timing         ERROR : Cmake build failure
ERROR   - see: /.../workspace/twister-out/tiac_magpie/tests/drivers/can/timing/drivers.can.timing/build.log
… … …
```

Build log is (`.../tests/drivers/can/timing/drivers.can.timing/build.log`):
```
Loading Zephyr module(s) (Zephyr base): zephyr_default:dts
-- Application: /.../workspace/zephyr/tests/drivers/can/timing
-- CMake version: 3.27.4
Loading Bridle default modules (Freestanding).
-- Bridle version: 3.4.99 (/.../workspace/bridle)
-- ZEPHYR_TOOLCHAIN_VARIANT: zephyr
-- SDK_VERSION: 0.16.1
-- ZEPHYR_SDK_INSTALL_DIR  : /opt/zephyr-sdk-0.16.1
-- Found Python3: /.../workspace/.env/bin/python (found suitable version "3.8.10", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/.../.cache/zephyr
-- Zephyr version: 3.4.99 (/.../workspace/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: tiac_magpie
CMake Error at zephyr/cmake/modules/extensions.cmake:2764 (message):
  zephyr_set(...  SCOPE snippets) doesn't exists.
Call Stack (most recent call first):
  twister-out/tiac_magpie/tests/drivers/can/timing/drivers.can.timing/zephyr/snippets_generated.cmake:27 (zephyr_set)
  zephyr/cmake/modules/snippets.cmake:94 (include)
  zephyr/cmake/modules/snippets.cmake:121 (zephyr_process_snippets)
  zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  zephyr/cmake/package_helper.cmake:115 (find_package)
```

–> **zephyr_set(...  SCOPE snippets) doesn't exists.**

rexut, 2023-09-28:
Related to https://github.com/zephyrproject-rtos/zephyr/pull/63232#issuecomment-1739387093

`cat twister-out/tiac_magpie/tests/drivers/can/timing/drivers.can.timing/build.log`:
```
… … …
-- Board: tiac_magpie
CMake Error at zephyr/cmake/modules/extensions.cmake:2764 (message):
  zephyr_set(...  SCOPE snippets) doesn't exists.
Call Stack (most recent call first):
  twister-out/tiac_magpie/tests/drivers/can/timing/drivers.can.timing/zephyr/snippets_generated.cmake:27 (zephyr_set)
… … …
```

`cat twister-out/tiac_magpie/tests/drivers/can/timing/drivers.can.timing/zephyr/snippets_generated.cmake`:
```
# WARNING. THIS FILE IS AUTO-GENERATED. DO NOT MODIFY!
#
# This file contains build system settings derived from your snippets.
# Its contents are an implementation detail that should not be used outside
# of Zephyr's snippets CMake module.
#
# See the Snippets guide in the Zephyr documentation for more information.

###############################################################################
# Global information about all snippets.

# The name of every snippet that was discovered.
set(SNIPPET_NAMES "can-timing-adj" "cdc-acm-console" "usb-console" "xen_dom0")
# The paths to all the snippet.yml files. One snippet
# can have multiple snippet.yml files.
set(SNIPPET_PATHS "/media/slz/lpndevel/devel/TIAC/workspace/bridle/snippets/can-timing-adj/snippet.yml" "/media/slz/lpndevel/devel/TIAC/workspace/bridle/snippets/usb-console/snippet.yml" "/media/slz/lpndevel/devel/TIAC/workspace/zephyr/snippets/cdc-acm-console/snippet.yml" "/media/slz/lpndevel/devel/TIAC/workspace/zephyr/snippets/xen_dom0/snippet.yml")

# Create variable scope for snippets build variables
zephyr_create_scope(snippets)

###############################################################################
# Snippet 'can-timing-adj'

# Common variable appends.
# Appends for board 'tiac_magpie'
if("${BOARD}" STREQUAL "tiac_magpie")
  zephyr_set(EXTRA_DTC_OVERLAY_FILE "/media/slz/lpndevel/devel/TIAC/workspace/bridle/snippets/can-timing-adj/boards/tiac_magpie.overlay" SCOPE snippets APPEND)
endif()
```

It seems `zephyr_create_scope(snippets)` has not the expected effect in case of `west twister …`?

rexut, 2023-09-28:
> It seems `zephyr_create_scope(snippets)` has not the expected effect in case of `west twister …`?

Cross check with `west build …` with and w/o PR #63232 
```
west build -b tiac_magpie -S can-timing-adj \
           -d build/tiac_magpie -p always \
           zephyr/tests/drivers/can/timing
west flash -d build/tiac_magpie
```
Build host output:
```
-- west build: making build dir /.../workspace/build/tiac_magpie pristine
-- west build: generating a build system
… … …
-- Board: tiac_magpie
-- Snippet(s): can-timing-adj
… … …
-- Found BOARD.dts: /.../workspace/bridle/boards/arm/tiac_magpie/tiac_magpie.dts
-- Found devicetree overlay: /.../workspace/bridle/snippets/can-timing-adj/boards/tiac_magpie.overlay
-- Generated zephyr.dts: /.../workspace/build/tiac_magpie/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /.../workspace/build/tiac_magpie/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /.../workspace/build/tiac_magpie/zephyr/dts.cmake
Parsing /.../workspace/zephyr/Kconfig
Loaded configuration '/.../workspace/bridle/boards/arm/tiac_magpie/tiac_magpie_defconfig'
Merged configuration '/.../workspace/zephyr/tests/drivers/can/timing/prj.conf'
Configuration saved to '/.../workspace/build/tiac_magpie/zephyr/.config'
Kconfig header saved to '/.../workspace/build/tiac_magpie/zephyr/include/generated/autoconf.h'
… … …
```
Target console output:
```
*** Booting Zephyr OS build zephyr-v3.4.0-4592-g2205092ae00e ***
Running TESTSUITE can_timing
… … …
===================================================================
START - test_timing
testing bitrate 20000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 = 2, prescaler = 150 OK, sample point error 0.0%
testing bitrate 50000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 = 2, prescaler = 60 OK, sample point error 0.0%
testing bitrate 125000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 = 2, prescaler = 24 OK, sample point error 0.0%
testing bitrate 250000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 = 2, prescaler = 12 OK, sample point error 0.0%
testing bitrate 500000, sample point 87.5% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 = 2, prescaler = 6 OK, sample point error 0.0%
testing bitrate 800000, sample point 80.0% (valid): prop_seg = 0, phase_seg1 = 15, phase_seg2 = 4, prescaler = 3 OK, sample point error 0.0%
testing bitrate 1000000, sample point 75.0% (valid): prop_seg = 0, phase_seg1 = 11, phase_seg2 = 4, prescaler = 3 OK, sample point error 0.0%
testing bitrate 125000, sample point 90.0% (valid): prop_seg = 0, phase_seg1 = 13, phase_seg2 = 2, prescaler = 24 OK, sample point error 2.5%
testing bitrate 125000, sample point 80.0% (valid): prop_seg = 0, phase_seg1 = 11, phase_seg2 = 4, prescaler = 24 OK, sample point error 5.0%
testing bitrate 125000, sample point 100.0% (invalid): OK
testing bitrate 1000001, sample point 75.0% (invalid): OK
 PASS - test_timing in 0.122 seconds
===================================================================
… … …
===================================================================
TESTSUITE can_timing succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [can_timing]: pass = 3, fail = 0, skip = 3, total = 6 duration = 0.127 seconds
 - SKIP - [can_timing.test_set_timing_data_max] duration = 0.001 seconds
 - SKIP - [can_timing.test_set_timing_data_min] duration = 0.001 seconds
 - PASS - [can_timing.test_set_timing_max] duration = 0.001 seconds
 - PASS - [can_timing.test_set_timing_min] duration = 0.001 seconds
 - PASS - [can_timing.test_timing] duration = 0.122 seconds
 - SKIP - [can_timing.test_timing_data] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

Result, the autogenerated file `snippets_generated.cmake` in build folder is exactly the same:
```
md5sum build/tiac_magpie/zephyr/snippets_generated.cmake \
       twister-out/tiac_magpie/tests/drivers/can/timing/drivers.can.timing/zephyr/snippets_generated.cmake
```

> ```
> 9c6a6a7cf63cc083a4722094853e4061  build/tiac_magpie/zephyr/snippets_generated.cmake
> 9c6a6a7cf63cc083a4722094853e4061  twister-out/tiac_magpie/tests/drivers/can/timing/drivers.can.timing/zephyr/snippets_generated.cmake
> ```

That leads me to the assertion that `zephyr_create_scope(snippets)` from `zephyr/cmake/modules/extensions.cmake` has side effects, if `west twister …` is the entry point.

rexut, 2023-09-28:
Here a last cross check that shows the unexpected different behaviour between `west build …` and `west twister …`. This check was done without our 3rd party downstream project – Zephyr main branch only (ddd2b490b006e36c2565ac5d53eda272a7a68488).

Also we do not try to use `--extra-args SNIPPET="<valid-snippet-name>"`, but the official recommended way with the test case spec `required_snippets` was used. Thus, to reproduce the CMake bug with original UART API test on board `nucleo_f767zi` we had add the following to lines locally:
```
diff --git a/tests/drivers/uart/uart_async_api/testcase.yaml b/tests/drivers/uart/uart_async_api/testcase.yaml
index e9914d48d9..8b26a7db84 100644
--- a/tests/drivers/uart/uart_async_api/testcase.yaml
+++ b/tests/drivers/uart/uart_async_api/testcase.yaml
@@ -13,6 +13,8 @@ tests:
     harness_config:
       fixture: gpio_loopback
     depends_on: gpio
+    required_snippets:
+      - cdc-acm-console
   drivers.uart.wide:
     filter: CONFIG_UART_CONSOLE and CONFIG_SERIAL_SUPPORT_ASYNC and not CONFIG_UART_MCUX_LPUART
     harness: ztest
```

To reproduce the CMake bug, it's enough to run with `--cmake-only` in both cases, `west build …` and `west twister …`.

### The good case `west build …`

```
west build --cmake-only -p always \
           -b nucleo_f767zi -S cdc-acm-console \
           zephyr/tests/drivers/uart/uart_async_api
```

results in:
```
-- west build: generating a build system
… … …
-- Board: nucleo_f767zi
-- Snippet(s): cdc-acm-console
… … …
-- Found devicetree overlay: /.../workspace/zephyr/snippets/cdc-acm-console/cdc-acm-console.overlay
… … …
Merged configuration '/.../workspace/zephyr/snippets/cdc-acm-console/cdc-acm-console.conf'
… … …
-- Build files have been written to: /.../workspace/build
```

**ALL FINE**

### The buggy path `west twister …`

Note, the patch (above) was applied:
```
west twister --verbose --inline-logs \
             --cmake-only --platform nucleo_f767zi \
             --test tests/drivers/uart/uart_async_api/drivers.uart.async_api
```

results in:
```
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-4625-gddd2b490b006
… … …
ERROR   - Cmake build failure: /.../workspace/zephyr/tests/drivers/uart/uart_async_api for nucleo_f767zi
INFO    - 1/1 nucleo_f767zi             tests/drivers/uart/uart_async_api/drivers.uart.async_api  ERROR Cmake build failure (cmake)
INFO    - /.../workspace/twister-out/nucleo_f767zi/tests/drivers/uart/uart_async_api/drivers.uart.async_api/build.log
ERROR   - Loading Zephyr module(s) (Zephyr base): zephyr_default:kconfig
… … …
-- Board: nucleo_f767zi
CMake Error at zephyr/cmake/modules/extensions.cmake:2764 (message):
  zephyr_set(...  SCOPE snippets) doesn't exists.
… … …
```

**NOT EXPECTED**

rexut, 2023-09-28:
All this is in serious contradiction to the test for `extra_args: SNIPPET="<valid-snippet-name>"` that shows no errors: `west twister --verbose --inline-logs --testsuite-root zephyr/tests/cmake/snippets`.

tejlmand, 2023-10-02:
@rexut could you try this fix: #63372

rexut, 2023-10-04:
> @rexut could you try this fix: #63372

Great, it fixes the origin of the issue:

```
west twister --verbose --inline-logs --cmake-only \
     --extra-args SNIPPET="cdc-acm-console" \
     --platform nucleo_f767zi \
     --test tests/drivers/uart/uart_async_api/drivers.uart.async_api
```

… works now as expected –> `zephyr,console = &snippet_cdc_acm_console_uart;`

Also, with a local patched `zephyr/tests/drivers/uart/uart_async_api/testcase.yaml`:

```
index e9914d48d9..8b26a7db84 100644
--- a/tests/drivers/uart/uart_async_api/testcase.yaml
+++ b/tests/drivers/uart/uart_async_api/testcase.yaml
@@ -13,6 +13,8 @@ tests:
     harness_config:
       fixture: gpio_loopback
     depends_on: gpio
+    required_snippets:
+      - cdc-acm-console
   drivers.uart.wide:
     filter: CONFIG_UART_CONSOLE and CONFIG_SERIAL_SUPPORT_ASYNC and not CONFIG_UART_MCUX_LPUART
     harness: ztest
```

… and simple call:

```
west twister --cmake-only \
     --platform nucleo_f767zi \
     --test tests/drivers/uart/uart_async_api/drivers.uart.async_api
```

–> `zephyr,console = &snippet_cdc_acm_console_uart;`

PR #63372 (2023-10-05): cmake: stub zephyr_set(... SCOPE ...) function in package helper
Fixes: #63011

Custom targets are not available in script mode, which cause snippet loading to fail when using package helper.

Lookup of snippets is supported by package helper, thus allowing twister to filter test cases based on snippets.

However, loading of snippets itself is not supported as snippets make use of Zephyr scoping, which uses custom targets, something that is not available in script mode.

Therefore overload the `zephyr_set()` function, so that CMake package helper can be used together with snippets.
commit c7779bec (2023-10-05): cmake: stub zephyr_set(... SCOPE ...) function in package helper

Fixes: #63011

Custom targets are not available in script mode, which cause snippet
loading to fail when using package helper.

Lookup of snippets is supported by package helper, thus allowing twister
to filter test cases based on snippets.

However, loading of snippets itself is not supported as snippets make
use of Zephyr scoping, which uses custom targets, something that is not
available in script mode.

Therefore overload the `zephyr_set()` function, so that CMake package
helper can be used together with snippets.
closed 2023-10-05 by fabiobaltieri