zephyrproject-rtos/zephyr #63017: drivers: can: mcan: RX filter synchronisation

gramsay0, 2023-09-24:
The MCAN driver does not have enough synchronisation around adding and removing CAN filters.

This could lead to bad situations like a callback function being called with an incorrect user data pointer.

**To Reproduce**

Here is a sample that _should_ be reproducible using MCAN in loopback mode.
500kbps bitrate worked well for me, but you may need to try a couple different bitrates (as we are trying to reproduce race conditions).

There are two threads, TX thread alternates between sending frame 1 and 2, the RX thread alternatives between adding/removing RX filter 1 and 2.
These threads are the same priority, neither sleep, they preempt each other through time-slicing.
Some error counts and stats are gathered and printed every 10 seconds.

The expected behavior is that:

- filter callback 1 should not receive CAN frames send to ID 2 (and vice-versa)
- filter callback functions are *never* called with a user data pointer that doesn't match what they configured

Kconfig:

    CONFIG_CAN=y
    CONFIG_TIMESLICE_SIZE=1

main:

    #include <zephyr/device.h>
    #include <zephyr/drivers/can.h>
    #include <zephyr/kernel.h>
    
    FUNC_NORETURN static void tx_thread_func(void *a1, void *a2, void *a3);
    FUNC_NORETURN static void rx_thread_func(void *a1, void *a2, void *a3);
    
    static const struct device *can_dev = DEVICE_DT_GET(DT_CHOSEN(zephyr_canbus));
    struct k_thread tx_thread;
    static K_THREAD_STACK_DEFINE(tx_thread_stack, 4096);
    static struct k_thread rx_thread;
    static K_THREAD_STACK_DEFINE(rx_thread_stack, 4096);
    
    // Stats
    uint32_t send_1_cnt = 0;
    uint32_t send_2_cnt = 0;
    uint32_t cb_missing_cnt = 0;
    uint32_t rx_callback_1_cnt = 0;
    uint32_t rx_callback_2_cnt = 0;
    uint32_t unexpected_user_data_1_cnt = 0;
    uint32_t unexpected_user_data_2_cnt = 0;
    uint32_t unexpected_frame_id_1_cnt = 0;
    uint32_t unexpected_frame_id_2_cnt = 0;
    
    int main(void)
    {
        can_set_mode(can_dev, CAN_MODE_LOOPBACK);
        can_start(can_dev);
    
        k_thread_create(
            &tx_thread, tx_thread_stack,
            K_THREAD_STACK_SIZEOF(tx_thread_stack), tx_thread_func,
            NULL, NULL, NULL, 5, K_ESSENTIAL, K_NO_WAIT);
    
        k_thread_create(
            &rx_thread, rx_thread_stack,
            K_THREAD_STACK_SIZEOF(rx_thread_stack), rx_thread_func,
            NULL, NULL, NULL, 5, K_ESSENTIAL, K_NO_WAIT);
    
        while (true) {
            k_sleep(K_SECONDS(10));
            printk("[%lld]\n"
                   "send_1_cnt = %u\n"
                   "send_2_cnt = %u\n"
                   "cb_missing_cnt = %u\n"
                   "rx_callback_1_cnt = %u\n"
                   "rx_callback_2_cnt = %u\n"
                   "unexpected_user_data_1_cnt = %u\n"
                   "unexpected_user_data_2_cnt = %u\n"
                   "unexpected_frame_id_1_cnt = %u\n"
                   "unexpected_frame_id_2_cnt = %u\n\n",
                   k_uptime_get(), send_1_cnt, send_2_cnt,
                   cb_missing_cnt, rx_callback_1_cnt, rx_callback_2_cnt,
                   unexpected_user_data_1_cnt, unexpected_user_data_2_cnt,
                   unexpected_frame_id_1_cnt, unexpected_frame_id_2_cnt);
        }
    }
    
    static void can_tx_callback(const struct device *dev, int error, void *user_data)
    {
    }
    
    static void can_rx_callback_1(const struct device *dev, struct can_frame *frame, void *user_data)
    {
        rx_callback_1_cnt++;
    
        if ((uintptr_t)user_data != 1) {
            unexpected_user_data_1_cnt++;
        }
        if (frame->id != 1) {
            unexpected_frame_id_1_cnt++;
        }
    }
    
    static void can_rx_callback_2(const struct device *dev, struct can_frame *frame, void *user_data)
    {
        rx_callback_2_cnt++;
    
        if ((uintptr_t)user_data != 2) {
            unexpected_user_data_2_cnt++;
        }
        if (frame->id != 2) {
            unexpected_frame_id_2_cnt++;
        }
    }
    
    FUNC_NORETURN static void tx_thread_func(void *a1, void *a2, void *a3)
    {
        const struct can_frame tx_frame_1 = {
            .flags = 0,
            .id = 1,
            .dlc = 1,
            .data = { 1 }
        };
        const struct can_frame tx_frame_2 = {
            .flags = 0,
            .id = 2,
            .dlc = 1,
            .data = { 2 }
        };
    
        while (true) {
            int res = can_send(can_dev, &tx_frame_1, K_NO_WAIT, can_tx_callback, NULL);
            if (res == 0) send_1_cnt++;
            res = can_send(can_dev, &tx_frame_2, K_NO_WAIT, can_tx_callback, NULL);
            if (res == 0) send_2_cnt++;
        }
    }
    
    FUNC_NORETURN static void rx_thread_func(void *a1, void *a2, void *a3)
    {
        const struct can_filter filter_1 = {
            .id = 1,
            .mask = 3,
            .flags = CAN_FILTER_DATA
        };
        const struct can_filter filter_2 = {
            .id = 2,
            .mask = 3,
            .flags = CAN_FILTER_DATA
        };
    
        while (true) {
            int filter_id_1 = can_add_rx_filter(can_dev, can_rx_callback_1, (void *)1, &filter_1);
            can_remove_rx_filter(can_dev, filter_id_1);
            int filter_id_2 = can_add_rx_filter(can_dev, can_rx_callback_2, (void *)2, &filter_2);
            can_remove_rx_filter(can_dev, filter_id_2);
        }
    }

Small patch to can_mcan.c for extra debug:

	diff --git a/drivers/can/can_mcan.c b/drivers/can/can_mcan.c
	index 103c80400d..f4446a1982 100644
	--- a/drivers/can/can_mcan.c
	+++ b/drivers/can/can_mcan.c
	@@ -755,6 +755,8 @@ static void can_mcan_get_message(const struct device *dev, uint16_t fifo_offset,
	                                cb(dev, &frame, user_data);
	                        } else {
	                                LOG_DBG("cb missing");
	+                               extern uint32_t cb_missing_cnt;
	+                               cb_missing_cnt++;
	                        }
	                } else {
	                        LOG_ERR("Frame is too big");

Example output:

	[10005]
	send_1_cnt = 41734
	send_2_cnt = 43044
	cb_missing_cnt = 21578
	rx_callback_1_cnt = 2601
	rx_callback_2_cnt= 1597
	unexpected_user_data_1_cnt = 291
	unexpected_user_data_2_cnt = 242
	unexpected_frame_id_1_cnt = 101
	unexpected_frame_id_2_cnt = 41

Possible fixes:

I tried reordering some of the calls and adding compiler barriers and cache flushes but the errors persisted.

I then tried adding a spinlock, this removed the worst error "unexpected_user_data", but some of the other errors are still there...

	[10005]
	send_1_cnt = 41842
	send_2_cnt = 42936
	cb_missing_cnt = 2152
	rx_callback_1_cnt = 13354
	rx_callback_2_cnt = 1998
	unexpected_user_data_1_cnt = 0
	unexpected_user_data_2_cnt = 0
	unexpected_frame_id_1_cnt = 4287
	unexpected_frame_id_2_cnt = 777

I'll have a look as to why this is happening when I get time, thought I'd share these notes for now

henrikbrixandersen, 2023-09-24:
@gramsay0 Can this be reproduced on any in-tree boards/SoCs?

gramsay0, 2023-09-24:
> @gramsay0 Can this be reproduced on any in-tree boards/SoCs?

It should be, although I do not have any handy to test on. I could probably find a CAN stm32 kit at work if needed.

It's a pretty simple test that just stresses that unlikely scenario

gramsay0, 2023-09-25:
I think this is why those "unexpected_frame_id" errors are still occurring:

MCAN user manual:

> Acceptance filtering is started after the complete identifier has been received. After acceptance
filtering has completed, and if a matching Rx Buffer or Rx FIFO has been found, the Message
Handler starts writing the received message data in portions of 32 bit to the matching Rx Buffer or
Rx FIFO.
...
Note: When an accepted message is written to one of the two Rx FIFOs, or into an Rx Buffer,
the unmodified received identifier is stored independent of the filter(s) used. The result
of the acceptance filter process is strongly depending on the sequence of configured
filter elements.

They key is that acceptance filtering is done after the CAN header / control bits are received, but before the full frame is received. If the filter matches, that filter could be removed and have a new one added in its place before the full frame is received and the RX ISR called.

I tested this sequence and got that error to occur reliably:

1. add RX filter that matches TX frame
2. send loopback TX frame
3. busy-wait for 34 CAN bits (header / control bits)
4. remove RX filter
5. add RX filter that does not match TX frame
6. incorrect filter always receives frame

Some possible fixes:

1. Ignore it, it's a pretty uncommon edge case
2. Double check the filter in software
3. Try to flush pending RX frames when adding/removing a filter
4. Do not immediately return filters to the free list
5. Select the oldest used filter when adding a new one

Either way, I think we should add a spinlock for the "user data" error.
That one seems more likely and more dangerous

closed 2024-02-25 by github-actions[bot]