zephyrproject-rtos/zephyr #63042: Improper build.log error checking in SizeCalculator

LukaszMrugala, 2023-09-25:
**Describe the bug**
Take notice of the code checking whether an objdump section is known or unknown to the size_calc if a buildlog file is provided:
https://github.com/zephyrproject-rtos/zephyr/blob/b6165e01caa42a5c2cc3e05744575e11ae142d4b/scripts/pylib/twister/twisterlib/size_calc.py#L260-L263

It seems obvious that the expected behaviour is that if the section name is in either of the three section name lists, we want to skip it and only take care of 'unknown ones. However, this `if` is written incorrectly. Current behaviour checks whether given section name is in `SizeCalculator.alloc_sections` and then whether the other two lists exist at all, which is true almost always. This makes us ignore all unknown sections if a buildlog file is provided.

**Expected behavior**
`name` is checked against all three lists.

**Impact**
We are currently ignoring all unknown sections in our size_calc.

**Logs and console output**
N/A

**Environment (please complete the following information):**
N/A

**Proposed fix**
```python
if name in SizeCalculator.alloc_sections or \ 
   name in SizeCalculator.rw_sections or \ 
   name in SizeCalculator.ro_sections: 
     continue 
 ```
```python
if any(name in section_names for section_names in [
           SizeCalculator.alloc_sections,
           SizeCalculator.rw_sections,
           SizeCalculator.ro_sections
]):
     continue 
 ```

nashif, 2023-09-25:
please submit a PR

LukaszMrugala, 2023-09-26:
> please submit a PR

Done in PR #63108.

PR #63108 (2023-09-26): scripts: pylib: twister: twisterlib: Fix size_calc error checking
Fixes #63042
commit 4b934551 (2023-09-26): scripts: pylib: twister: twisterlib: Fix size_calc error checking

Fixes #63042
closed 2023-09-26 by nashif