zephyrproject-rtos/zephyr #63082: The Enhancement Can be Made on the Cortex-A/R AArch32

SgrrZhf, 2023-09-26:
**Is your enhancement proposal related to a problem? Please describe.**
When I separated the Cortex-M and Cortex-A/R aarch32 architecture code, I found some improvements that can be made. So I list them here for me or other interested people.

**Describe the solution you'd like**

1. SMP support for Cortex-A/R
1. Refine the exception entry and exit code
1. Distinguish interrupt nested count and exception nested count.

**Additional context**

More details for the list above.

1. SMP support for Cortex-A/R
    - Implement _USE_SWITCH_ https://github.com/zephyrproject-rtos/zephyr/pull/61206
    - Implement _SMP_ https://github.com/zephyrproject-rtos/zephyr/pull/61206
    - Implement _USERSPACE_ with _USE_SWITCH_ and _SMP_
    - Implement _FPU_SHARING_ with _USE_SWITCH_ and _SMP_
    - Deprecate _arch_swap_

1. Refine the exception entry and exit code.
    Cortex-A/R arch32 has 5 exceptions. svc, fiq, irq, abt, und. The exception entry and exit code can be unified.
 
1. Distinguish interrupt nested count and exception nested count.
    For now, The Cortex-A/R code doesn't distinguish interrupt nested count and exception nested count. It will cause some issues with the context switch. For example, when IRQ occurred in an SVC handler routing, a context switch could happen after handling IRQ, but now it cannot occur because of nested count >= 1.

o7-machinehum, 2023-09-29:
Awesome! I would like to help with this, I'm currently working on a project using the RP2040, one of the only duel core M's around. I can give the PR a spin with my hardware if that would be helpful?

If there is any other way I can be of assistance I'm all ears.

SgrrZhf, 2023-10-08:
> Awesome! I would like to help with this, I'm currently working on a project using the RP2040, one of the only duel core M's around. If I give the PR a spin with my hardware would be helpful?

So, I guess you must be interested in this https://github.com/zephyrproject-rtos/zephyr/issues/59826 @o7-machinehum

microbuilder, 2023-12-12:
@SgrrZhf The linked issues in the description here have all been merged. Is there remaining work, or can this issue be closed?

SgrrZhf, 2023-12-12:
Yes, most of these have been finished. But there is something left, I marked them out in the description.