zephyrproject-rtos/zephyr #63115: zephyr/zephyr/cmake/modules/dts.cmake:366 [...]: No such file or directory

josuah, 2023-09-26:
**Describe the bug**
When running `west build`, the process stops with an error

**To Reproduce**
Steps to reproduce the behavior:
1. [NDA] (I will investigate this issue myself).

**Expected behavior**
The build runs to completion.

**Impact**
Showstopper (no zephyr.bin).

**Logs and console output**
```
$ python3 -m west build zephyr_clnx
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
[redacted]
-- Including generated dts.cmake file: /.../zephyr/build/zephyr/dts.cmake
CMake Error at /.../zephyr/zephyr/cmake/modules/dts.cmake:366 (message):
  command failed with return code: No such file or directory
Call Stack (most recent call first):
  /.../zephyr/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /.../zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /.../zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/.../.venv/bin/python3 -B/.../zephyr/build -GNinja -S/.../zephyr/zephyr_clnx
make: *** [Makefile:105: zephyr/build/zephyr/zephyr.bin] Error 1
```

**Environment (please complete the following information):**
 - OS: `Linux vm1 6.1.0-12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.52-1 (2023-09-07) x86_64 GNU/Linux`
 - Toolchain zephyr-sdk-0.16.3-rc1
 - Commit SHA: 7fca0aa8a6932bfd599552cd17be7b0d1d16b407

It also happened on this system, with the native toolchain (`riscv-none-elf-`):
`Linux lap1 6.1.46-0-lts #1-Alpine SMP PREEMPT_DYNAMIC Thu, 17 Aug 2023 14:11:55 +0000 x86_64 Linux`

**Additional context**
I only open this so that the error message shows-up in the issue search. :)

thedjnK, 2023-09-26:
I'll bet you anything in the part you've handily covered up with `[redacted]` is a message that reads something like "You have not provided a board using BOARD=x, this will likely fail"

josuah, 2023-09-26:
Sorry for involving your time, I planned to provide a message with the solution right away, but had a call immediately and did not have time to provide a proper open reproducer, which would have been allowing you to reproduce.

Perhaps I should have been waiting to have the reproducer before reporting...

I do have that message you mentioned!
The `CMakeLists.txt` is indeed providing the board rather than the command line.
Even without the warning, this appeared to be working lately, but switching to another environment and pulling  from a few Zephyr repos seemed to change the behavior.

With `strace`, I noticed this:

```
[pid 13587] execve("/.../zephyr/zephyr-sdk-0.16.3-rc1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc", ["/.../z"..., "-Wunique_unit_address_if_enabled", "-v"], 0x7ffe28abb488 /* 29 vars */) = -1 ENOENT (No such file or directory)
[pid 13587] write(10, "No such file or directory", 25) = 25
[pid 13587] exit_group(1)               = ?
[pid 13567] <... read resumed>"", 1)    = 0
[pid 13567] close(11)                   = 0
[pid 13567] rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
[pid 13567] read(9, "No such file or directory", 1024) = 25
[pid 13567] read(9, "", 999)            = 0
[pid 13567] close(9)                    = 0
[pid 13567] kill(13587, SIGSTOP)        = 0
[pid 13567] openat(AT_FDCWD, "/proc", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 9
...
```

So it appears that the "No such file or directory" is coming from the lack of the `dtc` command... Or problem related to dynamic libraries.

Either way, I will provide a reproducer soon. Thank you for your time and apologies for this incomplete bug report!

josuah, 2023-09-26:
This might be related to my Zephyr-SDK:

```
vm1$ dtc --version
Version: DTC 1.6.1
vm1$ zephyr-sdk-0.16.3-rc1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc --version
-bash: zephyr-sdk-0.16.3-rc1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: cannot execute: required file not found
vm1$127
```

Here is the full log (thanks to the power of sed!)

```
(venv) vm1$ west build -p -b litex_vexriscv zephyr_clnx 2>&1 | sed 's,path_with_the_project_name_not_so_confidential_but_well,...,g'
-- west build: making build dir /.../zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /.../zephyr/zephyr_clnx
-- CMake version: 3.25.1
-- Found Python3: /.../.venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /home/tinyvision/.cache/zephyr
-- Zephyr version: 3.4.99 (/.../zephyr/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: litex_vexriscv
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.3 (/.../zephyr/zephyr-sdk-0.16.3-rc1)
-- Found toolchain: zephyr 0.16.3 (/.../zephyr/zephyr-sdk-0.16.3-rc1)
-- Found Dtc: /.../zephyr/zephyr-sdk-0.16.3-rc1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (Required is at least version "1.4.6")
-- Found BOARD.dts: /.../zephyr/zephyr/boards/riscv/litex_vexriscv/litex_vexriscv.dts
-- Found devicetree overlay: app.overlay
-- Found devicetree overlay: litex.overlay
-- Generated zephyr.dts: /.../zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /.../zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /.../zephyr/build/zephyr/dts.cmake
CMake Error at /.../zephyr/zephyr/cmake/modules/dts.cmake:366 (message):
  command failed with return code: No such file or directory
Call Stack (most recent call first):
  /.../zephyr/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /.../zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /.../zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/.../.venv/bin/python3 -B/.../zephyr/build -GNinja -DBOARD=litex_vexriscv -S/.../zephyr/zephyr_clnx
(venv) vm1$
```

josuah, 2023-09-26:
Ah, a new SDK published 4 days ago. Downloading and trying again now...

josuah, 2023-09-26:
yes I did which did not affect the out come (but thank you for the suggestion).
The problem was closely related!

It appears that something wrong went with the SDK installation, and running the setup again fixed it for me:

```
(venv) vm1$ cd zephyr-sdk-0.16.3
(venv) vm1$ bash setup.sh -h
Zephyr SDK 0.16.3 Setup

Installing host tools ...

All done.
(venv) vm1$ sysroots/x86_64-pokysdk-linux/usr/bin/dtc --version
Version: DTC 1.6.0-dirty
(venv) vm1$
```

So reinstalling the host tools fixed the issue! \o/
It is possible that my installation was interrupted on the first attempt, or that anything else caused that issue.

This shows that in case the host tools have some problem, `dts.cmake` might be the first one to complain (through failed invocation of `dtc`).

josuah, 2023-09-26:
Thank you everyone! The bug was not related to Zephyr in the end, but to one particular host/setup (me).
Closing this :)

closed 2023-09-26 by josuah