zephyrproject-rtos/zephyr #63124: Misconfiguration of Message RAM for STM32H7

legoabram, 2023-09-26:
**Describe the bug**
The second instance of the FDCAN on an STM32H7 has improperly configured message RAM addresses which prevents operation.

The core of the problem comes from the call to `can_mcan_configure_mram` on line 137 in `drivers\can\can_stm32h7.c`. Since the same address is passed to both mrba, and mram, both M_CANs are configured to use the same message RAM at offset 0x0, even though FDCAN2 is supposed to be offset into it. If the second parameter is changed to `0x4000ac00`, correct operation is achieved. So this parameter should be given the absolute message ram base address instead of the base address for that particular instance.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure FDCAN2 for operation instead of FDCAN1
2. Send random/full message over FDCAN2
3. See empty message instead

[image]

**Expected behavior**
FDCAN2 is usable.

**Impact**
FDCAN2 is unusable.

**Environment**
 - OS: Windows 11
 - Toolchain: GCC arm-none-eabi 12.2.1
 - Zephyr v3.4.0 Release

**Additional context**
STM32H753VI

legoabram, 2023-09-27:
Already resolved. Fix wasn't applied to v3.4.0

closed 2023-09-27 by legoabram

henrikbrixandersen, 2023-09-27:
Just to be clear: The fix was done in https://github.com/zephyrproject-rtos/zephyr/pull/60935 which was backported to the `v3.4-branch` already (commit bd741f2f014499fe5c8f823de040140231f98048).