zephyrproject-rtos/zephyr #63159: Documentation build broken after CMSIS-DSP module update

SeppoTakalo, 2023-09-27:
**Describe the bug**

Documentation build fails.

**To Reproduce**
Steps to reproduce the behavior:
1. cd doc
2. cmake -GNinja -B_build .
3. cd _build
4. ninja html

**Expected behavior**
HTML documentation would build.

**Impact**
I cannot build the documentaion

**Logs and console output**
```
[1/2] Running Sphinx HTML build...
Running Sphinx v6.2.1
making output directory... done
Building Kconfig database...... failed

Traceback (most recent call last):
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 2188, in _enter_file
    self._readline = self._open(filename, "r").readline
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 3921, in _open
    open(filename, mode, encoding=self._encoding)
IsADirectoryError: [Errno 21] Is a directory: '/home/seta/src/ncs/zephyr/'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/seta/.virtualenvs/zephyr/lib/python3.8/site-packages/sphinx/events.py", line 96, in emit
    results.append(listener.handler(self.app, *args))
  File "/home/seta/src/ncs/zephyr/doc/_extensions/zephyr/kconfig/__init__.py", line 247, in kconfig_build_resources
    kconfig, module_paths = kconfig_load(app)
  File "/home/seta/src/ncs/zephyr/doc/_extensions/zephyr/kconfig/__init__.py", line 117, in kconfig_load
    return kconfiglib.Kconfig(ZEPHYR_BASE / "Kconfig"), module_paths
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 2988, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 3027, in _parse_block
    self._parse_block(_T_ENDMENU, node, node)
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 2988, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 2988, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 3027, in _parse_block
    self._parse_block(_T_ENDMENU, node, node)
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 2987, in _parse_block
    self._enter_file(filename)
  File "/home/seta/src/ncs/zephyr/scripts/kconfig/kconfiglib.py", line 2191, in _enter_file
    raise _KconfigIOError(
kconfiglib._KconfigIOError: /tmp/tmp7kq0k5e_/Kconfig.modules:26: Could not open '/home/seta/src/ncs/zephyr/' (in 'osource "$(ZEPHYR_CMSIS_DSP_KCONFIG)"') (EISDIR: Is a directory)

```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.6 LTS
 - Toolchain  Zephyr SDK 16.1
 - Commit SHA 104ea2ccca1f62048fcc9535e182809ffdf6fd8b

**Additional context**
I did bisecting and found out that the breaking commit was this:
```
104ea2ccca1f62048fcc9535e182809ffdf6fd8b is the first bad commit
commit 104ea2ccca1f62048fcc9535e182809ffdf6fd8b
Author: Ryan McClelland <ryanmcclelland@meta.com>
Date:   Sat Apr 15 17:26:09 2023 -0700

    modules: cmsis-dsp: add cmsis-dsp module
    
    Use CMSIS-DSP from its new realm. This also changes change how you
    initialize FFT tables as well to use arm_cfft_init_64_f32 if you
    know the FFT size in advance rather than the generic initialization
    arm_cfft_init_f32.
    
    Signed-off-by: Ryan McClelland <ryanmcclelland@meta.com>

 MAINTAINERS.yml                               |  15 +-
 modules/cmsis/Kconfig                         |   8 -
 modules/cmsis/Kconfig.cmsis_dsp               | 803 ----------------------
 modules/cmsis_dsp/CMakeLists.txt              | 931 ++++++++++++++++++++++++++
 modules/cmsis_dsp/Kconfig                     |  79 +++
 tests/benchmarks/cmsis_dsp/basicmath/prj.conf |   1 -
 tests/lib/cmsis_dsp/bayes/prj.conf            |   1 -
 tests/lib/cmsis_dsp/complexmath/prj.conf      |   1 -
 tests/lib/cmsis_dsp/distance/prj.conf         |   1 -
 tests/lib/cmsis_dsp/fastmath/prj.conf         |   1 -
 tests/lib/cmsis_dsp/filtering/prj.conf        |   1 -
 tests/lib/cmsis_dsp/filtering/prj_base.conf   |   1 -
 tests/lib/cmsis_dsp/interpolation/prj.conf    |   1 -
 tests/lib/cmsis_dsp/matrix/prj.conf           |   1 -
 tests/lib/cmsis_dsp/matrix/prj_base.conf      |   1 -
 tests/lib/cmsis_dsp/quaternionmath/prj.conf   |   1 -
 tests/lib/cmsis_dsp/statistics/prj.conf       |   1 -
 tests/lib/cmsis_dsp/support/prj.conf          |   1 -
 tests/lib/cmsis_dsp/svm/prj.conf              |   1 -
 tests/lib/cmsis_dsp/transform/prj.conf        |   1 -
 tests/lib/cmsis_dsp/transform/prj_base.conf   |   1 -
 tests/subsys/dsp/basicmath/prj.conf           |   1 -
 tests/subsys/dsp/basicmath/prj_arc.conf       |   1 -
 west.yml                                      |   3 +
 24 files changed, 1026 insertions(+), 831 deletions(-)

```

kartben, 2023-09-27:
Fix for this is in PR #63120

SeppoTakalo, 2023-09-28:
Fixed.

closed 2023-09-28 by SeppoTakalo