zephyrproject-rtos/zephyr #63166: Domains initialisation possibly insane

LukaszMrugala, 2023-09-27:
**Describe the bug**
Let's take a look at the initialisation of the Domains class:

https://github.com/zephyrproject-rtos/zephyr/blob/73b803ab4b50847f3865f3da1b175b07038a9f62/scripts/pylib/build_helpers/domains.py#L54-L69

We can see that a case where the data is incomplete, lacking domains, was considered, as a warning is then generated. However, such a consideration misses the point.

If `data` dictionary lacks the `domains` key entirely, the `__init__` will crash just after the warning is generated. If the key has an empty list, the object becomes a ticking timebomb, ready to crash at a `get_default_domain()` call. That is caused by a naming error, where a default value of `[]` is assigned to `self._domain_default` rather than `self._default_domain`. 

**Expected behavior**
Initialisation raises error or gracefully ends rather than crashing and does not leave the created object in an unstable state.

**Impact**
Domains module is unstable.

**Proposed solution**
```python
 def __init__(self, data): 
     self._domains = [] 
     self._domain_names = [] 
     self._default_domain = [] 
  
    if data is None:
      raise ValueError('Data not provided.') 
  
     self._build_dir = data.get('build_dir') 
     domain_list = data.get('domains') 
     if not domain_list: 
         raise ValueError('No domains provided.')
  
     for d in domain_list: 
         domain = Domain(d['name'], d['build_dir']) 
         self._domains.append(domain) 
         self._domain_names.append(domain.name) 
         if domain.name == data['default']: 
             self._default_domain = domain 
```

thedjnK, 2023-09-27:
CC @57300

57300, 2023-09-27:
This is improved somewhat on `main` now (https://github.com/zephyrproject-rtos/zephyr/commit/05d77690736b5ad022ee1d0f7dc9b4fea2bd2c24). `from_file()` and `get_default_domain()` will no longer crash.

But I can think of more improvements. The warning is indeed unnecessary, and more error checks can be added.

golowanow, 2023-10-05:
some new issue at CI:
see https://github.com/zephyrproject-rtos/zephyr/actions/runs/6417150473/job/17422293239?pr=63472

PR #63195 (2023-10-04): scripts: domains: More stability improvements
Fixes #63166

* Catch inconsistent `domains.yaml` contents early
* In the YAML schema, make "domains" a required key
* Remove unnecessary warnings
* Remove support for initializing `Domains` from a dictionary
* Bonus: Convert `Domain` into a dataclass (cosmetic)
commit f82e2ef0 (2023-10-04): scripts: domains: Tighten up initialization

Do extra checks to reject inconsistent `domains.yaml` contents early:

   default: app
   build_dir: /path/to/build
   domains:
     - name: image_1
       build_dir: /path/to/build/image_1
     - name: image_2
       build_dir: /path/to/build/image_2
   flash_order:
     - image_1
     - image_2
     - unknown

This example lists "image_1" and "image_2" as known domains, but an
entry for the default "app" is missing, which should never be allowed.
A valid `domains.yaml` file shall also not contain extra domains under
`flash_order`, such as "unknown [... cut]
closed 2023-10-11 by carlescufi