zephyrproject-rtos/zephyr #63191: Implement subsys/settings callback for notifying data saved

gwa-simon, 2023-09-28:
As was mentioned by `https://github.com/zephyrproject-rtos/zephyr/issues/31408` the operations are a little confused to be understood. It is not very clear for me when the handlers h_get, h_set, h_commit and h_export are called. 

I have tested registering all callbacks and notice that none of them is called when using mcumgr **settings write** command. It would be interesting to be able to register a notify callback that is called when the data is updated, so that the application can know about this update and react to it.

Is there already something like this implemented? If not, it would be a good improvement for the settings module.

nordicjm, 2023-09-28:
I don't know what you mean by they're not called, yes they are? You can see how this works in the test: https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/subsys/mgmt/mcumgr/settings_mgmt

There is also a notification hook system in MCUmgr which allows you to get callbacks when the functions themselves run, which includes what type of function and the key that was used.

gwa-simon, 2023-09-28:
Thanks @nordicjm for the quick answer. 

I don't mean that they are never called. What I mean is that I don't get any of the handlers (h_set, h_get, h_commit, h_export) called after `mcumgr shell exec write` command. Maybe I implemented it wrong. Should one of these be called by shell write?

nordicjm, 2023-09-28:
Oh, you are not really using MCUmgr then, just using it to remotely run a shell command. Not sure on how that works, but you could instead use the newly introduce MCUmgr settings mgmt group commands, these _will_ give you the callback to those functions and means you don't have to rely upon sending shell commands to a device

Laczen, 2023-09-29:
> Thanks @nordicjm for the quick answer.
> 
> I don't mean that they are never called. What I mean is that I don't get any of the handlers (h_set, h_get, h_commit, h_export) called after `mcumgr shell exec write` command. Maybe I implemented it wrong. Should one of these be called by shell write?

None of these handlers will be called when executing a shell write or a settings_save_one. The handlers are for use on "run time" representations of some stored data. When executing a shell write all that happens is that this is written to the backend, it is not applied in any way to the "run time" representation.

The shell write command is executed on a key, while handlers can be called for a part of a key.

If you would like to immediately apply the written data it would take two steps: write and then call the load command. In practical cases this is not needed because first the "run time" representation is changed and then a settings_save_one() is called.

*The term "run time" representation is used because there does not need to be a direct match between a key-value pair written to disk and a value in the code.*