zephyrproject-rtos/zephyr #63202: can: common: timing calculation: fill seg1 before using prop

alexanderwachter, 2023-09-28:
The SJW is limited my the min of seg1 and seg2. The current algorithm shares the first section of the timing (before the sp) between prop_seg and seg1. This artificially limits the sjw.

Put as much tqs in seg1 and only the rest into the prop seg.
 

Spottet in #63034

henrikbrixandersen, 2023-10-31:
I am not sure the proposed change brings much improvement. Distribution of `TSEG1` between `PROP_SEG` and `PHASE_SEG1` will always be specific to the given CAN network, since it depends on the maximum propagation delay for the transmitted signal to reach all receivers.

Many CAN controllers don't even support having a separate `PROP_SEG` and hence sets the minimum/maximum values for the `PROP_SEG` to 0, causing the calculations to use the entire `TSEG1` for `PHASE_SEG1`. For those CAN controllers, the proposed change will be a no-op,

FWIW, the Linux kernel implementation (https://elixir.bootlin.com/linux/latest/source/drivers/net/can/dev/calc_bittiming.c#L155) also calculates `PROP_SEG` as `TSEG1 / 2`.

I'd rather keep our current implementation and avoid introducing under-the-hood changes to the current timing calculations, which are difficult to prove correct in a working environment.

alexanderwachter, 2023-10-31:
The"problem" i thought off is that controllers with prop_seg could have smaller values for sjw than the one without. But it could also be that the limit in the sjw calculation will never trigger 🤷‍♂️.

alexanderwachter, 2023-10-31:
Should we close it?

henrikbrixandersen, 2023-10-31:
> Should we close it?

I think we should, yes.

closed 2023-10-31 by alexanderwachter