zephyrproject-rtos/zephyr #63206: No state of health in fuel gauge API

ThomasGooijers, 2023-09-28:
The experimental fuel gauge API offers a lot of common properties that can be retrieved from fuel gauges.

When utilizing this API to implement a custom driver for the BQ27520 gauge I noticed the API does not provide a state of health estimate (SOH), which this gauge and I assume many others can provide. 

The ability to monitor this property could be very useful because it could indicate whether or not the battery of your device needs to be replaced. 

It could look something like this:
""
FUEL_STATE_OF_HEALTH 
Battery state of health estimate (0-100%)
""

If this or something like it could be added to a future version of the API I think that would be a good improvement.

aaronemassey, 2023-09-28:
@teburd @rriveramcrus This seems similar to the charger health in https://github.com/zephyrproject-rtos/zephyr/pull/62551 How do we feel about exposing a type of health status from fuel gauges as well? Seems completely reasonable to me.

rriveramcrus, 2023-09-28:
@aaronemassey the state of health (SoH) in gauging-land is different than charger health in #62551, so there is no risk of conflict or overlap. I am okay with this.
`charger_health` is indicative of the conditional state of battery pack and/or charging circuit with respect to the ability to charge a battery, so it is of interest to the charger and the OS. The proposed fuel gauge STATE_OF_HEALTH is a dynamic characteristic of the battery pack only of interest to the gauge and OS.

aaronemassey, 2023-09-29:
Ack. My first priority at the moment is creating a set of generic API tests to make it easier for fuel gauge contributions and assuring drivers are consistent. Let me get to that first and I'll grab this. 

@ThomasGooijers If you'd like this done sooner I'm also happy to review a PR.

ThomasGooijers, 2023-10-02:
@aaronemassey I don't need it done right now, it's just a improvement suggestion. I wouldn't mind working on this myself but unfortunately I have other priorities right now.

ThomasGooijers, 2026-02-03:
@devbhavej 

It's been a while since I worked on this, so I had to refresh my memory.

I think what you are looking for is simply called 'StateOfCharge' which is filtered and rounded up to the nearest whole percent.
There is also a 'TrueSOC' register which is similar but unfiltered.

For the most common applications where you want a simple battery charge estimate display for the end user you should probably use the filtered value.

You can find a detailed description for all of the registers in the "Standard data commands" section of the Technical Reference Manual (I checked for the BQ27520-G4).

I probably can't be of much more help since I a not working with fuel gauges at the moment. But I hope I was able to point you in the right direction.