zephyrproject-rtos/zephyr #63217: Bluetooth: bt_conn_process_tx() does not destroy TX buffers when connection state is BT_CONN_DISCONNECT_COMPLETE

watsug, 2023-09-28:
**Describe the bug**
We have a peripheral BLE device based on Zephyr RTOS and another central one is based on iOS (iPhone mobile app).
The central sends a write request command (), and without waiting for the response drops the connection. This is the trace:

2122	14.507	Master_0xaf9a84ad	LE 2M	ATT	9	29763µs	0	0	False	51	Sent Write Request, Handle: 0x0013 (Unknown: Unknown: Client Characteristic Configuration)
2123	14.507	Slave_0xaf9a84ad	LE 2M	LE LL	0	149µs	0	1	False	51	Empty PDU
2124	14.537	Master_0xaf9a84ad	LE 2M	LE LL	2	29727µs	1	1	False	52	Control Opcode: LL_TERMINATE_IND
2125	14.537	Slave_0xaf9a84ad	LE 2M	LE LL	0	149µs	1	0	False	52	Empty PDU

On the peripheral side (Zephyr), the BLE host processes the write request command, creates a write response frame, and stores it in the TX queue. It looks like there is a timing or connection state machine issue, resulting in not releasing the net buffer (write response).

In the function bt_conn_process_tx() a connection is cleaned up only when the state is BT_CONN_DISCONNECTED. In our case sometimes this function is called when the state is BT_CONN_DISCONNECT_COMPLETE. In such a scenario the connection is not cleaned up, the function tries to process buffer from the TX queue, and send_buf() returns -128 (disconnected). With this error code, the function simply returns, leaving the buffer in the TX queue.  In such a situation buffer with 'write response' is not released, and finally, acl_tx_pool can be left empty making new connections unusable.

To investigate the problem I added an additional log to bt_conn_process_tx() function:
void bt_conn_process_tx(struct bt_conn *conn)
{
	struct net_buf *buf;
	int err;

	LOG_DBG("conn %p", conn);

	if (conn->state == BT_CONN_DISCONNECTED &&
	    atomic_test_and_clear_bit(conn->flags, BT_CONN_CLEANUP)) {
		LOG_DBG("handle %u disconnected - cleaning up", conn->handle);
		conn_cleanup(conn);
		return;
	}

	/* Get next ACL packet for connection. The buffer will only get dequeued
	 * if there is a free controller buffer to put it in.
	 *
	 * Important: no operations should be done on `buf` until it is properly
	 * dequeued from the FIFO, using the `net_buf_get()` API.
	 */
	buf = k_fifo_peek_head(&conn->tx_queue);
	BT_ASSERT(buf);

	/* Since we used `peek`, the queue still owns the reference to the
	 * buffer, so we need to take an explicit additional reference here.
	 */
	buf = net_buf_ref(buf);
	err = send_buf(conn, buf);
	net_buf_unref(buf);

	LOG_INF("bt_conn_process_tx(): send_buf() -> %d, conn->state -> %d", err, conn->state);
	if (err  == -EIO) {
		struct bt_conn_tx *tx = tx_data(buf)->tx;

And from time to time I can see this is the trace:
I: bt_conn_process_tx(): send_buf() -> -128, conn->state -> 1

**To Reproduce**
The problem was discovered on STM32WB55RG Nucleo. It is hard to reproduce it as this is a kind of timing issue. To reproduce it is required to simulate a call of bt_conn_process_tx() when the connection state is BT_CONN_DISCONNECT_COMPLETE.

**Expected behavior**
Buffers from conn->tx_queue should always be properly released.

**Impact**
When all buffers from acl_tx_pool have been used but not freed, even if the BLE connection can be opened it cannot be used to communicate over GATT.

**Logs and console output**
I: bt_l2cap_create_pdu_timeout():644: pool 0x20012824 size 526
I: allocated buf 0x2000af58
I: bt_ipm_send():393: buf 0x2000af58 ref 2 pool_id 0 frags (nil)
I: bt_conn_process_tx():941: buf 0x2000af58 ref 1 pool_id 0 frags (nil)
I: bt_conn_process_tx(): send_buf() -> 0, conn->state -> 7
I: bt_l2cap_create_pdu_timeout():644: pool 0x20012824 size 526
I: allocated buf 0x2000af58
D: CCC is being written from connection 0x20002378, attribute 0x200002dc, value 0
I: GATT unsubscribed so the connection is being closed.
D: CCC notify - attribute 0x200002dc, value 0
I: bt_conn_process_tx():941: buf 0x2000af58 ref 2 pool_id 0 frags (nil)
I: bt_conn_process_tx(): send_buf() -> -128, conn->state -> 1
D: Disconnected (reason 19)

**Environment (please complete the following information):**
 - OS: Linux, mobile application acting as a central on iPhone 12 mini, iOS 16.6.1
 - Zephyr SDK 0.15.1
 - Zephyr 3.4

**Additional context**

watsug, 2023-09-28:
Another idea is that the issue might be because of a non-atomic transition in conn_destroy() from BT_CONN_DISCONNECTING to BT_CONN_DISCONNECTED. See this:

static void conn_destroy(struct bt_conn *conn, void *data)
{
	if (conn->state == BT_CONN_CONNECTED ||
	    conn->state == BT_CONN_DISCONNECTING) {
		bt_conn_set_state(conn, BT_CONN_DISCONNECT_COMPLETE);
	}

	bt_conn_set_state(conn, BT_CONN_DISCONNECTED);
}

watsug, 2023-10-05:
An additional trace:
```
I: bt_l2cap_create_pdu_timeout():644: pool 0x20012824 size 526
I: allocated buf 0x2000af58
I: bt_conn_set_state(): 0x20002378, disconnect-complete, 0x20002848
I: bt_conn_process_tx(): conn 0x20002378, conn state: disconnect-complete, 0x20001fd0
I: bt_conn_process_tx():953: buf 0x2000af58 ref 2 pool_id 0 frags (nil)
I: send_buf(): result -128
I: bt_conn_set_state(): 0x20002378, disconnected, 0x200020a0
I: bt_conn_set_state(): 0x20002378, connecting-adv, 0x20001dd8
I: [0x20001800] BLE connection closed - 0x20002378
```
And the list of threads:
[image]

In this line:
```
bt_conn_set_state(): 0x20002378, disconnect-complete, 0x20002848
```
the last value is k_tid_t, so this means this is bt_ipm_rx_thread.

Same for this line:
```
I: bt_conn_process_tx(): conn 0x20002378, conn state: disconnect-complete, 0x20001fd0
```
the last value is k_tid_t, so this means this is hci_tx_thread.

carlescufi, 2023-10-05:
@watsug thanks for the thorough analysis. Based on your understanding of the issue, can you please open a Pull Request with a suggested fix so that @jhedberg but also others can have a chance to review it?

watsug, 2023-10-05:
@carlescufi 
Sure, I will prepare a PR right of the beginning of the next week.

carlescufi, 2023-10-06:
> @carlescufi Sure, I will prepare a PR right of the beginning of the next week.

Thank you!

theob-pro, 2023-10-12:
@watsug are you still planning on posting a PR for this?

watsug, 2023-10-19:
@theob-pro I had something urgent in the meantime, but I am going to create PR today.

jori-nordic, 2023-11-24:
So here's my analysis:
Starting from when the host receives the `DISCONNECT_COMPLETE` event.
- driver calls [`bt_recv`](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L3763)
- `bt_recv` processes the disconnect event [immediately](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L3788) *and* puts it [in the event RX queue](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L3792)
- `hci_disconn_complete_prio` [sets the state](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L839) to `BT_CONN_DISCONNECT_COMPLETE`
- `bt_conn_process_tx` runs, [tries](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/conn.c#L926) and fails to send the buffer. As you said, the buf is not popped from the queue. **This can end up leaking the buffers/metadata** because we don't check for disconnected before [popping the buf from the queue](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/conn.c#L681)
- `send_buf` returns, `buf` still has one ref. Maybe it's still in the queue, maybe not (e.g. if it's a fragment)
- `hci_disconn_complete` runs, puts the conn into [BT_CONN_DISCONNECTED](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/hci_core.c#L863)
- this transition [sets the `BT_CONN_CLEANUP` bit](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/conn.c#L1105)
- `bt_conn_process_tx` runs again, this time the stars are aligned (right bit and conn state) and [conn_cleanup](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/conn.c#L909) is called

So I think this is correct, you did find a leak, thanks! I'll approve the PR.

jori-nordic, 2023-11-24:
Regarding
> Is the transition from BT_CONN_CONNECTED to BT_CONN_DISCONNECT_COMPLETE proper

I don't have anymore insight than you on this, but it points to a commit addressing priority in handling the disconnected event.

closed 2024-04-08 by github-actions[bot]