zephyrproject-rtos/zephyr #63221: West: Bindesc: Crash when no argument is provided to `west bindesc`

kmeinhar, 2023-09-28:
**Describe the bug**

When executing `west bindesc` without any arguments Python crashes with backtrace:
```
Traceback (most recent call last):
  File "/home/zephyr-main/.venv/bin/west", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/zephyr-main/.venv/lib/python3.11/site-packages/west/app/main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "/home/zephyr-main/.venv/lib/python3.11/site-packages/west/app/main.py", line 244, in run
    self.run_command(argv, early_args)
  File "/home//zephyr-main/.venv/lib/python3.11/site-packages/west/app/main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "/home//zephyr-main/.venv/lib/python3.11/site-packages/west/app/main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/home/zephyr-main/.venv/lib/python3.11/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/home/zephyr-main/zephyr/scripts/west_commands/bindesc.py", line 194, in do_run
    self.is_big_endian = args.big_endian
                         ^^^^^^^^^^^^^^^
AttributeError: 'Namespace' object has no attribute 'big_endian'

```

**To Reproduce**
1. Download latest Zephyr main branch
2. Execute `west bindest`

**Expected behavior**

Pint the help page usually generated with `west bindesc --help`

kmeinhar, 2023-09-28:
@yonsch Please have look if you find the time.

nashif, 2023-09-28:
@yonsch  also please an area in the maintainer file.

PR #63225 (2023-09-28): bindesc: Add a few missing pieces
This PR adds a few missing pieces to bindesc:
1. Area in MAINTAINERS.yml
2. Release notes for 3.5
3. Improved documentation for the `hello_bindesc` sample
4. Fixed `west bindesc` crash when no subcommand is given

Fixes: #63221
closed 2023-09-28 by jhedberg