zephyrproject-rtos/zephyr #63280: drivers: spi_mcux_lpspi: spurious irq interrupt

jeppenodgaard, 2023-09-29:
**Describe the bug**
The `spi_mcux_lpspi.c` `spi_transceive` implementation sometimes fails.

This is due to a spurious interrupt in [LPSPI_MasterTransferHandleIRQ](https://github.com/zephyrproject-rtos/hal_nxp/blob/master/mcux/mcux-sdk/drivers/lpspi/fsl_lpspi.c#L1643C15-L1643C15).

The issue arises if the SPI clock speed is high and there are other interrupts occurring.

I have found a work around but not a solution:
```c
static void spi_mcux_isr(const struct device *dev)
...
	if (data->handle.state == (uint8_t)kLPSPI_Idle) {
		return;
	}

	LPSPI_MasterTransferHandleIRQ(base, &data->handle);
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Increase SPI clock frequency: `sed -i 's|6400000>|64000000>|g' zephyr/samples/drivers/led_ws2812/boards/mimxrt1050_evk.overlay`
2. Add `assert(handle->state != (uint8_t)kLPSPI_Idle);` in the top of `modules/hal/nxp/mcux/mcux-sdk/drivers/lpspi/fsl_lpspi.c:LPSPI_MasterTransferHandleIRQ`
3. Build sample: `west build -p always -b mimxrt1050_evk zephyr/samples/drivers/led_ws2812/ -- -DCONFIG_SAMPLE_LED_UPDATE_DELAY=0 -DCONFIG_SHELL=y -DCONFIG_ASSERT=y`
4. Since the issue is not consistent the error might occur by itself during boot. If not press and hold \<ENTER\> to stress the system.

```
*** Booting Zephyr OS build zephyr-v3.4.0-4642-g1084c8b083f0 ***
[00:00:00.000,000] <inf> main: Found LED strip device ws2812@0
[00:00:00.000,000] <inf> main: Displaying pattern on strip
uart:~$
uart:~$
uart:~$
uart:~$
uart:~$
uart:~$
uart:~$
uart:~$
uart:~$ assertion "handle->state != (uint8_t)kLPSPI_Idle" failed: file "WEST_TOPDIR/modules/hal/nxp/mcux/mcux-sdk/drivers/lpspi/fsl_lpspi.c", line 1566, function: LPSPI_MasterTransferHandleIRQ
abort()
[00:00:02.323,000] <err> os: r0/a1:  0x00000004  r1/a2:  0x80000ab0  r2/a3:  0x80000ab0
[00:00:02.323,000] <err> os: r3/a4:  0x800011b4 r12/ip:  0x0000ea60 r14/lr:  0x60008d31
[00:00:02.323,000] <err> os:  xpsr:  0x61000032
[00:00:02.323,000] <err> os: Faulting instruction address (r15/pc): 0x60008d3c
[00:00:02.323,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:02.323,000] <err> os: Fault during interrupt handling

[00:00:02.323,000] <err> os: Current thread: 0x80000ab0 (idle)
[00:00:02.370,000] <err> os: Halting system
```

**Expected behavior**
No spurious interrupt.

**Impact**
If thousands of `spi_transceive`s occurs every second at some point the spurious interrupt will run while a thread is in the middle of `spi_transceive` resulting in e.g. an absurdly high length which will result in the accessing an invalid address at some point:
```
[00:04:16.814,000] <err> spi_mcux_lpspi: Transfer could not start
[00:06:26.320,000] <err> spi_mcux_lpspi: Transfer could not start
[00:06:29.384,000] <err> spi_mcux_lpspi: Transfer could not start
[00:06:29.584,000] <err> spi_mcux_lpspi: Transfer could not start
[00:06:29.784,000] <err> spi_mcux_lpspi: Transfer could not start
[00:06:29.984,000] <err> spi_mcux_lpspi: Transfer could not start
[00:06:29.995,000] <err> os: ***** BUS FAULT *****
[00:06:29.995,000] <err> os:   Precise data bus error
[00:06:29.995,000] <err> os:   BFAR Address: 0x20280000
[00:06:29.995,000] <err> os: r0/a1:  0x20280000  r1/a2:  0x00000000  r2/a3:  0x00000000
[00:06:29.995,000] <err> os: r3/a4:  0x20280000 r12/ip:  0xaaaaaaaa r14/lr:  0x6005b76f
[00:06:29.995,000] <err> os:  xpsr:  0x81000032
[00:06:29.995,000] <err> os: Faulting instruction address (r15/pc): 0x6005b6ec
[00:06:29.995,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
[00:06:29.995,000] <err> os: Fault during interrupt handling
```

jeppenodgaard, 2023-09-29:
I just realized the reproduce steps requires waiting for https://github.com/zephyrproject-rtos/zephyr/pull/63275 to be merged or checking out https://github.com/jeppenodgaard/zephyr/tree/jepod/improve-led-ws2812

DerekSnell, 2023-10-03:
Hi @jeppenodgaard ,

We have not been able to replicate this issue yet, even at 64MHz SPI clock.  However, it seems you are testing this at 64MHz, and that violates the maximum LPSPI frequency of the RT1050.  From the [datasheet](https://www.nxp.com/docs/en/data-sheet/IMXRT1050CEC.pdf), the maximum LPSPI frequency is 30MHz.

Can you test on your hardware at 30MHz or lower, and see if you still have these spurious interrupts?  Thanks

jeppenodgaard, 2023-10-04:
Hi @DerekSnell.
I have also tried with 30 MHz which also causes the issue for me:
[Screencast from 10-04-2023 09:49:10 AM.webm](https://github.com/zephyrproject-rtos/zephyr/assets/4233729/5613dbe4-563d-4574-9308-3667307e8cdb)

danieldegrasse, 2023-10-05:
Hi @jeppenodgaard

I have been able to reproduce the spurious interrupt you mention. However, I do not think this is an issue, but rather an expected result of the way that the LPSPI HAL driver manages the TX FIFO. You can see in the implementation of the `LPSPI_MasterTransferFillUpTxFifo` function that the driver will continue to write data until the FIFO is filled:
https://github.com/zephyrproject-rtos/hal_nxp/blob/d3770a0434cd15df3f89cd7d38932395679979ee/mcux/mcux-sdk/drivers/lpspi/fsl_lpspi.c#L1504.

With a fast SPI clock, it is possible that the LPSPI will partially drain the FIFO as the core fills it. I see this happening with a SPI clock of 30MHz. In the 48 byte transfer, I see the following:
- After the first call to `LPSPI_MasterTransferFillUpTxFifo` (from `LPSPI_MasterTransferNonBlocking`), 26 bytes remain to be written
- After the second call to `LPSPI_MasterTransferFillUpTxFifo` (from `LPSPI_MasterTransferHandleIRQ`) 4 bytes remain to be written
- After the third call to `LPSPI_MasterTransferFillUpTxFifo` (from `LPSPI_MasterTransferHandleIRQ`), 0 bytes remain to be written

At this point, the FIFO can drain rather quickly. The HAL next checks if all bytes have been written: https://github.com/zephyrproject-rtos/hal_nxp/blob/d3770a0434cd15df3f89cd7d38932395679979ee/mcux/mcux-sdk/drivers/lpspi/fsl_lpspi.c#L1743, and since they have it will check to see if the transfer has completed.

At higher clock speeds, the FIFO will have drained at this point, so we will take the conditional branch here, since the transmit complete flag will be set in the LPSPI: https://github.com/zephyrproject-rtos/hal_nxp/blob/d3770a0434cd15df3f89cd7d38932395679979ee/mcux/mcux-sdk/drivers/lpspi/fsl_lpspi.c#L1748

**However, we have yet to disable the transmit data interrupt (TDIE is still set), so there will be another interrupt pending at this point**

At this point, we will disable the interrupt request flag and set the module as idle, but the last "spurious interrupt" is already pending and will fire. This appears to be the critical difference at high clock speeds- the FIFO will drain before we have a chance to disable the transmit data interrupt, so one additional interrupt will fire.

However, I don't see why this would cause any issues. According to the ARMv7 architecture reference manual, exceptions should chain in this case: 
> On an exception return, there can be a pending exception with sufficient priority to preempt the execution being
returned to. This results in an exception entry sequence immediately after the exception return sequence. This is
referred to as chaining the exceptions.

So the next "spurious interrupt" on the LPSPI should fire before the running application can configure the SPI to transfer additional data. The `LPSPI_MasterTransferHandleIRQ` will handle this case, and simply not change the LPSPI configuration.

To summarize, as far as I can tell this "spurious interrupt" is expected by the HAL driver when the LPSPI is using a fast clock speed, and should not cause issues. I assume there was a fault that caused you to report this issue initially. Do you have any details on how I could reproduce that?

jeppenodgaard, 2023-10-05:
Hi @danieldegrasse..

Thank you for looking into this!

Unfortunately I cannot share the code that caused the fault due to licenses and custom hardware, so I am not able to share the setup that can reproduce it.

I think what happens is that because a thread is allowed to use `spi_transceive` before the pending interrupt has fired, the ISR might run in the middle of e.g. `spi_context_buffers_setup`.

I reported the issue due to this fault https://github.com/zephyrproject-rtos/zephyr/pull/62243#issuecomment-1721398086 which shows the state of `struct spi_mcux_data` where `totalByteCount` has a very high value.

danieldegrasse, 2023-10-05:
> I think what happens is that because a thread is allowed to use spi_transceive before the pending interrupt has fired, the ISR might run in the middle of e.g. spi_context_buffers_setup.

This should not be possible if running in the thread context- since the LPSPI interrupt will already be pending at the exit of the last call to `LPSPI_MasterTransferHandleIRQ`. Are you possibly using the SPI async API? In that case I could see this spurious interrupt causing issues, because it would be possible to start another SPI transfer from an interrupt context

jeppenodgaard, 2023-10-05:
True.
The synchronous SPI API is used.
I must admit I have been trying but not been able to figure out how it happens.

danieldegrasse, 2023-10-05:
> I must admit I have been trying but not been able to figure out how it happens.

Could the issue be reproduced with a toy sample, that calls `spi_transceive` in a loop? Also, are you only transmitting data, or are you receiving it as well?

jeppenodgaard, 2023-10-05:
I'll try to remove the assert and see if I can reproduce the crash with the sample.
I both transmit and receive. I can share code snippets to show exactly how the SPI buffers are configured.

I get access to the source and hardware tomorrow.

jeppenodgaard, 2023-10-06:
Here's the snippet/implementation:

`EscALEvent.Byte` is a two uint8_t array.

```
static uint8_t spi_read_wait_byte = 0xFF;
static uint8_t spi_read_terminate_byte = 0xFF;

void HW_EscRead( MEM_ADDR *pData, UINT16 Address, UINT16 Len )
{
	uint8_t addr_cmd_buf[2];

	const struct spi_buf rx_buf[] = {
		{
			.buf = EscALEvent.Byte,
			.len = sizeof(EscALEvent),
		},
		{
			.buf = NULL,
			.len = 1,
		},
		{
			.buf = (uint8_t *)pData,
			.len = Len,
		},
	};
	const struct spi_buf_set rx_bufs = {
		.buffers = rx_buf,
		.count = ARRAY_SIZE(rx_buf)
	};

	const struct spi_buf tx_buf[] = {
		{
			.buf = addr_cmd_buf,
			.len = sizeof(addr_cmd_buf),
		},
		{
			.buf = &spi_read_wait_byte,
			.len = sizeof(spi_read_wait_byte),
		},
		{
			/* If NULL spi_mcux_lpspi writes 0x00 */
			.buf = NULL,
			.len = Len > 1 ? Len - 1 : 0,
		},
		{
			.buf = &spi_read_terminate_byte,
			.len = sizeof(spi_read_terminate_byte),
		},
	};
	const struct spi_buf_set tx_bufs = {
		.buffers = tx_buf,
		.count = ARRAY_SIZE(tx_buf)
	};

	addr_cmd_buf[0] = Address >> 5;
	addr_cmd_buf[1] = Address << 3 | ESC_RDW;

        spi_transceive(spi_dev, &spi_cfg, &tx_bufs, &rx_bufs);
}

void HW_EscWrite( MEM_ADDR *pData, UINT16 Address, UINT16 Len )
{
	uint8_t addr_cmd_buf[2];

	const struct spi_buf rx_buf[] = {
		{
			.buf = EscALEvent.Byte,
			.len = sizeof(EscALEvent),
		},
		{
			.buf = NULL,
			.len = Len,
		}
	};
	const struct spi_buf_set rx_bufs = {
		.buffers = rx_buf,
		.count = ARRAY_SIZE(rx_buf)
	};

	const struct spi_buf tx_buf[] = {
		{
			.buf = addr_cmd_buf,
			.len = sizeof(addr_cmd_buf),
		},
		{
			.buf = (uint8_t *)pData,
			.len = Len,
		},
	};
	const struct spi_buf_set tx_bufs = {
		.buffers = tx_buf,
		.count = ARRAY_SIZE(tx_buf)
	};

	addr_cmd_buf[0] = Address >> 5;
	addr_cmd_buf[1] = Address << 3 | ESC_WR;

	spi_transceive(spi_dev, &spi_cfg, &tx_bufs, &rx_bufs);
}
```

danieldegrasse, 2023-10-09:
Hi @jeppenodgaard, thank you for providing the snippet. Is it clear which function starts the transfer that causes the crash, between `HW_EscWrite` and `HW_EscRead`?

jeppenodgaard, 2023-10-09:
Hi @danieldegrasse!
I have not been able to figure out if one or both functions cause the issue.

jeppenodgaard, 2023-10-25:
Hi @danieldegrasse.
I know you have probably been busy since v3.5 has just been release, but any news on this issue?

jeppenodgaard, 2023-12-25:
The issue is not resolved. The label should be removed.

danieldegrasse, 2024-01-24:
> The issue is not resolved. The label should be removed.

Hi @jeppenodgaard- sorry for the delay responding here. I'm not sure how I can make much more progress here without some sample app that reproduces the issue. I understand that you can't share the code itself, but could you possibly reproduce the issue with some sort of application that writes data via the LPSPI in a loop? A reproducible sample will make it easier to narrow down what could be going wrong here.

jeppenodgaard, 2024-04-09:
The issue has not been resolved and a workaround might be required as described in the [issue description](https://github.com/zephyrproject-rtos/zephyr/issues/63280#issue-1918907511).

decsny, 2025-01-24:
Hi, I have just rewritten the entire LPSPI driver in https://github.com/zephyrproject-rtos/zephyr/pull/82877 , can you please try that version of the driver to see if it fixes your problem.  

I have an idea of what this spurious interrupt might be since I had to spend some days debugging a spurious RX interrupt as well and found out that what was happening was that the RX interrupt was being pended to the core before being disabled during the transfer handler, meaning it will happen despite no intent to handle RX.  Maybe the SDK has a similar issue that is not fixed.  The rewrite of the driver does not use the SDK.

jeppenodgaard, 2025-01-25:
Hi @decsny !

I am happy to hear you were able to reproduce it.

I would like to test and hope to get access to the hardware soon.

decsny, 2025-01-25:
> Hi [@decsny](https://github.com/decsny) !
> 
> I am happy you were able to reproduce it.
> 
> I would like to test and hope to get access to the hardware soon.

To be clear, I would not say that I reproduced it, because I was essentially using a totally different driver, I just think it's possible that maybe the SDK driver had a similar problem that I ran into

jeppenodgaard, 2025-01-29:
> To be clear, I would not say that I reproduced it, because I was essentially using a totally different driver, I just think it's possible that maybe the SDK driver had a similar problem that I ran into

The issue cannot reproduce it with the steps described here 👍 
I experienced the issue on with custom hardware and software but I cannot currently test that.
Closing the issue.

jeppenodgaard, 2025-01-29:
Since the PR has not yet been merged I probably closed this prematurely.
However I assume the PR will be merged thus keeping this closed.

decsny, 2025-01-29:
> Since the PR has not yet been merged I probably closed this prematurely. However I assume the PR will be merged thus keeping this closed.

I'll reopen it, because GitHub has a linking feature that will close an issue automatically when linked PR is merged

PR #82877 (2025-01-30): NXP LPSPI: Fix native CS on non-DMA driver
Have to rework this driver completely to fix the CS because the SDK handle is based on the idea of a transfer being a single buffer, so we can't use the SDK transfer function

depends on:

- https://github.com/zephyrproject-rtos/zephyr/pull/82650
- https://github.com/zephyrproject-rtos/zephyr/pull/82806
- https://github.com/zephyrproject-rtos/zephyr/pull/82750
- https://github.com/zephyrproject-rtos/zephyr/pull/84406

Fixes #16544
Fixes #63280
closed 2025-01-30 by kartben