zephyrproject-rtos/zephyr #63284: NET_EVENT_L4_CONNECTED not ready

MTsimon, 2023-09-29:
On branch:  v3.4-branch

When connection manager post event `NET_EVENT_L4_CONNECTED`, I expect the net interface to be ready, at the very moment this event occur.
This seems not to be the case every time.
If I try to resolve a host address using `dns_get_addr_info` ( i've made a wrapper arround it `dns_get_addr_info_blocking`) it messes something up with the iface. Meaning even the shell command `net ping 8.8.8.8` is timing out. It happens quite rarely, but often enough to be a serious issue.

If I add a small delay, before doing the DNS lookup i works.

```
[00:00:10.428,375] <dbg> net_dns_resolve: dns_write: (main): [0] submitting work to server idx 0 for id 47778 hash 38030
[00:00:10.428,588] <dbg> net_if: net_if_tx: (main): Processing (pkt 0x200370a0, prio 0) network packet iface 0x200018b0/1
[00:00:10.438,415] <dbg> net_if: net_if_tx: (main): Calling context send cb 0x20006520 status 71
[00:00:10.438,446] <dbg> dns: dns_get_addr_info_blocking: Waiting for DNS lookup to finish
[00:00:15.428,436] <dbg> net_dns_resolve: query_timeout: (sysworkq): Query timeout DNS req 47778 type 1 hash 38030
[00:00:15.428,497] <dbg> net_dns_resolve: dns_resolve_cancel_with_hash: (sysworkq): Cancelling DNS req 47778 (name google.dk type 1 hash 38030)
[00:00:15.428,527] <dbg> dns: dns_resolve_cb: DNS lookup cb status: -101
[00:00:15.428,588] <err> dns: DNS lookup for google.dk
[00:00:16.428,955] <dbg> net_dns_resolve: dns_write: (main): [0] submitting work to server idx 0 for id 21038 hash 38030
[00:00:16.429,168] <dbg> net_if: net_if_tx: (main): Processing (pkt 0x200370a0, prio 0) network packet iface 0x200018b0/1
[00:00:16.439,178] <dbg> net_if: net_if_tx: (main): Calling context send cb 0x20006520 status 71
[00:00:16.439,208] <dbg> dns: dns_get_addr_info_blocking: Waiting for DNS lookup to finish
[00:00:21.429,016] <dbg> net_dns_resolve: query_timeout: (sysworkq): Query timeout DNS req 21038 type 1 hash 38030
[00:00:21.429,077] <dbg> net_dns_resolve: dns_resolve_cancel_with_hash: (sysworkq): Cancelling DNS req 21038 (name google.dk type 1 hash 38030)
```

This is working:
```c++
if (mgmt_event == NET_EVENT_L4_CONNECTED) {
  LOG_INF("Network connected");
  struct net_if_config *cfg;
  cfg = net_if_get_config(iface);
  if (!cfg) {
    LOG_ERR("Cannot get network config");
    return;
  }
  char buf[NET_IPV4_ADDR_LEN];
  LOG_WRN("Device IP address: %s", net_addr_ntop(AF_INET, &cfg->ip.ipv4->unicast[0].address.in_addr, buf, sizeof(buf)));
  internet_connected = true;
  k_sleep(K_MSEC(1000)); // add small delay
  sys_evt_post(SYS_EVT_INTERNET_OK);
  return;
}
```

This is not:

```c++
if (mgmt_event == NET_EVENT_L4_CONNECTED) {
  LOG_INF("Network connected");
  struct net_if_config *cfg;
  cfg = net_if_get_config(iface);
  if (!cfg) {
    LOG_ERR("Cannot get network config");
    return;
  }
  char buf[NET_IPV4_ADDR_LEN];
  LOG_WRN("Device IP address: %s", net_addr_ntop(AF_INET, &cfg->ip.ipv4->unicast[0].address.in_addr, buf, sizeof(buf)));
  internet_connected = true;

  sys_evt_post(SYS_EVT_INTERNET_OK);
  return;
}
```

rlubos, 2023-09-29:
Please provide more details, what L2/driver are you using, do you configure IP statically or with DHCP?

> sys_evt_post(SYS_EVT_INTERNET_OK);

Also please note that `NET_EVENT_L4_CONNECTED` does not imply internet connectivity, we don't have a mechanism in Zehpyr to verify that (at least not yet).

MTsimon, 2023-09-29:
I am using PPP as L2, with a BG77 modem. 

I have tested with shell command `net ppp ping 1` to check if ppp is working, it is. I get a reply within milliseconds.

Also the ppp Iface is up, and has obtained an IP address.

Using ´net iface show´ command to check.

jukkar, 2023-10-02:
Did you try with some other value for the sleep delay? The 1sec delay does not look very small actually, just wondering if you see the same issue "fixed" with much smaller delay. If yes, then it could indicate some thread priority issue i.e., some important thread might get chance to run.

rlubos, 2023-10-02:
> Also the ppp Iface is up, and has obtained an IP address.

So the conditions to tirgger `NET_EVENT_L4_CONNECTED` event by connection manager are satisfied. So perhaps the issue is with the PPP interface, reporting the operational UP state too early.

rlubos, 2023-10-02:
FYI @bjarki-trackunit

bjarki-trackunit, 2023-10-02:
The event you should be listening for is `NET_EVENT_DNS_SERVER_ADD`. The following is taken from the `samples/net/cellular_modem sample`
```
	printk("Waiting for L4 connected\n");
	ret = net_mgmt_event_wait_on_iface(net_if_get_default(),
					   NET_EVENT_L4_CONNECTED, &raised_event, &info,
					   &info_len, K_SECONDS(120));

	if (ret != 0) {
		printk("L4 was not connected in time\n");
		return -1;
	}

	printk("Waiting for DNS server added\n");
	ret = net_mgmt_event_wait_on_iface(net_if_get_default(),
					   NET_EVENT_DNS_SERVER_ADD, &raised_event, &info,
					   &info_len, K_SECONDS(10));

	printk("Performing DNS lookup of %s\n", SAMPLE_TEST_ENDPOINT_HOSTNAME);
	ret = sample_dns_request();
	if (ret < 0) {
		printk("DNS query failed\n");
		return -1;
	}
```

jukkar, 2023-10-02:
> The event you should be listening for is `NET_EVENT_DNS_SERVER_ADD`.

Hmm, why so? I mean that if you are interested in noticing when DNS server is added to the system, then yes. But if you want to know about the system connectivity status then the `L4_CONNECTED` should be just fine.

MTsimon, 2023-10-03:
@bjarki-trackunit 

The `NET_EVENT_DNS_SERVER_ADD` event, is comming before the `NET_EVENT_L4_CONNECTED` event - the `sys_evt_post` I have created, is also posting an event internally to my program, when the DNS server is added. 

I have a "wait for all events" function waiting for both `NET_EVENT_DNS_SERVER_ADD` and `NET_EVENT_L4_CONNECTED`, before trying to resolve any hostnames using the `dns_get_addr_info`. 

@jukkar 

Yes, I have tried with other delays than 1000ms. I cant say for sure that 1000 ms is the sweet spot. 
Although I have not tried any lower values, only higher.

bjarki-trackunit, 2023-10-03:
Have you enabled `CONFIG_NET_L2_PPP_OPTION_DNS_USE=y`

MTsimon, 2023-10-03:
> Have you enabled `CONFIG_NET_L2_PPP_OPTION_DNS_USE=y`

Yes

bjarki-trackunit, 2023-10-03:
Using the following test setup:
- board: nrf9160dk
- modem: Quectel BG95
- driver: modem_cellular.c
- sample: net/cellular_modem
- zephyr version: v3.5.0-rc1

With these additional options:
- CONFIG_LOG=y
- CONFIG_NET_LOG=y
- CONFIG_NET_MGMT_EVENT_LOG_LEVEL_DBG=y
- CONFIG_NET_MGMT_EVENT_INFO=y
- CONFIG_LOG_BUFFER_SIZE=8192

I observe the following log output
```
...
[00:00:36.226,226] <inf> modem_cellular: event registered
[00:00:36.226,287] <dbg> net_mgmt: mgmt_thread: (net_mgmt): Handling events, forwarding it relevantly
[00:00:36.226,318] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Event layer 1 code 1 cmd 2
[00:00:36.226,440] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Running callback 0x2000524c : 0x1a591
[00:00:36.226,501] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Running callback 0x20004574 : 0x9d49
[00:00:36.342,163] <dbg> net_mgmt: net_mgmt_event_notify_with_info: (rx_q[0]): Notifying Event layer 3 code 276 type 3
[00:00:36.342,712] <dbg> net_mgmt: mgmt_thread: (net_mgmt): Handling events, forwarding it relevantly
[00:00:36.342,773] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Event layer 3 code 276 cmd 3
[00:00:36.342,803] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Unlocking 0x20007b6c synchronous call
Waiting for DNS server added
[00:00:36.345,275] <dbg> net_mgmt: mgmt_event_wait_call: (main): Synchronous event 0xf9140003 wait 0x20007b6c
[00:00:36.345,336] <dbg> net_mgmt: net_mgmt_add_event_callback: (main): Adding event callback 0x20007b6c
[00:00:36.358,489] <dbg> net_mgmt: net_mgmt_event_notify_with_info: (rx_q[0]): Notifying Event layer 2 code 4 type 1
[00:00:36.358,673] <dbg> net_mgmt: mgmt_thread: (net_mgmt): Handling events, forwarding it relevantly
[00:00:36.358,703] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Event layer 2 code 4 cmd 1
[00:00:36.358,764] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Running callback 0x20005224 : 0x1a4d5
[00:00:36.358,947] <dbg> net_mgmt: net_mgmt_event_notify_with_info: (): Notifying Event layer 3 code 276 type 1
[00:00:36.359,069] <dbg> net_mgmt: mgmt_thread: (net_mgmt): Handling events, forwarding it relevantly
[00:00:36.359,100] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Event layer 3 code 276 cmd 1
[00:00:36.359,161] <dbg> net_mgmt: mgmt_run_callbacks: (net_mgmt): Unlocking 0x20007b6c synchronous call
Performing DNS lookup of test-endpoint.com
[00:00:36.731,658] <dbg> net_dns_resolve: dns_resolve_name: (main): DNS id will be 3596
[00:00:36.731,781] <dbg> net_dns_resolve: dns_write: (main): [0] submitting work to server idx 0 for id 3596 hash 44215
Opening UDP socket
...
```
Parsing the events:
- Event layer 1 code 1 cmd 2 -> L2, NET_IF_CORE, NET_IF_UP
- Event layer 3 code 276 cmd 3 -> L4, L4_CORE, DNS_SERVER_ADD
- Event layer 3 code 276 cmd 1 -> L4, L4_CORE, CONNECTED
- Event layer 2 code 4 cmd 1 -> L3, IPV4_CORE, ADDR_ADD

@MTsimon  is correct that the DNS_SERVER_ADD event occurs before L4_CONNECTED, the function `net_mgmt_event_wait_on_iface` is not working as was expecting it to (I will create an issue for this later).

In any case, my sample is working, as long as the DNS request is performed after the L4_CONNECTED event.

Trying to recreate your issue, if I try to make a DNS request before servers are added, i get
```
<dbg> net_dns_resolve: dns_resolve_init_locked: (main): No DNS servers defined.
```

If I try to make a DNS request right after servers are added, i get
```
[00:00:40.593,078] <dbg> net_dns_resolve: dns_resolve_name: (main): DNS id will be 14726
[00:00:40.593,200] <dbg> net_dns_resolve: dns_write: (main): [0] submitting work to server idx 0 for id 14726 hash 44215
[00:00:40.593,383] <dbg> net_dns_resolve: dns_write: (main): Cannot send query (-22)
[00:00:40.593,414] <dbg> net_dns_resolve: dns_resolve_name: (main): DNS query failed 1 times
```

I can not recreate your specific issue, leading me to believe the error is not related to the DNS_SERVER_ADD or L4_CONNECTED events. Specifically, if the server has not been added, we get the error `No DNS servers defined`, if L4 is not connected, we get `Cannot send query (-22)` which both seem appropriate. In your case, it seemingly has a server defined, and is able to send the query, its just timing out (seemingly)

I suspect it is your wrapper `dns_get_addr_info_blocking` which is not working properly, I believe this given that there is no log from the DNS library indicating that a query has been received (which we should probably add TBH), see the first log in this message.

For inspiration, this is the wrapper I used for the modem_cellular sample
```
K_SEM_DEFINE(dns_query_sem, 0, 1);

static void sample_dns_request_result(enum dns_resolve_status status, struct dns_addrinfo *info,
				      void *user_data)
{
	if (sample_test_dns_in_progress == false) {
		return;
	}

	if (status != DNS_EAI_INPROGRESS) {
		return;
	}

	sample_test_dns_in_progress = false;
	sample_test_dns_addrinfo = *info;
	k_sem_give(&dns_query_sem);
}

static int sample_dns_request(void)
{
	static uint16_t dns_id;
	int ret;

	sample_test_dns_in_progress = true;
	ret = dns_get_addr_info(SAMPLE_TEST_ENDPOINT_HOSTNAME,
				DNS_QUERY_TYPE_A,
				&dns_id,
				sample_dns_request_result,
				NULL,
				19000);
	if (ret < 0) {
		return -EAGAIN;
	}

	if (k_sem_take(&dns_query_sem, K_SECONDS(20)) < 0) {
		return -EAGAIN;
	}

	return 0;
}
```

Set a breakpoint in `subsys/lib/dns/resolve.c` here to catch the response if it is received.
```
line 636
			invoke_query_callback(DNS_EAI_INPROGRESS, &info,
					      &ctx->queries[*query_idx]);
```

carlescufi, 2023-10-04:
@MTsimon this happens only in Zephyr 3.4.0 or can you reproduce it also on the main branch?

carlescufi, 2023-10-11:
@MTsimon ping

MTsimon, 2023-10-18:
@carlescufi: Apologies for not providing an update earlier; I've been quite busy. I haven't had a chance to test it on the main branch yet. Currently, my project isn't compiling on it, so I'm unable to conduct the test.

@bjarki-trackunit: Thank you for delving deeply into the issue. It's not a DNS lookup problem; it's a general network issue. I have to insert a delay with every network transmit attempt after receiving the `NET_EVENT_L4_CONNECTED` event. Interestingly, your DNS lookup inspiration code is actually identical to the one I had created.

I dont know if it has something to do with the really slow modem connection. Its connected to a LTE-M network. 
Not sure if the problem would exist, if it were on an ethernet connection.

carlescufi, 2023-12-13:
@MTsimon any progress or additional info on this issue?

carlescufi, 2024-01-10:
Closing due to inactivity. Please reopen if still an issue.

closed 2024-01-10 by carlescufi