zephyrproject-rtos/zephyr #63307: Can not use USE_SEGGER_RTT

KlemenDEV, 2023-09-29:
**Describe the bug**

When trying to use USE_SEGGER_RTT, I get the following fault

> warning: USE_SEGGER_RTT (defined at modules\segger\Kconfig:12) was assigned the value 'y' but got
> the value 'n'. Check these unsatisfied dependencies: HAS_SEGGER_RTT (=n), 0 (=n). See
> http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_USE_SEGGER_RTT and/or look up
> USE_SEGGER_RTT in the menuconfig/guiconfig interface. The Application Development Primer, Setting
> Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
> too.

**To Reproduce**
Steps to reproduce the behavior:

* Clone fresh zephyr install using west
* Build this sample: west build -p always -b nrf52833dk_nrf52833 samples/bluetooth/hci_uart

**Expected behavior**

RTT should be enabled.

This bug prevents from using RTT in the current latest release (west update) of zepyhr.

**More context**

I have checked what HAS_SEGGER_RTT is selected by (https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_HAS_SEGGER_RTT) and it is selected by CONFIG_SOC_SERIES_NRF52X (https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SOC_SERIES_NRF52X) so not sure why this is not working.

It used to work properly in older versions of zepyhr in the same sample, but since I updated it, this is happening.

nordicjm, 2023-09-29:
Works fine for me, you probably forgot to do `west update` or you have a manifest which blocks getting the segger repo

KlemenDEV, 2023-09-29:
This has happened after I ran `west update`. I was able to use RTT just fine before running this command after picking up the project after a while

KlemenDEV, 2023-09-29:
> you have a manifest which blocks getting the segger repo

Where can I check that?

nordicjm, 2023-09-29:
> > you have a manifest which blocks getting the segger repo
> 
> Where can I check that?

Are you using a custom manifest or `west init && west update` directly?

KlemenDEV, 2023-09-29:
Ok, I see what you mean, I use https://github.com/zephyrproject-rtos/example-application for the base of the workspace.

Strange that it worked before.

What do I need to add in west.yml for this to work?

KlemenDEV, 2023-09-29:
Ok got it, sorry for the confusion.

For anyone having a similar problem, one needs to either remove `name-allowlist` or add segger to it if using https://github.com/zephyrproject-rtos/example-application

closed 2023-09-29 by KlemenDEV