zephyrproject-rtos/zephyr #63308: drivers: disk_access and sd: Hardcoded SD clock .

Dgthhaivu, 2023-09-29:
**Describe the bug**
In the Zephyr "Disk Access" driver (`disk_access.c`), when looking into the initialization function which is the `disk_access_init()`, it's calling the `sd_init_io()` function in `sd.c` for initialization, and in that function the clock frequency is hard code to 400KHz using this macro `SDMMC_CLOCK_400KHZ`, is this the correct way to set the SD clock frequency ? Can we update it while initializing ?

```
static int sd_init_io(struct sd_card *card)
{
	struct sdhc_io *bus_io = &card->bus_io;
	int ret;

	/* SD clock should start gated */
	bus_io->clock = 0;
	
        ...
	
	bus_io->clock = SDMMC_CLOCK_400KHZ;
	ret = sdhc_set_io(card->sdhc, bus_io);
	if (ret) {
		LOG_ERR("Could not start bus clock");
		return ret;
	}
	return 0;
}
```
I'm using STM32 Nucleo_G474RE as the target.

**Expected behavior**
It will cause an error in `spi_ll_stm32(): Unsupported frequency ...` if the minimum clock frequency (664062Hz in my case) is greater than the default SD clock frequency (400KHz), please find the error image attached in below.

**Impact**
If the rcc frequency is too high, the only way to address the mentioned error is to update a new value for the SD clock speed by hardcoding the `bus->clock` using `sdhc_clock_speed` macros defined in `sd_spec.h`.

**Logs and console output**
[image]

**Additional context**
Is there a way to update the SD bus clock via devicetree or Kconfig during the initialization so that it won't cause an error in `spi_ll_stm32()` ? Otherwise I think it will need a new feature to update the clock frequency for the SD frquency.

danieldegrasse, 2023-10-05:
> I was wondering if hardcoding the frequency in sd card driver is correct or not.
Is there a way to just configure the SPI frequency to a specific frequency while using the sd driver (e.g 400KHz) without effecting other drivers using it?

The problem here is that it is correct- while a given card will *probably* respond at 600KHz, the spec dictates that cards initialize at 100-400KHz.

@erwango, do you have any insight into what we could do here? We could add a workaround in the SD subsystem to allow the user to specify an initialization frequency manually, but I'd prefer to avoid this if possible.

erwango, 2023-10-05:
> @erwango, do you have any insight into what we could do here? We could add a workaround in the SD subsystem to allow the user to specify an initialization frequency manually, but I'd prefer to avoid this if possible.

Unfortunately, there is no way to change SPI clock other than APB2 prescaler. So maybe this can be done dynamically, but this is not possible in tree today.
This being said this is a specific case, and I'm not sure that it is a reason to deviate from spec in the SD driver.

Dgthhaivu, 2023-10-10:
Thank you both for clarifying my queries.

Rohan-RSA, 2025-05-29:
Hello @Dgthhaivu. I am running into the same/similar issue. What have you done to solve this?