zephyrproject-rtos/zephyr #63319: stm32: vref sensor conflicts with uart

TilmannUnte, 2023-09-29:
**Describe the bug**
In https://github.com/zephyrproject-rtos/zephyr/commit/4dc16527fd0795b85e7b064176a1dd3c470fa97d vref and vbat sensor functionality was added to and activated by default on many stm32 boards.
On disco_l475_iot1 this feature conflicts with UART4 (at least), also active by default.
The UART no longer transmits any data, without any indication as to what is wrong.

vref and vbat need to be set to disabled in the devicetree for functionality to return.
Alternatively CONFIG_SENSOR=n can be used, however this will break applications relying on any sensors.

**To Reproduce**
1. clone my test program: https://github.com/TilmannUnte/vref_uart_test
2. west build -p always -b disco_l475_iot1
3. west flash
4. Connect jumper wires from Pins D0->D10 and Pins D1->D4
5. screen /dev/ttyACM0 115200 and press Reset Button
6. (Optional) Experiment with workarounds in Devicetree Overlay or KConfig

**Expected behavior**
The program should report:
> [...]
> Device serial@40004c00 sent: "Hello from device serial@40004c00, num 0"
> Device serial@40004400 received: "Hello from device serial@40004c00, num 0"

Zephyr should not have conflicting default settings and ideally should warn about them.

**Impact**
Now that I know what the issue is, it's easily dealt with.
However, tracking down the problem was annoying, because it was not clear where it originates.
Many more boards could be affected!

**Environment (please complete the following information):**
 - OS: Ubuntu Linux 22.04 LTS
 - Zephyr SDK 0.16.1
 - Commit 3168623abb

erwango, 2023-10-02:
I'm not sure exactly where the conflict happens (pins maybe), but there is one thing that bother me which is the fact that Vref sensing happens directly when enabling CONFIG_SENSOR.
At least it should require enabling ADC drive, which is not the case because of: 
https://github.com/zephyrproject-rtos/zephyr/blob/0badef8091d737e465f75c8a1c011c9bb996bf77/drivers/sensor/stm32_vref/Kconfig#L10
which should rather be 
https://github.com/zephyrproject-rtos/zephyr/blob/0badef8091d737e465f75c8a1c011c9bb996bf77/drivers/sensor/stm32_vbat/Kconfig#L10

This would be a first fix to do.

Then, assuming that conflict is a issue of sharing of limited hw resources, there is not much that can be done. For those who need ADC + SENSOR + UART4, following remains an option:
```
&vref {
	status = "disabled";
};
```

Note that `disco_l475_iot1.dts` is only a default board configuration which aims at:
- demonstrating most of the board capabilities
- On board CI non regression on test bench

When using in and application, it expected for user to tune the configuratio (using overlay) to satisfy application needs.

TilmannUnte, 2023-10-02:
> [T]here is one thing that bother me which is the fact that Vref sensing happens directly when enabling CONFIG_SENSOR. At least it should require enabling ADC drive, which is not the case because [...]

Thank you for taking a closer look!
I thought I might have missed something regarding KConfig, but it seems now that indeed the current state is not ideal.

I agree with you that proper configuration is of course left to the user.
I'm assuming a check for hardware conflicts is out of scope for Zephyr and/or the SDK?

erwango, 2023-10-03:
> I'm assuming a check for hardware conflicts is out of scope for Zephyr and/or the SDK?

We should be able to arrange the most obvious and easy to fix in order to be able to be compatible with most use cases (samples, tests). But there can't be a requirement stating: "no conflicts are allowed in the default configuration".
On a case by case basis, conflicts could be dealt with .overlay files in samples/tests if need be.

erwango, 2023-10-03:
@TilmannUnte How did you conclude this was a conflict with vref ? Gt bisect to 4dc16527fd0795b85e7b064176a1dd3c470fa97d ?

FRASTM, 2023-10-05:
The pb you report here, @TilmannUnte  is a pin conflict on PA0: 
the uart0: &arduino_serial is the uart4  and 
&uart4 {
	pinctrl-0 = <&**uart4_tx_pa0** &uart4_rx_pa1>;

on the other hand, by activating the SENSOR, it enables the ADC1 which also has PA0 for its input5
&adc1 {
	pinctrl-0 = <&adc1_in5_**pa0**

closed 2023-10-11 by fabiobaltieri

TilmannUnte, 2023-10-13:
Sorry for being quiet, I was traveling.

@erwango 
Yes, I used git bisect and an application I'm working on. I was then confused about the origin of the problem, because I couldn't reproduce it with zephyr samples and vref seemed unrelated to anything I was using. This led to me taking pieces of my application and putting them into the native_tty sample.

@FRASTM 
Thank you for taking a closer look and proposing a fix!