zephyrproject-rtos/zephyr #63322: NXP - MIMXRT1050 Ethernet fix (HAL_NXP)

ShahinHaque97-CT, 2023-09-29:
Hello,

Currently Ethernet does not work on the IMX RT1050, I have a fix that requires a change in HAL_NXP. However, it would not let me post an issue there. 

Create Project
	west init rt1050EthSupport
	cd rt1050EthSupport
	west update

Load DHCP example project
	cd ..\rt1050EthSupport\zephyr\samples\net\dhcpv4_client

Flash
	west build -b mimxrt1050_evk -p

Console reply

```
*** Booting Zephyr OS build zephyr-v3.4.0-4747-g23ed21c0a028 ***
[00:00:00.500,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.500,000] <inf> net_dhcpv4_client_sample: Start on ethernet@402d8000: index=1
[00:00:00.501,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
uart:~$
```

No further responses

-----------------------------

Change IGPIO module features from 1 to 0 in ..\rt1050EthSupport\modules\hal\nxp\mcux\mcux-sdk\devices\MIMXRT1052\MIMXRT1052_features.h

/* IGPIO module features */

```
/* @brief Has data register set DR_SET. */
#define FSL_FEATURE_IGPIO_HAS_DR_SET (0)
/* @brief Has data register clear DR_CLEAR. */
#define FSL_FEATURE_IGPIO_HAS_DR_CLEAR (0)
/* @brief Has data register toggle DR_TOGGLE. */
#define FSL_FEATURE_IGPIO_HAS_DR_TOGGLE (0)
```

Console reply

```
*** Booting Zephyr OS build zephyr-v3.4.0-4747-g23ed21c0a028 ***
[00:00:00.500,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.500,000] <inf> net_dhcpv4_client_sample: Start on ethernet@402d8000: index=1
[00:00:02.501,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:07.535,000] <inf> net_dhcpv4: Received: 192.168.0.49
[00:00:07.535,000] <inf> net_dhcpv4_client_sample:    Address[1]: 192.168.0.49
[00:00:07.535,000] <inf> net_dhcpv4_client_sample:     Subnet[1]: 255.255.255.0
[00:00:07.535,000] <inf> net_dhcpv4_client_sample:     Router[1]: 192.168.0.1
[00:00:07.535,000] <inf> net_dhcpv4_client_sample: Lease time[1]: 86400 seconds
uart:~$
```

ShahinHaque97-CT, 2023-09-29:
@danieldegrasse Is this something you can review?

robertinant, 2023-09-29:
Thanks for reporting this. This will save me quite a bit of time figuring out what the problem is. Looking at the history of that file it has been like it is in the current version from the start. So either Ethernet never worked or something else changed that affects these bits. I might have some time later today to look into this and will post here if I find something.

robertinant, 2023-09-29:
Alright, with a mimxrt1060_evkb, nxp_hal @ ad142f5612d927e29b1f9606e8edade871b8a526 and Zephyr @ v3.4.0-4757-ga6f8935e4fca the out of the box samples/net/dhcpv4_client works like a charm. I looked in our stash of boards to see if I could find a RT1050 but unfortunately I came up empty. Given that I don't have the hardware it will be hard from me to reproduce / fix. I'll ask around on Monday if anybody has a RT1050 board.. As a first I would give the latest Zephyr a try just in case. I'll try the other version of the 1060 board (there are quite a few differences and it has a different dts) and see what that tells me.

```
*** Booting Zephyr OS build zephyr-v3.4.0-4757-ga6f8935e4fca ***
[00:00:00.501,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.501,000] <inf> net_dhcpv4_client_sample: Start on ethernet@402d8000: index=1
[00:00:02.502,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:02.531,000] <inf> net_dhcpv4: Received: 10.0.0.173
[00:00:02.531,000] <inf> net_dhcpv4_client_sample:    Address[1]: 10.0.0.173
[00:00:02.531,000] <inf> net_dhcpv4_client_sample:     Subnet[1]: 255.255.255.0
[00:00:02.531,000] <inf> net_dhcpv4_client_sample:     Router[1]: 10.0.0.1
[00:00:02.531,000] <inf> net_dhcpv4_client_sample: Lease time[1]: 172800 seconds
```

ShahinHaque97-CT, 2023-09-30:
@robertinant 

I believe this issue is isolated to the RT1050 as I believe it is untested, Unfortunately thought github/discord there isn'tany mention of users using the RT1050 for ethernet projects, however many are using a variant of the RT1060

robertinant, 2023-10-02:
I do have a contact at NXP that might be able to help. I'll reach out and see who over there is handling Zephyr.