zephyrproject-rtos/zephyr #63336: ESP32-S3-DevkitC not support settings

sam131208, 2023-09-30:
*Describe the bug**
I build and flash "samples/bluetooth/peripheral_hids" to ESP32-S3-DevkitC-1 board( esp32-s3-wroom-2  D3N32R8V).
While it run and boot to zephyr os, get the message "<err> bt_settings: settings_subsys_init failed (err -45)
".

**To Reproduce**
Steps to reproduce the behavior:
1. update to last zephyr
2. cd zephyr
3. west build -b esp32s3_devkitm  samples/bluetooth/peripheral_hids
4. west flash
5. run

**Expected behavior**
Identify the flash capacity and run the settings normally.

**Impact**
Inability to store, recover critical information.

**Logs and console output**
I (62) boot: ESP-IDF 635063971 2nd stage bootloader
I (62) boot: compile time 15:26:43
I (62) boot: chip revision: 0
I (64) boot.esp32s3: Boot SPI Speed : 40MHz
I (69) boot.esp32s3: SPI Mode       : SLOW READ
I (74) boot.esp32s3: SPI Flash Size : 2MB
I (78) boot: Enabling RNG early entropy source...
I (84) boot: Partition Table:
I (87) boot: ## Label            Usage          Type ST Offset   Length
I (95) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (102) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (110) boot:  2 factory          factory app      00 00 00010000 00100000
I (117) boot: End of partition table
I (122) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28)                                                                             
I (130) esp_image: segment 1: paddr=00010044 vaddr=3fc9d100 size=00ebch (  3772)                                                                              load
I (140) esp_image: segment 2: paddr=00010f08 vaddr=3fc9dfbc size=00478h (  1144)                                                                              load
I (147) esp_image: segment 3: paddr=00011388 vaddr=40374000 size=09140h ( 37184)                                                                              load
I (170) esp_image: segment 4: paddr=0001a4d0 vaddr=00000000 size=05b68h ( 23400)                                                                             
I (179) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=06b60h ( 27488)                                                                              map
I (190) esp_image: segment 6: paddr=00026ba8 vaddr=00000000 size=09450h ( 37968)                                                                             
I (204) esp_image: segment 7: paddr=00030000 vaddr=42030000 size=24674h (149108)                                                                              map
I (262) boot: Loaded app from partition at offset 0x10000
I (262) boot: Disabling RNG early entropy source...
*** Booting Zephyr OS build zephyr-v3.4.0-4777-ga956f7aef10b ***
[00:00:00.299,000] <err> bt_settings: settings_subsys_init failed (err -45)
Bluetooth init failed (err -45)

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain:  Zephyr SDK 0.16.1
 - last version zephyr
 - ESP32-S3-DevkitC-1 board ( esp32-s3-wroom-2  D3N32R8V)

sylvioalves, 2023-10-02:
@sam131208, thanks for spotting this. Will submit a fix.

sylvioalves, 2023-10-11:
@sam131208, would you take a look on this?

sam131208, 2023-10-11:
@sylvioalves Yes, after updating to the latest version of Zephyr, it already works. Thank you!

PR #63783 (2023-10-13): west.yml: update esp32s3 to enable all flash ranges
Current hal only enable up to 16MB flash size in esp32s3 SoC.  
This updates hal_espressif to include 32MB, 64MB and 128MB.

This also fixes "random.h" include header.

Fixes #63336
closed 2023-10-11 by sam131208