zephyrproject-rtos/zephyr #63364: Retention: retention_is_valid is missing check on disabled mutexes

Laczen, 2023-10-02:
**Describe the bug**
The retention subsystem has the possibility to disable the mutex using the config value `CONFIG_RETENTION_MUTEX_FORCE_DISABLE` when multithreading is enabled. In this case the validity of the checksum can no longer be guaranteed.

The retention subsystem `retention_is_valid` routine should report `-ENOTSUP` when the option `CONFIG_RETENTION_MUTEX_FORCE_DISABLE` is set.

nordicjm, 2023-10-02:
Not true, this is application defined behaviour. A bootloader might set a region up (i.e. write only), an application might check a region (i.e. read only), therefore mutex support for retention subsystem is not needed, the application can therefore safely check if the data is valid. That would not be the case for an application which has multiple threads all reading and writing to a retention area, but it does not pertain to all applications thus is left to the user to decide.

Laczen, 2023-10-02:
> Not true, this is application defined behaviour. A bootloader might set a region up (i.e. write only), an application might check a region (i.e. read only), therefore mutex support for retention [...]

Your first example is a very rare case and this exception should be part of `blinfo_mcuboot`.

The usecase for `CONFIG_RETENTION_MUTEX_FORCE_DISABLE` according to the documentation is for use in an ISR. In this case the `retention_is_valid` function might return 'valid' on non-valid data, to avoid that it is needed to return `-ENOTSUP` when the mutex is disabled.

nordicjm, 2023-10-02:
> Your first example is a very rare case

How have you determined this, have you ran a survey to quiz the users of this subsystem to determine it is true? Audited open source projects that use it?

Laczen, 2023-10-02:
> > Your first example is a very rare case
> 
> How have you determined this, have you ran a survey to quiz the users of this subsystem to determine it is true? Audited open source projects that use it?

Simple, the naming of the system and its documentation implies a broad usage.

I don't understand the problem, it is a simple correction. It would also be possible to specify this per retention area (in the dts) but this would be a lot more effort.

jhedberg, 2023-10-10:
I think it makes more sense to close an issue as "not planned" rather than using a "Won't Fix" label. This can be reopened if necessary.

Laczen, 2023-10-10:
I have put in on my planned list.

jhedberg, 2023-10-10:
> I have put in on my planned list.

Then it needs a proper priority rather than "Won't Fix". I've fixed the labels now.

closed 2023-12-24 by github-actions[bot]