zephyrproject-rtos/zephyr #63369: USB boot hang

astonkov, 2023-10-02:
**Describe the bug**
I have a SAMR34 XPLAINED PRO board, and I'm trying to bring up USB. I've built the usb_cdc example using the west build command, but I've encountered an issue where the program seems to hang at the line of code https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/usb/device/usb_dc_sam0.c#L151:

Additionally to [board code](https://github.com/zephyrproject-rtos/zephyr/tree/14348c53d41c5f080648b71f8fa1817b0da6b304/boards/arm/atsamr34_xpro), I've made some modifications to the device tree files (dts and dtsi) to configure the USB pins as follows:

In .[dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/14348c53d41c5f080648b71f8fa1817b0da6b304/boards/arm/atsamr34_xpro/atsamr34_xpro-pinctrl.dtsi) file:
```
usb_dc_default: usb_dc_default {
    group1 {
        pinmux = <PA25G_USB_DP>,
                 <PA24G_USB_DM>;
    };
};
```

In .[dts](https://github.com/zephyrproject-rtos/zephyr/blob/14348c53d41c5f080648b71f8fa1817b0da6b304/boards/arm/atsamr34_xpro/atsamr34_xpro.dts) file:
```
zephyr_udc0: &usb0 {
    status = "okay";
    pinctrl-0 = <&usb_dc_default>;
    pinctrl-names = "default";
};
```
I inserted and removed the USB, but it doesn't solve the problem. I also tried the example for the board from Microchip Studio USB_CDC, it works correctly and initializes USB.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b atsamr34_xpro zephyr/samples/subsys/usb/cdc_acm
2. openocd -f board/atmel_saml21_xplained_pro.cfg
3. gdb-multiarch -iex "target extended-remote localhost:3333" ./build/zephyr/zephyr.elf 
4. load
5. program hang

**Expected behavior**
Connection Establishment with host computer

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.2 LTS
 - Zephyr SDK 0.16.0
 - v3.4-branch / 3e75c22

jfischer-no, 2023-10-02:
Quickly tested on atsamr21_xpro, seems the driver works fine, maybe board configuration issue, clock or something else.

attie-argentum, 2023-10-03:
I can confirm this, but unfortunately I don't have a lot of time to look into this right now. Strange, because I thought this was working when I did support for this SoC...

Backtrace from 54efc02:

```
(gdb) bt
#0  usb_sam0_wait_syncbusy () at /p/zephyr/drivers/usb/device/usb_dc_sam0.c:151
#1  0x000042c4 in usb_dc_attach () at /p/zephyr/drivers/usb/device/usb_dc_sam0.c:248
#2  0x00002bdc in usb_enable (status_cb=status_cb@entry=warning: (Internal error: pc 0x0 in read in CU, but not in symtab.)
warning: (Error: pc 0x0 in address map, but not in symtab.)
0x0 <_vector_table>) at /p/zephyr/subsys/usb/device/usb_device.c:1622
#3  0x00000650 in main () at /p/zephyr/samples/subsys/usb/cdc_acm/src/main.c:168
```

https://github.com/zephyrproject-rtos/zephyr/blob/54efc028ec366dffc3cffccdb05f96e1aefa346d/drivers/usb/device/usb_dc_sam0.c#L246-L248

---

I wonder if it might be a power or clock config issue - but I'd be surprised, as far as I know the SAMR34 has SAML21 silicon inside (SiP)... I've skimmed the SAMR34/35 datasheet, but can't see anything calling out differences for USB.

https://github.com/zephyrproject-rtos/zephyr/blob/54efc028ec366dffc3cffccdb05f96e1aefa346d/soc/arm/atmel_sam0/common/soc_saml2x.c#L19-L30

attie-argentum, 2023-10-03:
I just tried this on a SAML21-XPro, with the same result on 48c83a1f4268e539afa03abe2ecaed43a0650b7e.

@jfischer-no did you do anything special / any patches?

jfischer-no, 2023-10-03:
> I just tried this on a atsaml21_xpro, with the same result on [48c83a1](https://github.com/zephyrproject-rtos/zephyr/commit/48c83a1f4268e539afa03abe2ecaed43a0650b7e).
> 
> @jfischer-no did you do anything special / any patches?

"Quickly tested on atsam**r**21_xpro, seems the driver works fine"
**R** :smile:

nandojve, 2023-10-07:
I only have samr21 so I can not reproduce on my side.
@attie-argentum if you donˋt have time to look now we need schedule for the next release and try check using loopback app + Linux.

PR #65783 (2023-12-05): Setup USB clock and pinctrl for SAML21-derrived boards
This patch resolves issues with USB Device mode on SAML21-derrived boards, including ATSAML21-XPRO and ATSAMLR34-XPRO (and tested on both with [`.../samples/subsys/usb/hid/`](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/usb/hid))

cc: @astonkov, @nandojve (apologies for the delay!)

Fixes #63369
closed 2023-12-05 by MaureenHelm