zephyrproject-rtos/zephyr #63373: net: lwm2m: notify-messages aren't generated due to duplicate observation and bad timing

hung-gcx, 2023-10-02:
**Describe the bug**
If a resource is observed and if there happens to be duplicate observation(s) on that resource, the resource's content
might not be reported correctly anymore.

**To Reproduce**
- Establish observation on a resource whose content changes regularly (for example: device uptime).
- Have the resource's content reported to the server periodically (e.g. every second) by generating notify messages, either manually or automatically.
The smaller the report interval, the easier it is to reproduce the problem.
- Trigger another observation request on the resource to cause a duplicate observation.

-> The notify-messages won't be generated periodically (e.g. every second) anymore.
What this also means is that, looking from the server/backend's point of view, many values of the resource will be missing.

**Why?**
When `generate_notify_message()` is called, the boolean `active_tx_operation` is set to true. In the meantime, an observation request is received by the device, and the resource's token is updated (while the device hasn't received the corresponding ACK for its notify-message yet). 
Then the ACK comes and is handled by `notify_message_reply_cb()`, but since the token has been updated, `notify_message_reply_cb()` cannot find the observer anymore:
https://github.com/zephyrproject-rtos/zephyr/blob/48c83a1f4268e539afa03abe2ecaed43a0650b7e/subsys/net/lib/lwm2m/lwm2m_message_handling.c#L2893C1-L2907C1
```
        } else {
                obs = engine_observe_node_discover(&msg->ctx->observer, &prev_node, NULL,
                                                   reply->token, reply->tkl);

                if (obs) {
                        obs->active_tx_operation = false;
                        if (msg->ctx->observe_cb) {
                                msg->ctx->observe_cb(LWM2M_OBSERVE_EVENT_NOTIFY_ACK,
                                                     lwm2m_read_first_path_ptr(&obs->path_list),
                                                     reply->user_data);
                        }
                        notify_cached_pending_data_trig(obs);
                }
        }

```
`if (obs)` resolving to false is what we suspect to have caused the issue.

rlubos, 2023-10-03:
It seems to me that a natural consequence of updating the observer token [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_observation.c#L596) or [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/lwm2m/lwm2m_observation.c#L679) should be canceling any ongoing transactions. The question is how to tackle this best. We could simply loop over `lwm2m_ctx->replies` array and try to find a message with a matching token, or we could replace `active_tx_operation` with a `struct lwm2m_message` pointer, so that it could be retrieved from the observer context directly (non-null would mean there is an active transaction). Any thoughts @SeppoTakalo?

rlubos, 2023-10-03:
In a meantime, I've opened a PR with a fix proposal: https://github.com/zephyrproject-rtos/zephyr/pull/63458.

@hung-gcx Could you please give it a try?

hung-gcx, 2023-10-03:
@rlubos I just gave https://github.com/zephyrproject-rtos/zephyr/pull/63458 a spin, seems to have solved it (:
One titbit, not sure if relevant:
- The value right before the (additional/duplicate) observation request is successfully reported to the server in a `CON` message (say the value is 300).
- Observation request comes in, device responds with and `ACK` containing the previous value (300)
-> The same value is reported twice, but I'd say it's expected behavior.

rlubos, 2023-10-03:
> @rlubos I just gave #63458 a spin, seems to have solved it (:

Glad to hear that! 

> One titbit, not sure if relevant:
>
> * The value right before the (additional/duplicate) observation request is successfully reported to the server in a `CON` message (say the value is 300).
> * [...]

If the observed resource value did not change inbetween notification/new observe request, then I'd say there's nothing wrong here. The device **has to** respond to the observation request even if the resource value is still the same.

PR #63458 (2023-10-04): net: lwm2m: Reset ongoing notification in case token changes
In case observation token changes (the LwM2M server re-sends observation request to the client), the LwM2M engine should cancel any ongoing notifications based on the old token. Otherwise, it will be impossible to match the pending notification reply (ACK) with the observer context anymore, causing new notifications for this observation to stall.

Fixes #63373
closed 2023-10-04 by carlescufi