zephyrproject-rtos/zephyr #63414: gd32l233r_eval fails to build sensor_shell sample with async

yperess, 2023-10-02:
During the introduction of https://github.com/zephyrproject-rtos/zephyr/pull/60063 the CI was failing to build `gd32l233r_eval` with an assembler error which I believe is a part of the toolchain (see below for the build log). To replicate, simply remove the `gd32l233r_eval` from the excluded platforms list.

```
INFO    - 166/671 gd32l233r_eval            samples/sensor/sensor_shell/sample.sensor.shell     ERROR Build failure (build)
INFO    - /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/build.log
ERROR   - Loading Zephyr module(s) (Zephyr base): zephyr_default:kconfig
-- Application: /__w/zephyr/zephyr/samples/sensor/sensor_shell
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /github/home/.cache/zephyr
-- Zephyr version: 3.5.0-rc1 (/__w/zephyr/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: gd32l233r_eval
-- Found host-tools: zephyr 0.16.3 (/opt/toolchains/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (/opt/toolchains/zephyr-sdk-0.16.3)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/arm/gd32l233r_eval/gd32l233r_eval.dts
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/dts.cmake
Parsing /__w/zephyr/zephyr/samples/sensor/sensor_shell/Kconfig
Loaded configuration '/__w/zephyr/zephyr/boards/arm/gd32l233r_eval/gd32l233r_eval_defconfig'
Merged configuration '/__w/zephyr/zephyr/samples/sensor/sensor_shell/prj.conf'
Merged configuration '/__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/misc/generated/extra_kconfig_options.conf'
Configuration saved to '/__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/.config'
Kconfig header saved to '/__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/include/generated/autoconf.h'
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/zephyr/zephyr/samples/sensor/sensor_shell
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /github/home/.cache/zephyr
-- Zephyr version: 3.5.0-rc1 (/__w/zephyr/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: gd32l233r_eval
-- Found host-tools: zephyr 0.16.3 (/opt/toolchains/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (/opt/toolchains/zephyr-sdk-0.16.3)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/arm/gd32l233r_eval/gd32l233r_eval.dts
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/dts.cmake
Parsing /__w/zephyr/zephyr/samples/sensor/sensor_shell/Kconfig
Loaded configuration '/__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/.config'
No change to configuration in '/__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/.config'
No change to Kconfig header in '/__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /opt/toolchains/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    TC_RUNID

-- Build files have been written to: /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell
[1/156] Preparing syscall dependency handling

[2/156] Generating include/generated/version.h
-- Zephyr version: 3.5.0-rc1 (/__w/zephyr/zephyr), build: v3.5.0-rc1-33-gac6ddeb4a2ea
[3/156] Generating misc/generated/syscalls_subdirs.trigger
[4/156] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[5/156] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
[6/156] Generating include/generated/driver-validation.h
[7/156] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[8/156] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[9/156] Generating include/generated/offsets.h
[10/156] Building C object CMakeFiles/app.dir/src/main.c.obj
[... 103 lines cut ...]
[110/156] Building C object modules/hal_gigadevice/CMakeFiles/hal_gigadevice.dir/__w/zephyr/modules/hal/gigadevice/gd32l23x/cmsis/gd/gd32l23x/source/system_gd32l23x.c.obj
[111/156] Building C object zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/cortex_m_systick.c.obj
[112/156] Linking C static library zephyr/drivers/serial/libdrivers__serial.a
[113/156] Linking C static library zephyr/subsys/rtio/libsubsys__rtio.a
[114/156] Building C object modules/hal_gigadevice/CMakeFiles/hal_gigadevice.dir/__w/zephyr/modules/hal/gigadevice/gd32l23x/standard_peripheral/source/gd32l23x_gpio.c.obj
[115/156] Building C object zephyr/kernel/CMakeFiles/kernel.dir/banner.c.obj
[116/156] Building C object zephyr/kernel/CMakeFiles/kernel.dir/device.c.obj
[117/156] Building C object modules/hal_gigadevice/CMakeFiles/hal_gigadevice.dir/__w/zephyr/modules/hal/gigadevice/gd32l23x/standard_peripheral/source/gd32l23x_rcu.c.obj
[118/156] Building C object modules/hal_gigadevice/CMakeFiles/hal_gigadevice.dir/__w/zephyr/modules/hal/gigadevice/gd32l23x/standard_peripheral/source/gd32l23x_usart.c.obj
[119/156] Building C object zephyr/kernel/CMakeFiles/kernel.dir/errno.c.obj
[120/156] Building C object zephyr/kernel/CMakeFiles/kernel.dir/main_weak.c.obj
[121/156] Building C object zephyr/kernel/CMakeFiles/kernel.dir/busy_wait.c.obj
[122/156] Building C object zephyr/drivers/sensor/CMakeFiles/drivers__sensor.dir/sensor_shell.c.obj
[123/156] Building C object zephyr/kernel/CMakeFiles/kernel.dir/kheap.c.obj
[124/156] Building C object zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj
[125/156] Building C object zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj
[126/156] Building C object zephyr/drivers/sensor/CMakeFiles/drivers__sensor.dir/default_rtio_sensor.c.obj
FAILED: zephyr/drivers/sensor/CMakeFiles/drivers__sensor.dir/default_rtio_sensor.c.obj 
ccache /opt/toolchains/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DGD32L233 -DIRC32K_VALUE=32000 -DKERNEL -DPICOLIBC_DOUBLE_PRINTF_SCANF -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/__w/zephyr/zephyr/include -I/__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/zephyr/include/generated -I/__w/zephyr/zephyr/soc/arm/gigadevice/gd32l23x -I/__w/zephyr/zephyr/lib/posix/getopt/. -I/__w/zephyr/zephyr/soc/arm/gigadevice/common/. -I/__w/zephyr/zephyr/soc/arm/gigadevice/gd32l23x/. -I/__w/zephyr/zephyr/subsys/rtio -I/__w/zephyr/modules/hal/cmsis/CMSIS/Core/Include -I/__w/zephyr/zephyr/modules/cmsis/. -I/__w/zephyr/modules/hal/ [... cut ...] MakeFiles/drivers__sensor.dir/default_rtio_sensor.c.obj -c /__w/zephyr/zephyr/drivers/sensor/default_rtio_sensor.c
/tmp/ccnw8Ls2.s: Assembler messages:
/tmp/ccnw8Ls2.s:[275](https://github.com/zephyrproject-rtos/zephyr/actions/runs/6383134587/job/17323294466?pr=60063#step:13:276)8: Error: only lo regs allowed with immediate -- `mov ip,#0'
ninja: build stopped: subcommand failed.

INFO    - /__w/zephyr/zephyr/twister-out/gd32l233r_eval/samples/sensor/sensor_shell/sample.sensor.shell/build.log
```

fabiobaltieri, 2023-10-03:
@microbuilder could you take a look into this?

fabiobaltieri, 2023-10-03:
cc @carlocaione @ithinuel

fabiobaltieri, 2023-10-03:
Actually this (`west build -p -b gd32l233r_eval -T samples/sensor/sensor_shell/sample.sensor.shell`) bisects down to https://github.com/zephyrproject-rtos/zephyr/pull/63204, is there something wrong with that SoC setup? cc @ithinuel

ithinuel, 2023-10-03:
No, it is an overlook in that previous PR. I should've also updated the constraint on `tmp`. `mov` on armv8-m.baseline, when used with an immediate value is limited to registers `r0` to `r7`.
`=r` tells the compiler any register can be used. `=l` needs to be used instead.
I have a PR incoming to fix this.

_#63204 may have subtly broken other part of the kernel for armv8-m.baseline._
_While working on something else, I noticed that sometimes interrupts are masked directly instead of going through `arch_irq_lock` (and conversely for the unlock)._

fabiobaltieri, 2023-10-03:
coo, thanks @ithinuel, assigning back to you then

PR #63854 (2023-10-13): arch: arm: cortex_m: Fix constraint on inline assembly for armv8-m.baseline
`mov` used with an immediate value on armv8-m.baseline is limited to registers `r0` to `r7`. `=r` tells the compiler any register can be used. `=l` needs to be used instead.

Fixes: #63414