zephyrproject-rtos/zephyr #63453: Compile error

alikovvl, 2023-10-03:
zephyr/include/zephyr/kernel.h:712:9: error: designator order for field '_static_thread_data::init_name' does not match declaration order in '_static_thread_data'

ycsin, 2023-10-03:
@alikovvl could you provide the full build log?
cc @keith-packard

fabiobaltieri, 2023-10-03:
Hi, could you change the issue description so that it uses the [bug report template](https://github.com/zephyrproject-rtos/zephyr/issues/new?template=001_bug_report.md)? It asks for some important information that are very useful for us to understand the issue. Thanks.

keith-packard, 2023-10-03:
I think I understand the problem -- the indicated patch changes the order of members of the _static_thread_data struct to avoid inserting padding before the 8-byte init_delay member. C++ appears to complain about a struct initialization where the order of initializers doesn't match the struct order. Which is silly as that's kinda the point of named initializers, but one doesn't generally get far arguing with the compiler.

Closed by: #63468

PR #63468 (2023-10-03): kernel: Re-order _static_thread_data initializer to match type
C++ compilers complain if the order of designated initializers doesn't match the order of the members in the type. Re-order the initializers for the _static_thread_data struct to match the new order of the type.

Closes: #63453
closed 2023-10-03 by fabiobaltieri