zephyrproject-rtos/zephyr #63488: Filter for libcxx tests use TOOLCHAIN_HAS_PICOLIBC rather than CONFIG_PICOLIBC_SUPPORTED

alpsayin, 2023-10-03:
https://github.com/zephyrproject-rtos/zephyr/blob/99ab1ab06cda167a633c4047d327141a19cbafaa/tests/lib/cpp/libcxx/testcase.yaml#L27

All other `picolibc` tests use the config flag. Is this an issue?

henrikbrixandersen, 2023-10-04:
If you are reporting a bug, please use our bug template (You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md).

alpsayin, 2023-10-04:
> If you are reporting a bug, please use our bug template (You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md).

My bad, this is what I get for using "Create a new issue referencing this line". Github clearly ignores the template.

keith-packard, 2023-10-04:
Hrm. Any C++ tests using the C++ standard library from the toolchain must also use the C library from the toolchain, and that's what this test is trying to check for. Maybe this check should *also* include `PICOLIBC_SUPPORTED`? That would have seemed redundant when this was written as the set of targets with picolibc support was greater than the set of targets with picolibc in the SDK

alpsayin, 2023-10-09:
I genuinely can't comment. It's a minor detail, and I only brought it to attention due to seeming inconsistency. 
Another way to look at this is: as of this moment `sdk-ng` always marks `TOOLCHAIN_HAS_PICOLIBC` so it's akin to not having a filter.

keith-packard, 2023-10-10:
yes, and sdk-ng also always supports picolibc. But presumably some other toolchain wouldn't set this? Like clang?

closed 2023-12-24 by github-actions[bot]