zephyrproject-rtos/zephyr #63493: call malloc() failed from a userspace thread with zephyr qemu_x86_64

mengyaliu, 2023-10-04:
**Description**
1. When calling malloc() from a userspace thread, got an exception of permission fail, with binary built for target qemu_x86_64.
2. Run my test on main branch with version: d489ec087ed5a0b4a8c104e2f5fb9a302419d96b
3. After some debugging, it turns out the exception caused by visiting data structure **z_malloc_heap** in zephyr/lib/libc/common/source/stdlib/malloc.c:144.
4. After checking zephyr.map file, it turns out the z_malloc_partition built correctly, both z_malloc_heap & malloc_arena are included:
`z_data_smem_z_malloc_partition_part_size = (z_data_smem_z_malloc_partition_part_end -                                  z_data_smem_z_malloc_partition_part_start)`
5. But in the function malloc_prepare() from zephyr/lib/libc/common/source/stdlib/malloc.c, the z_malloc_parttion is wrongly changed, which will lead the z_malloc_heap not included by the partition.
6. After the apply following change, malloc() can be successfully called.
```
diff --git a/lib/libc/common/source/stdlib/malloc.c b/lib/libc/common/source/stdlib/malloc.c
index a98b3e234d..591667f70b 100644
--- a/lib/libc/common/source/stdlib/malloc.c
+++ b/lib/libc/common/source/stdlib/malloc.c
@@ -225,12 +225,6 @@ static int malloc_prepare(void)
        heap_size = HEAP_SIZE;
 #endif

-#if Z_MALLOC_PARTITION_EXISTS
-       z_malloc_partition.start = POINTER_TO_UINT(heap_base);
-       z_malloc_partition.size = heap_size;
-       z_malloc_partition.attr = K_MEM_PARTITION_P_RW_U_RW;
-#endif
-
        sys_heap_init(&z_malloc_heap, heap_base, heap_size);

        return 0;
```

**To Reproduce**
1. my configuration:
```
CONFIG_PICOLIBC=y
CONFIG_POSIX_API=y
CONFIG_POSIX_MQUEUE=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
# CONFIG_GETOPT is not set
# CONFIG_CPP is not set
CONFIG_MAIN_STACK_SIZE=2048
# CONFIG_NEWLIB_LIBC is not set
CONFIG_SNTP=y
CONFIG_NET_CONFIG_CLOCK_SNTP_INIT=y
CONFIG_NET_CONFIG_SNTP_INIT_SERVER="au.pool.ntp.org"
CONFIG_FILE_SYSTEM=y
CONFIG_POSIX_FS=y
CONFIG_PIPES=y
CONFIG_POLL=y
CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS=y
CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE=67108864
CONFIG_SRAM_SIZE=524288
CONFIG_MAX_PTHREAD_MUTEX_COUNT=100
CONFIG_MSG_COUNT_MAX=128
CONFIG_MSG_SIZE_MAX=512
CONFIG_MQUEUE_NAMELEN_MAX=64
CONFIG_POSIX_MAX_OPEN_FILES=1024
CONFIG_MAX_PTHREAD_COUNT=100
CONFIG_MAX_THREAD_BYTES=24
CONFIG_FLASH_SIZE=262144
CONFIG_FLASH_BASE_ADDRESS=0x0
CONFIG_KERNEL_VM_SIZE=0x20000000
CONFIG_HEAP_MEM_POOL_SIZE=10240

CONFIG_X86_EXTRA_PAGE_TABLE_PAGES=1
CONFIG_FILE_SYSTEM=y
CONFIG_DISK_ACCESS=y
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_DISK_DRIVER_FLASH=y
CONFIG_CONSOLE_SUBSYS=y
CONFIG_CONSOLE_GETCHAR=y
CONFIG_USERSPACE=y
```
2. my unit test:
```
#include <stdio.h>
#include <stdlib.h>

#include <zephyr/kernel.h>
#include <zephyr/app_memory/app_memdomain.h>
#include <zephyr/sys/libc-hooks.h>

#define USER_STACKSIZE 2048
static struct k_mem_domain test_domain;

struct k_thread test_thread;
K_THREAD_STACK_DEFINE(test_stack, USER_STACKSIZE);

static void do_test(void *p1, void *p2, void *p3)
{
  printk("malloc address: %p\n", malloc(4));
  printk("test libc done\n");
}

static void test_libc(void *p1, void *p2, void *p3)
{
  printk("test libc started...\n");
  k_mem_domain_add_thread(&test_domain, k_current_get());
  k_thread_user_mode_enter(do_test, NULL, NULL, NULL);
}

int main(void)
{
  struct k_mem_partition *parts[] = {
#ifdef Z_MALLOC_PARTITION_EXISTS
    &z_malloc_partition,
    &z_libc_partition,
#endif
  };

  k_mem_domain_init(&test_domain, ARRAY_SIZE(parts), parts);

  k_thread_create(&test_thread, test_stack, USER_STACKSIZE,
                  test_libc, NULL, NULL, NULL,
                  -1, 0, K_MSEC(80));
}
```

**Environment (please complete the following information):**
 - OS: docker image: ghcr.io/zephyrproject-rtos/zephyr-build
 - Toolchain: /opt/toolchains/zephyr-sdk-0.16.1/sysroots/x86_64-pokysdk-linux/
 - Commit SHA or Version used: d489ec087ed5a0b4a8c104e2f5fb9a302419d96b

mengyaliu, 2023-10-04:
1.  Following change is just to illustrate the bug, and this is absolutely not a solution.
```
diff --git a/lib/libc/common/source/stdlib/malloc.c b/lib/libc/common/source/stdlib/malloc.c
index a98b3e234d..591667f70b 100644
--- a/lib/libc/common/source/stdlib/malloc.c
+++ b/lib/libc/common/source/stdlib/malloc.c
@@ -225,12 +225,6 @@ static int malloc_prepare(void)
        heap_size = HEAP_SIZE;
 #endif

-#if Z_MALLOC_PARTITION_EXISTS
-       z_malloc_partition.start = POINTER_TO_UINT(heap_base);
-       z_malloc_partition.size = heap_size;
-       z_malloc_partition.attr = K_MEM_PARTITION_P_RW_U_RW;
-#endif
-
        sys_heap_init(&z_malloc_heap, heap_base, heap_size);

        return 0;
```

2. Consider following related commit, when CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE  is -1, z_malloc_partition should be correctly setup to support all kinds of scenarios. 
```
commit 32a437cbc8cca7fa7be0d286ee71837cd7a10aee
Author: Keith Packard <keithp@keithp.com>
Date:   Wed Mar 29 09:25:35 2023 -0700

    libc/common: Handle MMU/MPU/USERSPACE and alignment requirements in malloc

    When CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE is set to -1, make the malloc
    arena use all remaining RAM. When an MMU exists, allocate the arena at
    startup using k_mem_map. Otherwise, compute the available memory
    automatically and use that.

    When an MPU is being used to manage the malloc arena, make sure the heap
    respects any MPU alignment requirements. Otherwise, align the heap to
    sizeof(double).

    Signed-off-by: Keith Packard <keithp@keithp.com>
```

nordicjm, 2023-10-04:
CC @keith-packard

keith-packard, 2023-10-09:
The fix for this #63543 is ready for review.

PR #63543 (2023-10-10): Move malloc data structures to libc partition
Leave the malloc partition so that it only contains the heap itself; this lets the initialization code adjust the address range when configuring the arena at startup.

This could be done differently -- when the malloc arena is allocated at build time as a static array, we could stick
the malloc variables in the same partition and avoid needing the libc partition. That's a bit more complicated as it would require detecting that condition when deciding whether to create a libc partition ...

Closes: #63493
closed 2023-10-10 by jhedberg