zephyrproject-rtos/zephyr #63500: SOF build broken since commit f30f08a3ce75

lyakh, 2023-10-04:
**Describe the bug**
Since that commit, cloning a fresh Zephyr tree with SOF and trying to build it, e.g. for MTL, now results in:
```
/tmp/west/modules/audio/sof/app/prj.conf:20: warning: attempt to assign the value 'n' to the undefined symbol HAVE_AGENT
/tmp/west/modules/audio/sof/app/prj.conf:37: warning: attempt to assign the value 'n' to the undefined symbol DEBUG_MEMORY_USAGE_SCAN
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:1: warning: attempt to assign the value 'y' to the undefined symbol METEORLAKE
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:2: warning: attempt to assign the value 'y' to the undefined symbol IPC_MAJOR_4
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:5: warning: attempt to assign the value 'n' to the undefined symbol INTEL_DMIC
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:6: warning: attempt to assign the value 'n' to the undefined symbol INTEL_DMIC_NHLT
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:7: warning: attempt to assign the value 'n' to the undefined symbol INTEL_SSP
/tmp/west/modules/audio/sof/appParsing /tmp/west/zephyr/Kconfig
Loaded configuration '/tmp/west/zephyr/boards/xtensa/intel_adsp_ace15_mtpm/intel_adsp_ace15_mtpm_defconfig'
Merged configuration '/tmp/west/modules/audio/sof/app/prj.conf'
Merged configuration '/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf'
Merged configuration '/tmp/west/modules/audio/sof/app/debug_overlay.conf'
/boards/intel_adsp_ace15_mtpm.conf:8: warning: attempt to assign the value 'n' to the undefined symbol INTEL_ALH
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:9: warning: attempt to assign the value 'y' to the undefined symbol COMP_SRC
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:10: warning: attempt to assign the value 'y' to the undefined symbol COMP_SRC_IPC4_FULL_MATRIX
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:11: warning: attempt to assign the value 'y' to the undefined symbol COMP_DRC
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:27: warning: attempt to assign the value 'y' to the undefined symbol ZEPHYR_NATIVE_DRIVERS
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:36: warning: attempt to assign the value 'y' to the undefined symbol ZEPHYR_DP_SCHEDULER
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:41: warning: attempt to assign the value 'y' to the undefined symbol INTEL_MODULES
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:42: warning: attempt to assign the value 'y' to the undefined symbol LIBRARY_MANAGER
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:45: warning: attempt to assign the value 'y' to the undefined symbol AMS
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:49: warning: attempt to assign the value 'y' to the undefined symbol L3_HEAP
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:50: warning: attempt to assign the value '"mtl"' to the undefined symbol RIMAGE_SIGNING_SCHEMA
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:52: warning: attempt to assign the value 'n' to the undefined symbol FORMAT_CONVERT_HIFI3
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:53: warning: attempt to assign the value 'y' to the undefined symbol PCM_CONVERTER_FORMAT_S16_C16_AND_S16_C32
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:54: warning: attempt to assign the value 'y' to the undefined symbol PCM_CONVERTER_FORMAT_S16_C32_AND_S32_C32
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:55: warning: attempt to assign the value 'y' to the undefined symbol PCM_CONVERTER_FORMAT_S16_C32_AND_S24_C32
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:56: warning: attempt to assign the value 'y' to the undefined symbol PCM_CONVERTER_FORMAT_S24_C24_AND_S24_C32
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:57: warning: attempt to assign the value 'y' to the undefined symbol PCM_CONVERTER_FORMAT_S24_C32_AND_S24_C24
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:58: warning: attempt to assign the value 'y' to the undefined symbol PCM_CONVERTER_FORMAT_S16_C32_AND_S16_C32
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:65: warning: attempt to assign the value 'y' to the undefined symbol COMP_VOLUME_WINDOWS_FADE
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:66: warning: attempt to assign the value 'y' to the undefined symbol COMP_UP_DOWN_MIXER
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:67: warning: attempt to assign the value 'y' to the undefined symbol COMP_CHAIN_DMA
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:68: warning: attempt to assign the value 'y' to the undefined symbol SAMPLE_KEYPHRASE
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:73: warning: attempt to assign the value 'y' to the undefined symbol SOF_LOG_LEVEL_INF
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:74: warning: attempt to assign the value 'y' to the undefined symbol ZEPHYR_LOG
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:79: warning: attempt to assign the value 'y' to the undefined symbol LL_WATCHDOG
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:82: warning: attempt to assign the value 'n' to the undefined symbol TRACE
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:83: warning: attempt to assign the value 'y' to the undefined symbol COMP_KPB
/tmp/west/modules/audio/sof/app/boards/intel_adsp_ace15_mtpm.conf:84: warning: attempt to assign the value 'y' to the undefined symbol COMP_ARIA
error: Aborting due to Kconfig warnings
```

**To Reproduce**
```
mkdir /tmp/test
cd /tmp/test
west init
west update
west update sof
west build -d build -b intel_adsp_ace15_mtpm modules/audio/sof/app/
```

**Expected behavior**
Building should succeed

**Impact**
An existing installation stopped working

**Logs and console output**
see above

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK 0.16.3
 - west update as of today, Zephyr commit 89db200117aa4d0604060d65318dcc0ab30b073d

kv2019i, 2023-10-04:
Let me route to @nashif for commit f30f08a3ce75

nashif, 2023-10-10:
you need 

` west config manifest.project-filter -- +sof`

nashif, 2023-10-12:
see https://docs.zephyrproject.org/latest/develop/manifest/index.html#inactive-and-optional-projects-modules

closed 2023-10-12 by nashif