zephyrproject-rtos/zephyr #63504: intel_ehl_crb board does not work with ACPI IRQ allocation

finikorg, 2023-10-04:
Interrupt allocation does not work for `intel_ehl_crb` board. The following code is not working:
https://github.com/zephyrproject-rtos/zephyr/blob/22f09e9fa03fcf88fcc0832857a4d77f8ea0db8c/drivers/pcie/host/pcie.c#L293-L299

since `CONFIG_ACPI` is always enabled. If I remove all this and leave only `irq = arch_irq_allocate();` then interrupt allocation works again.

jhedberg, 2023-10-04:
Would it make sense to fix this with a CONFIG_ACPI_IRQ option that'd be disabled for boards where this is broken?

```
config ACPI_IRQ
	bool "Use ACPI to allocate IRQs"
	default y if ACPI
```

najumon1980, 2023-10-04:
Created a draft PR https://github.com/zephyrproject-rtos/zephyr/pull/63519. The issue might need to address further (like support for shared interrupt etc.) and this PR might be short team resolution for this issue.

jhedberg, 2023-10-04:
> Created a draft PR #63519. The issue might need to address further (like support for shared interrupt etc.) and this PR might be short team resolution for this issue.

Thanks @najumon1980! I added the appropriate "Fixes ..." to it so that the PR and this issue are correctly linked together. Could you please move it out from Draft state so we get full reviews on it? I think it would be good to try to get a fix merged before Friday this week, so that it makes it to rc2.

finikorg, 2023-10-04:
I was thinking that for those boards we might disable ACPI, what are other usages?

PR #63519 (2023-10-06): drivers: pcie: add config support for enable PRT PCI Routing Table
Add config support for enable IRQ information retrieve via ACPI PRT (PCI Routing Table). 

Fixes #63504
closed 2023-10-06 by carlescufi