zephyrproject-rtos/zephyr #63514: IS_ENABLED(CONFIG_BT_CENTRAL) is used instead of conn->role == BT_HCI_ROLE_CENTRAL

Thalley, 2023-10-04:
**Describe the bug**
Several places in the host we have code guarded by `IS_ENABLED(CONFIG_BT_CENTRAL)` intended to only be executed when the device is a central device. The same applies to `IS_ENABLED(CONFIG_BT_PERIPHERAL)`.

However a device can have both `CONFIG_BT_CENTRAL=y` and `CONFIG_BT_PERIPHERAL=y`, and thus it is not enough to guard code with `IS_ENABLED(CONFIG_BT_CENTRAL)` without also checking `conn->role == BT_HCI_ROLE_CENTRAL`.

**To Reproduce**
N/A

**Expected behavior**
Peripheral behavior should not be affected by whether `CONFIG_BT_CENTRAL` is enabled or not. 
Similarly central behavior should not be affected by `CONFIG_BT_PERIPHERAL`

**Impact**
Can cause unexpected behavior that can be hard to track, so all cases where we check if a role is enabled, we should also check whether the `conn` role is the expected role. 

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - Commit SHA or Version used: 89db200117aa4d0604060d65318dcc0ab30b073d

**Additional context**
Just a few examples of behavior that is affected directly by the Kconfig option, rather than the actual role: 
bt_conn_set_state:
```c
	case BT_CONN_CONNECTING:
		if (IS_ENABLED(CONFIG_BT_CENTRAL) &&
		    conn->type == BT_CONN_TYPE_LE) {
			k_work_cancel_delayable(&conn->deferred_work);
		}
		break;
```

bt_conn_disconnect:
```c
	if (IS_ENABLED(CONFIG_BT_CENTRAL) &&
	    conn->type == BT_CONN_TYPE_LE) {
		bt_le_set_auto_conn(&conn->le.dst, NULL);
	}
```

bt_smp_encrypt_change:
```c
	if (IS_ENABLED(CONFIG_BT_CENTRAL) &&
	    IS_ENABLED(CONFIG_BT_PRIVACY) &&
	    !(smp->remote_dist & BT_SMP_DIST_ID_KEY)) {
		uint8_t smp_err;

		smp_err = smp_id_add_replace(smp, conn->le.keys);
		if (smp_err) {
			smp_pairing_complete(smp, smp_err);
		}
	}
```

The last one here is fixed by https://github.com/zephyrproject-rtos/zephyr/pull/63509

carlescufi, 2023-10-06:
I have gone through all uses of `CONFIG_BT_CENTRAL` and `CONFIG_BT_PERIPHERAL` in `subsys/bluetooth/host` and have not been able to find any other misuses. As a result of this I have posted a small PR to document the states:
https://github.com/zephyrproject-rtos/zephyr/pull/63615

I will use that PR to close this issue.

Thalley, 2023-10-06:
> I have gone through all uses of `CONFIG_BT_CENTRAL` and `CONFIG_BT_PERIPHERAL` in `subsys/bluetooth/host` and have not been able to find any other misuses. As a result of this I have posted a small PR to document the states: #63615
> 
> I will use that PR to close this issue.
> 
> @Thalley I have documented the cases you found in the description of this issue

If there not other issues associated it, I think it is OK to not consider this a `bug`, but nevertheless I still think we should apply a `conn->role` check to them all to avoid any future issues as well as optimizing the behavior (i.e. not calling functions when in the "wrong" role). 

Should we modify this issue to an Enhancement, or is it better to create a new GH issue for that?

carlescufi, 2023-10-06:
> If there not other issues associated it, I think it is OK to not consider this a `bug`, but nevertheless I still think we should apply a `conn->role` check to them all to avoid any future issues as well as optimizing the behavior (i.e. not calling functions when in the "wrong" role).

I don't think I agree here. In all cases I've seen the check for the role would've been superfluous due to the nature of the code and flow. If anything in that case I would add assert statements to avoid growing the flash footprint. I honestly doubt this deserves an enhancement at all. The code is designed this way and it does the work it's supposed to do, additional redundant checks are not going to enhance anything IMHO.

Thalley, 2023-10-06:
So you are saying that the following code in `bt_conn_disconnect`
```c
	if (IS_ENABLED(CONFIG_BT_CENTRAL) &&
	    conn->type == BT_CONN_TYPE_LE) {
		bt_le_set_auto_conn(&conn->le.dst, NULL);
	}
```

Is fine to call when the peripheral disconnect an ACL? It is not doing anything that breaks anything, but it does do a few lookups etc. which can be avoided. 

If you do not think verifying the `conn->role` is relavant, do you still think that the checks for `conn->type` is? 

I could also argue that we should add the `conn->role` check for consistency. We have 6 places where we do not check `conn->role` together with `IS_ENABLED(CONFIG_BT_CENTRAL)` where we could, and 25 places where we do (didn't see any places for peripheral where this may happen).

carlescufi, 2023-10-06:
Closing since this is not a bug, an enhancement will be opened by @Thalley.

closed 2023-10-06 by carlescufi