zephyrproject-rtos/zephyr #63535: llext mmu/mpu support

teburd, 2023-10-04:
**Is your enhancement proposal related to a problem? Please describe.**
Systems with memory protection schemes such as the arm mpu or an mmu with permission bits require
some support in llext for elf section placement and permissions.

**Describe the solution you'd like**
ELF sections have correct permissions set on the memory they reside in

.text => RX
.data => RW
.rodata => RO
.bss => RW

So perhaps llext_loader is setup with a set of callbacks that take a memory section and produce a pointer to that section with the appropriate permissions setup. This could be viewed as an allocator+copier+permissions setting call in one.

```c
struct llext_loader {
    void *(*section_init)(struct llext_loader *ldr, struct elf_shdr_t shdr);
}
```

teburd, 2024-01-31:
Fixed by #67857

3Nigma, 2024-02-15:
> Fixed by #67857

Can this be closed, then?

teburd, 2024-02-15:
Yes :-)

closed 2024-02-15 by teburd