zephyrproject-rtos/zephyr #63539: ISO Broadcast Source with 2 BIS has significant packet loss

Thalley, 2023-10-04:
**Describe the bug**
When running the https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/broadcast_audio_source and https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/broadcast_audio_sink samples, there is significant packet loss. The packet loss is oddly enough precisely 60% (i.e. when receiving 10000 HCI ISO packet events, 4000 are valid and 6000 are invalid. This is evident when using e.g. https://github.com/zephyrproject-rtos/zephyr/pull/63537 to check the RX stats

**To Reproduce**
Steps to reproduce the behavior:
2. 
1. Build and flash https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/hci_rpmsg with `-DCONF_FILE=nrf5340_cpunet_iso-bt_ll_sw_split.conf` to the netcores of 2 x nRF5340
2. Build and flash the two samples to the appcores of 2 x nRF5340. 
4. They should automatically works together
5. `ISO receive error` should be very visible in the log

Even if BROADCAST_ENQUEUE_COUNT is updated to 4 and `CONFIG_BT_ISO_TX_BUF_COUNT=8` the issue is visible. 
If reducing the number of streams to 1, then it works flawlessly, so this is clearly an issue for multiple BIS. I have not tested with more than 2. 

**Expected behavior**
Expect the broadcast source and sink samples to work without significant packet loss with the default settings. 

**Impact**
TBD

**Logs and console output**

Extract of an HCI log from the broadcast side. This shows that 8 packets are being enqueued, and that for the most part, the host receive `HCI Event: Number of Completed Packets` switching between handle 2 and 3, which is what is expected. 

The timing of the `HCI Event: Number of Completed Packets` events are not good enough though, as they are clearly not coming every 10ms (on average) as expected. 

Furthermore, events #4851, #4853 and #4855 show that occasionally we do not get the `HCI Event: Number of Completed Packets` for handle 2 back for a short while (we get 3 events for handle 3 in a row). 

```
< HCI Command: LE Create Broadcast Isochronous Group (0x08|0x0068) plen 31                                                                  #4822 48.269800
        Handle: 0x00
        Advertising Handle: 0x00
        Number of BIS: 2
        SDU Interval: 10000 us (0x002710)
        Maximum SDU size: 40
        Maximum Latency: 10 ms (0x000a)
        RTN: 0x02
        PHY: LE 2M (0x02)
        Packing: Sequential (0x00)
        Framing: Unframed (0x00)
        Encryption: 0x00
        Broadcast Code: 00000000000000000000000000000000
> HCI Event: Command Status (0x0f) plen 4                                                                                                   #4823 48.270400
      LE Create Broadcast Isochronous Group (0x08|0x0068) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 23                                                                                                   #4824 48.279400
      LE Broadcast Isochronous Group Complete (0x1b)
        Status: Success (0x00)
        Handle: 0x00
        BIG Synchronization Delay: 0 us (0x000000)
        Transport Latency: 0 us (0x000000)
        PHY: LE 2M (0x02)
        NSE: 4
        BN: 1
        PTO: 1
        IRC: 3
        Maximum PDU: 40
        ISO Interval: 0.00 msec (0x0000)
        Connection Handle #0: 2
        Connection Handle #1: 3
< HCI Command: LE Setup Isochronous Data Path (0x08|0x006e) plen 29                                                                         #4825 48.279500
        invalid packet size
        02 00 00 00 06 00 00 00 00 00 00 00 10 02 01 03  ................
        02 02 01 05 03 01 00 00 00 03 04 28 00           ...........(.   
> HCI Event: Command Complete (0x0e) plen 6                                                                                                 #4826 48.279800
      LE Setup Isochronous Data Path (0x08|0x006e) ncmd 1
        Status: Success (0x00)
        Handle: 2
< HCI Command: LE Setup Isochronous Data Path (0x08|0x006e) plen 29                                                                         #4827 48.279900
        invalid packet size
[... 96 lines cut ...]
        Handle: 2 Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Count: 1
        #4838: len 44 (4 Kb/s)
        Latency: 87 msec (21-87 msec ~65 msec)
< ISO Data TX: Handle 2 flags 0x02 dlen 44                                                                                                  #4858 48.388900
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                      #4859 48.399500
        Num handles: 1
        Handle: 3 Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Count: 1
        #4848: len 44 (4 Kb/s)
        Latency: 75 msec (20-75 msec ~66 msec)
< ISO Data TX: Handle 3 flags 0x02 dlen 44                                                                                                  #4860 48.399700
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                      #4861 48.410600
        Num handles: 1
        Handle: 2 Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Count: 1
        #4842: len 44 (3 Kb/s)
        Latency: 98 msec (21-98 msec ~81 msec)
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: 2f64ac9ab6d44f910bcd81df0a6ac27599bed148

**Additional context**
N/A

Thalley, 2023-10-04:
A similar issue can be seen in BSIM test running for this: https://github.com/zephyrproject-rtos/zephyr/actions/runs/6408192010/job/17396629032

cvinayak, 2023-10-05:
Could you please try this patch:
[0001-tests-bsim-Bluetooth-BAP-Broadcast-Advertising-Inter.patch](https://github.com/zephyrproject-rtos/zephyr/files/12818224/0001-tests-bsim-Bluetooth-BAP-Broadcast-Advertising-Inter.patch)

basically have the advertising interval non-overlapping with ISO Interval:

[image]

cvinayak, 2023-10-05:
The sample does not provide sequential packet sequence number to the Controller when sending ISO data for two or more streams.

I used on bsim `CONFIG_BT_CTLR_ADVANCED_FEATURES=y` and  `CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y`  to get the below (full change patch file here: [0001-samples-Bluetooth-Broadcast-audio-source-and-sink-co.patch](https://github.com/zephyrproject-rtos/zephyr/files/12821571/0001-samples-Bluetooth-Broadcast-audio-source-and-sink-co.patch))

[image]

closed 2023-10-06 by jhedberg